aws/amazon-freertos #3065: ESP32 WIFI_SetMode fails before calling WIFI_On

jonathanfoster, 2021-03-25:
**Describe the bug**

Calling `WIFI_SetMode` to set the Wi-Fi mode prior before calling `WIFI_On` on the ESP32 causes an assertion failure.

```c
../freertos/freertos_kernel/queue.c:1473 (xQueueSemaphoreTake)- assert failed!
```

**System information**
- Hardware board: Espressif ESP32 DevKitC
- IDE: CLion
- Operating System: macOS
- Version of FreeRTOS: 202012.00
- Project: Custom Application

**Expected behavior**

You should be able to call `WIFI_SetMode` to set the Wi-Fi mode before calling `WIFI_On`.

**Code to reproduce the bug**

```c
nvs_flash_init();
tcpip_adapter_init();
WIFI_SetMode(eWiFiModeStation);
WIFI_On();
```

**Additional context**

The issue appears to be caused by only initializing the Wi-Fi sync mutex in `WIFI_On`. https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/ports/wifi/iot_wifi.c#L417

muneebahmed10, 2021-03-25:
Hi @jonathanfoster,

Is there a reason you need to call `WIFI_SetMode` before `WIFI_On`? Looking at [`WIFI_On`](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/wifi/include/iot_wifi.h#L517),
> This function turns on Wi-Fi module,initializes the drivers and must be called
> before calling any other Wi-Fi API

it's necessary to initialize the module first before calling `WIFI_SetMode`.

Let me know if you have further questions.

jonathanfoster, 2021-03-26:
Thanks for the clarification, @muneebahmed10.

closed 2021-03-26 by jonathanfoster