aws/amazon-freertos #3067: Fleet provisioning demo on AWS FreeRTOS

ncohen17, 2021-03-26:
Hi, 

I wanted to know if there is a demo or support for the fleet provisioning feature in AWS FreeRTOS in order to automate the certificate provisioning between the ESP32 and AWS. 

Looking forward to hearing from you!

muneebahmed10, 2021-03-26:
Hi @ncohen17,

While we do not have a demo for Fleet Provisioning in this repository, you can use the existing [coreMQTT library](https://github.com/FreeRTOS/coreMQTT/tree/ff9830a338b6a759c5574f991b02d6f99abadc42) submoduled in this repo to make requests on [Fleet Provisioning topics](https://docs.aws.amazon.com/iot/latest/developerguide/fleet-provision-api.html). Demos for the MQTT library can be found [here](https://github.com/aws/amazon-freertos/tree/master/demos/coreMQTT)

Additionally, the [v4_beta_deprecated](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta_deprecated) branch of the AWS IoT Embedded C SDK has a [Fleet Provisioning Demo](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta_deprecated/demos/src/Provisioning). Although it uses a deprecated MQTT library, it is largely compatible with the [MQTT Compatibility Layer](https://github.com/aws/amazon-freertos/tree/master/libraries/c_sdk/standard/mqtt) in this repository, save for some differences in function names. We do not recommend this as the library has been deprecated, but it remains an option if you are willing to work with a deprecated library.

The development of a new Fleet Provisioning library - along with accompanying demos - is currently on our roadmap. However, we are currently unable to provide any timelines as for when this may be released.

Let us know if you have any further questions.

ncohen17, 2021-03-29:
Hi @muneebahmed10 ,

Thanks for your reply,

which option out of the two would you recommend to go for in order to achieve the fleet provisioning feature in the FreeRTOS on the ESP32 ?

muneebahmed10, 2021-03-29:
Hi @ncohen17,

Since the v4_beta branch of the AWS IoT Embedded C SDK has been deprecated, I would recommend the former option using coreMQTT. You would have to subscribe and publish to Fleet Provisioning topics yourself, but you would be using the latest releases of our libraries.

ncohen17, 2021-03-29:
Hi @muneebahmed10 ,

thanks for your reply, 

Is there more information or example/sample on how this could be done ?

muneebahmed10, 2021-03-30:
Hi @ncohen17,

For Fleet Provisioning, you just need to make requests to the [Fleet Provisioning MQTT APIs](https://docs.aws.amazon.com/iot/latest/developerguide/fleet-provision-api.html). You can find a general demo on subscribing and publishing with coreMQTT [here](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/mqtt_demo_mutual_auth.c). You could use that as a starting point and make the necessary changes for Fleet Provisioning topics and payloads.

I don't believe it would be worthwhile for you to try porting the [AWS IoT C++ Device SDK](https://github.com/aws/aws-iot-device-sdk-cpp-v2) to FreeRTOS, as that SDK not only uses a different MQTT stack, but is not geared towards embedded devices like those running FreeRTOS. It would be simpler to just make the Fleet Provisioning requests using coreMQTT.

closed 2021-05-22 by aggarw13

ncohen17, 2021-09-20:
Hi @muneebahmed10 , 

What about now ? Is there anything new updates? I saw that there is fleet provisioning library for the embedded C SDK available now. 

`https://github.com/aws/fleet-provisioning-for-aws-iot-embedded-sdk`

Is that an option to implement in the FreeRTOS or only through making requests to the Fleet Provisioning MQTT APIs that you mentioned in the previous reply.

sidd2217, 2021-11-30:
> Hi @muneebahmed10 ,
>
> What about now ? Is there anything new updates? I saw that there is fleet provisioning library for the embedded C SDK available now. [...]

Hi ,
Have you found out a way to burn this on esp32?

AlvaroModuly, 2022-05-06:
Any update on this?
I'm just starting to code the fleet provisions in the ESP32, and want to know if there are some examples/samples on how this could be done.