aws/amazon-freertos #3070: Why OTA Agent using c_sdk mqtt instead of coreMQTT

samuelbles07, 2021-03-26:
Hi,

I am wondering, based on recommendation from [Readme](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/mqtt/README.md) file of c_sdk mqtt, to use coreMQTT library because it's newer version, memory usage, etc. But i am just curious why MQTT OTA Agent still using c_sdk mqtt? Is there any reason or just not the priority to update the OTA library? And how long c_sdk mqtt will be supported?

Maybe this is out of topic but still related to OTA. Is it possible using 1 channel MQTT connection for main application and OTA? Or maybe the main application receive notification from IOT Core when there is a new update then main application start OTA agent by using main MQTT connection context?

Thank you!

muneebahmed10, 2021-03-27:
Hello @samuelbles07,

We are still in the process of bringing the [latest release of OTA](https://github.com/aws/ota-for-aws-iot-embedded-sdk/tree/v3.0.0) to this repository. One reason we have not done so yet is that the OTA demo uses the MQTT connection from multiple tasks, while the coreMQTT library is neither thread-aware nor thread-safe.

With coreMQTT, a single MQTT connection cannot be shared among multiple tasks. We are currently developing an [MQTT Agent library](https://github.com/FreeRTOS/coreMQTT-Agent) to serialize calls to coreMQTT in a thread-safe manner. While the MQTT Agent has not been released yet, we have written and tested [several demos](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent) with AWS IoT libraries and FreeRTOS, including one for [OTA](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent/blob/main/source/ota_over_mqtt_demo.c). We recommend you try these demos if you wish to share an MQTT connection with multiple tasks.

Please let us know if you have any further questions.

samuelbles07, 2021-03-27:
Hi @muneebahmed10 , thanks for the clarification, i'll try what you recommend. While waiting for the release, i'll just use coreMQTT for my main application and disconnect when an update available from my application for production.

Thank you for your response and looking forward for the release!

closed 2021-03-27 by samuelbles07