aws/amazon-freertos #3075: _httpReconnect always fails

Sherryzhang2, 2021-03-30:
**Describe the bug**
In the OTA process, I choice the HTTPS to download the image data. But it always fails in _httpReconnect.
Th reason is that in IotHttpsClient_Connect, it disconnects the connection by calling IotHttpsClient_Disconnect. IotHttpsClient_Disconnect  firstly calls  _networkDisconnect to disconnect the connection in which the _FLAG_RECEIVE_TASK_EXITED flag is cleared and the _FLAG_SHUTDOWN flag is set. Then IotHttpsClient_Disconnect  calls _networkDestroy to destroy the connect in which it asserts that connection flag _FLAG_RECEIVE_TASK_EXITED bit is set.

Proposed fix, in IotNetworkAfr_Destroy, the connection flag _FLAG_SHUTDOWN should be used in the assert instead of the _FLAG_RECEIVE_TASK_EXITED flag which has already be cleared in  IotNetworkAfr_Close.

Sherryzhang2, 2021-03-30:
I think the root cause of https://github.com/aws/amazon-freertos/issues/3073 is the same as this issue.

muneebahmed10, 2021-03-31:
Hi @Sherryzhang2,

Thank you for bringing this to our attention. I've created #3082 to fix this. Can you pull that change and confirm that the issue is resolved?

Thank you

muneebahmed10, 2021-04-15:
Hi @Sherryzhang2,

Thank you again for reporting this bug. I will close this issue. Please open another issue if you have any other questions or bug reports.

Thank you

closed 2021-04-15 by muneebahmed10

Sherryzhang2, 2021-04-15:
@muneebahmed10 Thanks for your prompt fix.