aws/amazon-freertos #3078: How to make an OTA Rollback using AWS FreeRTOS V202002.00 and

Yoimer, 2021-03-31:
Hello there. Based on https://docs.aws.amazon.com/freertos/latest/userguide/ota-agent-library.html, the document mentions that `the OTA Agent library lets you define application-specific logic for testing, committing, or rolling back a firmware update`.

My question is: What is the correct procedure to accomplish a rollback firmware update?. Is there any documentation available to follow along?

Currently, I am running  an OTA + MQTT code using `FreeRTOS V202002.00`, and I like to make a ota rollback on using my code here:
https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/cb8e28d03e59292bd489a2b3a037a23f9116a05d/demos/ota/V202002.00_aws_iot_ota_update_demo_and_mqtt.c.

However, my goal is to make ota rollbacks using latest version,  `FreeRTOS V202012.00`, as seen here: https://github.com/Yoimer/aws_freertos_esp32_components/blob/main/demos/ota/aws_iot_ota_update_demo.c

Looking forward to seeing your comments soon. Thanks in advance.

pvyawaha, 2021-03-31:
Hello,

The OTA Agent library supports rollback only on self-test failure during the update process. So when a new firmware is downloaded and its crypto signature is verified it is executed in self-test mode. The application receives `App_OTACompleteCallback` with event of type `eOTA_JobEvent_StartTest` where before setting the image state any application specific self-test can be performed. The OTA Agent library does not support general rollback so the firmware version must be of higher version than the version on the device. 

Please let me know if you have more question on this.

Yoimer, 2021-03-31:
@pvyawaha Thanks a lot for responding. Now this part is clear thanks to you explanation. Have a great day.

closed 2021-03-31 by Yoimer