aws/amazon-freertos #3079: OTA Agent VS AWS IoT Jobs

Yoimer, 2021-03-31:
Hello there. I hope you're doing fine.

Is there any difference, advantage using `OTA Agent `  (ota demo) instead of https://docs.aws.amazon.com/freertos/latest/userguide/ota-agent-library.html and `jobs_for_aws demo` https://docs.aws.amazon.com/freertos/latest/userguide/freertos-lib-jobs.html?

I am asking this since because I've tested many times the OTA demo and basically it's an aws ota job made on the AWS IoT Console for delivering a new firmware as a bin file located on an `s3` bucket. Would you be kind enough to help me out by explaining which benefits, advantage, disadvantage or specific use case where I choose one or another and why?

Looking forward to hearing from you soon. Thanks in advance.

pvyawaha, 2021-04-02:
Hello ,

The OTA Agent library provides functionality to receive job notifications, parse job document , update job status and use the metadata to download the update file, verify, test and activate it. So it is client library providing features required for doing complete firmware update process. The OTA Agent library with latest version https://freertos.org/ota/index.html also has feature to download other filetypes and use the OTA pal and callbacks provided to extend it. 

The Jobs library provides a convenient API to compose and recognize the MQTT topic strings used by the Jobs service. Please refer https://freertos.org/iot-jobs/index.html.

Yoimer, 2021-04-02:
@pvyawaha Thanks lot for responding. Clearer now. Have a wonderful day!

pvyawaha, 2021-04-05:
You are welcome , closing this issue now.

closed 2021-04-05 by pvyawaha