aws/amazon-freertos #3080: Running multiple demos

tjsp10, 2021-03-31:
Hello, i am trying to run OTA + MQTT demo together. i followed [this](https://github.com/aws/amazon-freertos/issues/2569). in that MQTT works fine nut when it complete OTA Complete callback it gives some abort. i am using amazon-freertos (202012.00-branch). i attached log below. can someone help this?
`
[iot_thread] State: WaitingForFileBlock  Received: 276   Queued: 0   Processed: 0   Dropped: 0
[iot_thread] [INFO ][MQTT][321380] (MQTT connection 0x3ffee148) MQTT PUBLISH operation queued.
 [iot_thread] [INFO ][MQTT][321380] (MQTT connection 0x3ffee148, PUBLISH operation 0x3ffe96d0) Waiting for operation completion.
 [OTA Agent Task] [prvIngestDataBlock] File receive complete and signature is valid.
 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
 [OTA Agent Task] [prvPublishStatusMessage] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x90005"}}
[OTA Agent Task] [INFO ][MQTT][323010] (MQTT connection 0x3ffee148) MQTT PUBLISH operation queued.
 [OTA Agent Task] [INFO ][MQTT][323010] (MQTT connection 0x3ffee148, PUBLISH operation 0x3ffeaaf8) Waiting for operation completion.
 [iot_thread] [INFO ][MQTT][323050] (MQTT connection 0x3ffee148, PUBLISH operation 0x3ffe96d0) Wait complete with result SUCCESS.
 [OTA Agent Task] [INFO ][MQTT][323140] (MQTT connection 0x3ffee148, PUBLISH operation 0x3ffeaaf8) Wait complete with result SUCCESS.
[OTA Agent Task] [prvPublishStatusMessage] 'IN_PROGRESS' to $aws/things/OTA_Device/jobs/AFR_OTA-ID_OTAMQTT2/update
 [OTA Agent Task] Received eOTA_JobEvent_Activate callback from OTA Agent.
 [OTA Agent Task] [INFO ][MQTT][323140] (MQTT connection 0x3ffee148) Disconnecting connection.
 [OTA Agent Task] [INFO] [MQTT] [core_mqtt.c:2149] 2795 32315 [OTA Agent Task] Disconnected from the broker.2796 32315 [OTA Agent Task]
 [OTA Agent Task] [INFO ][MQTT][323150] (MQTT connection 0x3ffee148) Network connection closed.
I (323224) boot_comm: chip revision: 1, min. application chip revision: 0
I (323224) esp_image: segment 0: paddr=0x001a0020 vaddr=0x3f400020 size=0x30d20 (199968) map
I (323404) esp_image: segment 1: paddr=0x001d0d48 vaddr=0x3ffbdb60 size=0x03990 ( 14736)
I (323424) esp_image: segment 2: paddr=0x001d46e0 vaddr=0x40080000 size=0x00400 (  1024)
0x40080000: _WindowOverflow4 at D:\AWS\amazon-freertos\build/../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (323424) esp_image: segment 3: paddr=0x001d4ae8 vaddr=0x40080400 size=0x0b528 ( 46376)
I (323464) esp_image: segment 4: paddr=0x001e0018 vaddr=0x400d0018 size=0xc07d0 (788432) map
0x400d0018: _flash_cache_start at ??:?

../freertos_kernel/queue.c:1477 (xQueueSemaphoreTake)- assert failed!
abort() was called at PC 0x4008f609 on core 0
0x4008f609: xQueueSemaphoreTake at D:\AWS\amazon-freertos\build/../freertos_kernel/queue.c:2429

ELF file SHA256: 89931430984849d2

Backtrace: 0x4008d1d0:0x3ffb3c70 0x4008d401:0x3ffb3c90 0x4008f609:0x3ffb3cb0 0x40151523:0x3ffb3cf0 0x4015153d:0x3ffb3d10 0x40146ca0:0x3ffb3d30 0x401486b0:0x3ffb3d60 0x4014709e:0x3ffb3d80 0x40147a5c:0x3ffb3db0 0x400d30d1:0x3ffb3de0 0x400d31f2:0x3ffb3e20 0x400d32d9:0x3ffb3e80 0x400d2e4a:0x3ffb3ea0 0x401513fb:0x3ffb3ed0
0x4008d1d0: invoke_abort at D:\AWS\amazon-freertos\build/../vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008d401: abort at D:\AWS\amazon-freertos\build/../vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008f609: xQueueSemaphoreTake at D:\AWS\amazon-freertos\build/../freertos_kernel/queue.c:2429

0x40151523: prIotMutexTimedLock at D:\AWS\amazon-freertos\build/../libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x4015153d: IotMutex_Lock at D:\AWS\amazon-freertos\build/../libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x40146ca0: _IotMqtt_IncrementConnectionReferences at D:\AWS\amazon-freertos\build/../libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:758

0x401486b0: _IotMqtt_CreateOperation at D:\AWS\amazon-freertos\build/../libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c:355

0x4014709e: IotMqtt_Publish at D:\AWS\amazon-freertos\build/../libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:1670

0x40147a5c: IotMqtt_TimedPublish at D:\AWS\amazon-freertos\build/../libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:1820

0x400d30d1: prvPublishMessage at D:\AWS\amazon-freertos\build/../demos/ota/aws_iot_ota_update_demo.c:227

0x400d31f2: vRunOTAUpdateDemo at D:\AWS\amazon-freertos\build/../demos/ota/aws_iot_ota_update_demo.c:290

0x400d32d9: vStartOTAUpdateDemoTask at D:\AWS\amazon-freertos\build/../demos/ota/aws_iot_ota_update_demo.c:437

0x400d2e4a: runDemoTask at D:\AWS\amazon-freertos\build/../demos/demo_runner/iot_demo_freertos.c:314

0x401513fb: _threadRoutineWrapper at D:\AWS\amazon-freertos\build/../libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

Rebooting...`

ncohen17, 2021-04-01:
Hi @muneebahmed10,
Is it possible to run both the MQTT + Shadow demo examples at the same time ?

Yoimer, 2021-04-01:
@tjsp10 Hi. I've tested this https://github.com/aws/amazon-freertos/issues/2569#issuecomment-708716699 and it works just fine. I can make any OTA and at the same time it sends data to AWS using mqtt

tjsp10, 2021-04-02:
@Yoimer. did u used using amazon-freertos (202012.00-branch)? because link given in [#2569 (comment)](https://github.com/aws/amazon-freertos/issues/2569#issuecomment-708716699) using FreeRTOS V202002.00

Yoimer, 2021-04-02:
> @Yoimer. did u tried amazon-freertos (202012.00-branch)? because link given in [#2569 (comment)](https://github.com/aws/amazon-freertos/issues/2569#issuecomment-708716699) using FreeRTOS V202002.00

@tjsp10 The version for the current `ota + mqtt` you can see it here, [version](https://github.com/aws/amazon-freertos/blob/7dcec10866d45a2f7207f0c15dd38dfb7b2a25a6/demos/ota/aws_iot_ota_update_demo.c#L2).

You can't use that code on what you call `202012.00-branch` repository.

muneebahmed10, 2021-04-10:
Hi @tjsp10,

Did the [above response](https://github.com/aws/amazon-freertos/issues/3080#issuecomment-811384571) answer your question? Were you able run the [OTA demo with the MQTT Agent](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent/blob/main/source/ota_over_mqtt_demo.c)?

@ncohen17, The MQTT and shadow demos in this repository are not thread safe as they directly use coreMQTT, so they cannot be run concurrently. If you want to run MQTT and shadow demos concurrently, I would recommend checking out the [demos using the thread-safe MQTT Agent](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent/tree/main/source).

@Yoimer, that version header was not updated for the 202012 release as the demo still uses the old MQTT API, the same as the 202002 release. That OTA demo can be run in the 202012 release using the [MQTT Compatibility layer](https://github.com/aws/amazon-freertos/tree/master/libraries/c_sdk/standard/mqtt). However, since the [MQTT demo](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/mqtt_demo_mutual_auth.c) in the 202012 release uses the latest MQTT APIs, the two demos are not compatible with each other.

Yoimer, 2021-04-10:
@muneebahmed10 Thanks a lot for clarifying.!!

AniruddhaKanhere, 2021-04-15:
Hello @tjsp10, did the above comment answer your question?

tjsp10, 2021-04-21:
@muneebahmed10. thanks for your reply. i am new in this aws freertos. Example [here](https://github.com/FreeRTOS/Lab-Project-coreMQTT-Agent) shown is for "Winsim" Hardware Platfarm. how i can use it for ESP32?

muneebahmed10, 2021-04-26:
Hi @tjsp10,

While there are build files for only Winsim and QEMU, the [source](https://github.com/FreeRTOS/coreMQTT-Agent-Demos/tree/main/source) and [library](https://github.com/FreeRTOS/coreMQTT-Agent-Demos/tree/main/lib) files are not designed for any specific hardware platform. You should add those files to your project and build as you would any ESP32 project. You can find instructions for getting started with ESP32 and FreeRTOS [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html).

Raghav3107, 2021-05-29:
Hello @tjsp10, 
you can use this repository [here](https://github.com/Raghav3107/amazon-freertos-workspace-v202012). I run the MQTT, shadow and OTA  simultaneously here.

muneebahmed10, 2021-07-06:
We will close this issue as all questions here have been answered. Please open a new issue for any further questions.

closed 2021-07-06 by muneebahmed10