aws/amazon-freertos #3081: multi_heap_assert called when calling SecureSocketsTransport_Connect()

samuelbles07, 2021-03-31:
**Is your feature request related to a problem? Please describe.**  
A clear and concise description of what the problem is. What are the use cases of the feature to be added? 

**Describe the solution you would like.**  
A clear and concise description of what you want to happen.

**Additional context**  
Add any other information (including rationale, requirements, or references) about the feature request here.

Hi,

My case is i am using amazon-freertos as submodule, and currently i want to integrate coreMQTT `mqtt_demo_mutual_auth` from demo directory to my main project. Everything in the main code is basically from the demo code, from wifi and mqtt. But when the code start execute `SecureSocketsTransport_Connect()`, multi_heal is called. Below is the log.

```
0 0 [main] Initializing lwIP TCP stack
I (635) PKCS11: Initializing NVS partition: "storage"
E (671) PKCS11: failed nvs get file size 4354 0
1 2 [main] Write certificate...
I (675) wifi:wifi driver task: 3ffb9568, prio:23, stack:6656, core=0
I (675) system_api: Base MAC address is not set
I (675) system_api: read default base MAC address from EFUSE
I (705) wifi:wifi firmware version: 1865b55
I (705) wifi:wifi certification version: v7.0
I (705) wifi:config NVS flash: enabled
I (705) wifi:config nano formating: disabled
I (705) wifi:Init data frame dynamic rx buffer num: 32
I (705) wifi:Init management frame dynamic rx buffer num: 32
I (715) wifi:Init management short buffer num: 32
I (715) wifi:Init dynamic tx buffer num: 32
I (725) wifi:Init static rx buffer size: 1600
I (725) wifi:Init static rx buffer num: 10
I (735) wifi:Init dynamic rx buffer num: 32
I (735) wifi_init: rx ba win: 6
I (735) wifi_init: tcpip mbox: 32
I (745) wifi_init: udp mbox: 6
I (745) wifi_init: tcp mbox: 6
I (755) wifi_init: tcp tx win: 5744
I (755) wifi_init: tcp rx win: 5744
I (755) wifi_init: tcp mss: 1440
I (765) wifi_init: WiFi IRAM OP enabled
I (765) wifi_init: WiFi RX IRAM OP enabled
2 15 [main] Success enable wifi, connecting wifi...
3 15 [main] [INFO ][DEMO][150] Credential OK, connecting...
W (785) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (925) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 2
I (935) wifi:mode : sta (3c:61:05:3d:de:7c)
I (935) WIFI: SYSTEM_EVENT_STA_START
I (2025) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
I (2565) wifi:state: init -> auth (b0)
I (2575) wifi:state: auth -> assoc (0)
I (2585) wifi:state: assoc -> run (10)
I (2845) wifi:connected with samuelbles.com, aid = 1, channel 9, BW20, bssid = 3c:84:6a:3a:9a:7a
I (2845) wifi:security: WPA2-PSK, phy: bgn, rssi: -70
I (2845) wifi:pm start, type: 1

I (2855) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (2915) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3625) esp_netif_handlers: sta ip: 192.168.0.200, mask: 255.255.255.0, gw: 192.168.0.1
I (3625) WIFI: SYSTEM_EVENT_STA_GOT_IP
4 300 [sys_evt] [INFO ][DEMO][3000] _wifiEventHandler called!
5 300 [sys_evt] [INFO ][DEMO][3000] Connected to WiFi access point, ip address: 255.255.255.0.
6 301 [main] Wifi connected, start MQTT
E (4565) PKCS11: failed nvs get file size 4354 0
CORRUPT HEAP: multi_heap.c:438 detected at 0x3ffb639c

abort() was called at PC 0x4008afd9 on core 0
0x4008afd9: multi_heap_assert at /Users/inventory-1/tools/amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) multi_heap_malloc_impl at /Users/inventory-1/tools/amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:438

Backtrace:0x40087773:0x3ffb6370 0x40087d85:0x3ffb6390 0x4008e4ba:0x3ffb63b0 0x4008afd9:0x3ffb6420 0x40081709:0x3ffb6440 0x4008194f:0x3ffb6460 0x40080ee8:0x3ffb6480 0x4013df69:0x3ffb64a0 0x40134666:0x3ffb64c0 0x40134753:0x3ffb64e0 0x40134ff2:0x3ffb6500 0x401350ad:0x3ffb6540 0x401353e4:0x3ffb6570 0x401355af:0x3ffb65f0 0x401357f3:0x3ffb6610 0x4013ea0c:0x3ffb6c80 0x4013f931:0x3ffb6cb0 0x4013f9c1:0x3ffb6ce0 0x4013c649:0x3ffb6d10 0x4013c337:0x3ffb6d50 0x4013c35e:0x3ffb6d90 0x4013c4eb:0x3ffb6dc0 0x401453ab:0x3ffb6e00 0x40145491:0x3ffb6e50 0x4014556b:0x3ffb6e80 0x40145835:0x3ffb6ec0 0x40145ef6:0x3ffb6f10 0x4014460a:0x3ffb6f90 0x4014b4dc:0x3ffb6fd0 0x40142e99:0x3ffb6ff0 0x40142ec9:0x3ffb7010 0x400dc3de:0x3ffb7030 0x400d7fd9:0x3ffb7050 0x400d86ea:0x3ffb70b0 0x400d7770:0x3ffb70d0 0x400d7897:0x3ffb7100 0x400d7b25:0x3ffb7120 0x400d2b71:0x3ffb7140 0x400d2e2d:0x3ffb71a0 0x400f385b:0x3ffb72f0 0x40087db9:0x3ffb7320
[... 66 lines cut ...]

0x400d86ea: Sockets_MetricsConnect at /Users/inventory-1/project/build/../freertos/libraries/abstractions/platform/freertos/iot_metrics.c:205

0x400d7770: connectToServer at /Users/inventory-1/project/build/../freertos/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c:336

0x400d7897: establishConnect at /Users/inventory-1/project/build/../freertos/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c:465

0x400d7b25: SecureSocketsTransport_Connect at /Users/inventory-1/project/build/../freertos/libraries/abstractions/transport/secure_sockets/transport_secure_sockets.c:540

0x400d2b71: prvConnectToServerWithBackoffRetries at /Users/inventory-1/project/build/../src/main.c:386

0x400d2e2d: app_main at /Users/inventory-1/project/build/../src/main.c:624

0x400f385b: main_task at /Users/inventory-1/tools/amazon-freertos/vendors/espressif/esp-idf/components/esp32/cpu_start.c:614

0x40087db9: vPortTaskWrapper at /Users/inventory-1/project/build/../freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:154

ELF file SHA256: 3a24354dd7b5c909
```

I don't know what happen, but my guess is when read the client key from NVS? 

Please help,

Thank you!

samuelbles07, 2021-04-01:
I check free heap size before calling `SecureSocketsTransport_Connect` using `esp_get_free_heap_size` is 239564 bytes. I wonder if it's something?

mahavirj, 2021-04-01:
@samuelbles07 

Did you reconfigure stack size of main task? Default value is 3584 bytes, configured using `CONFIG_ESP_MAIN_TASK_STACK_SIZE` which may not be sufficient for TLS operation. I recommend increasing this to at-least 6-7KB and then fine tune it based on high water mark (usage).

samuelbles07, 2021-04-01:
Hi @mahavirj ,

No, i don't configure anything of that scope. Where do i change it, from `menuconfig`?

mahavirj, 2021-04-01:
> No, i don't configure anything of that scope. Where do i change it, from menuconfig?

`idf.py menuconfig → Component config → Common ESP-related → Main task stack size`

Alternatively, you may create new task to start your demo application code.

samuelbles07, 2021-04-01:
Yes that is the problem, thank you!

closed 2021-04-01 by samuelbles07