aws/amazon-freertos #3086: Wifi Provisioning over BLE Crashes on mqtt demo FreeRTOS V202002.00

Yoimer, 2021-04-01:
Hello there team. With `ESP-IDFV3.3` I tested the mqtt demo [iot_demo_mqtt.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/mqtt/iot_demo_mqtt.c) using the ble wifi_provisioning section. 

The `#define IOT_BLE_ENABLE_WIFI_PROVISIONING 1` macro is here [iot_ble_config_defaults.h](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/e1181315727afbfa62aa9aa73d8bd119d7b26225/libraries/c_sdk/standard/ble/include/iot_ble_config_defaults.h). I decided to use the [ble just work pairing](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/libraries/c_sdk/standard/ble/include/iot_ble_config_defaults.h) as shown here.

For the test I used the `IOS` app from the workshop [wifi-provisioning](https://teuteuguy.github.io/amazon-freertos-workshop-docs/labs/04-wifi-provisioning/) which is available on the AppStore [WiFi credentials over BLE](https://apps.apple.com/app/wifi-provision/id1500121277)

Unfortunately, it does not get connected to my local router. I am attaching the whole log. Kindly, would you be kind enough to help me out on this as you always do, please?

Looking forward to hearing from you soon.

```
Microsoft Windows [Version 10.0.19042.906]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS>python vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p COM6 -B demos/mqtt/build
Note: You are using Python 3.8.0. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Setting IDF_PATH environment variable: C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\vendors\espressif\esp-idf
Checking Python dependencies...
Python requirements from C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\vendors\espressif\esp-idf\requirements.txt are satisfied.
Running esptool.py in directory C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\demos\mqtt\build
Executing "C:\Users\Net-prog\AppData\Local\Programs\Python\Python38-32\python.exe C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\vendors\espressif\esp-idf\components/esptool_py/esptool/esptool.py -p COM6 -b 460800 --after hard_reset erase_flash"...
esptool.py v2.8-dev
Serial port COM6
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:8f:f7:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.4s
Hard resetting via RTS pin...
Running ninja in directory C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\demos\mqtt\build
Executing "ninja all"...
[0/2] Re-checking globbed directories...
[1/3] Performing build step for 'bootloader'
ninja: no work to do.
Running esptool.py in directory C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\demos\mqtt\build
Executing "C:\Users\Net-prog\AppData\Local\Programs\Python\Python38-32\python.exe C:\Users\Net-prog\Desktop\esp32-aws-freertos\FreeRTOS\vendors\espressif\esp-idf\components/esptool_py/esptool/esptool.py -p COM6 -b 460800 --after hard_reset write_flash @flash_project_args"...
esptool.py -p COM6 -b 460800 --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin 0x16000 ota_data_initial.bin 0x20000 aws_demos.bin
esptool.py v2.8-dev
Serial port COM6
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:8f:f7:c0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 26656 bytes to 15847...
Wrote 26656 bytes (15847 compressed) at 0x00001000 in 0.4 seconds (effective 587.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 118...
Wrote 3072 bytes (118 compressed) at 0x00008000 in 0.0 seconds (effective 2442.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00016000 in 0.0 seconds (effective 7613.3 kbit/s)...
Hash of data verified.
Compressed 941216 bytes to 585643...
Wrote 941216 bytes (585643 compressed) at 0x00020000 in 13.5 seconds (effective 559.0 kbit/s)...
[... 118 lines cut ...]
19 1929 [iot_thread] [INFO ][DEMO][19290] memory_metrics::demo_task_stack::before::bytes::5852
20 1929 [iot_thread] [INFO ][DEMO][19290] memory_metrics::demo_task_stack::after::bytes::4012
21 1929 [iot_thread] [ERROR][DEMO][19290] Error running demo.
GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19
22 1937 [iot_thread] [ERROR][BLE_GAP][19370] Failed disconnect with Bluetooth status = 1
23 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
24 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:VM5364426-2g )
25 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
26 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:SKYJURTF )
27 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
28 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:SKYJURTF )
29 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
30 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:DIRECT-N9-BRAVIA )
31 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
32 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:Vatra Dornei )
33 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
34 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:DIRECT-47-HP DeskJet 2600 serie )
35 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
36 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:Peirov Mousavi )
37 1982 [iot_thread] [ERROR][BLE_TXRX][19820] TX Failed, channel closed.
38 1982 [iot_thread] [ERROR][BLE_WIFI_PROV][19820] Failed to send scanned network network ( SSID:BTWi-fi )
39 1983 [iot_thread] [INFO ][INIT][19830] SDK cleanup done.
40 1983 [iot_thread] [INFO ][DEMO][19830] -------DEMO FINISHED-------

```

ravibhagavandas, 2021-04-01:
It seems you are trying to run MQTT demo, but the demo is configured to use either a WiFi or a BLE network [here](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/e1181315727afbfa62aa9aa73d8bd119d7b26225/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L55)
So in this case demo starts as soon as BLE is connected for WiFi provisioning, but since the BLE app is not configured for demo it fails and disconnects BLE.
```
5 1272 [ble] [INFO ][DEMO][12720] BLE Connected to remote device, connId = 0

6 1272 [iot_thread] [INFO ][DEMO][12720] Successfully initialized the demo. Network type for the demo: 2
```
Could you set the demo to run only on WiFi by configuring to:
#define democonfigNETWORK_TYPES    ( AWSIOT_NETWORK_TYPE_WIFI )

Once set, demo should only start once a WiFi network is provisioned successfully.

Yoimer, 2021-04-01:
@ravibhagavandas Thanks a lot for your support. I will test that and report here. Have a great day

ravibhagavandas, 2021-04-09:
@Yoimer 
Just checking back, did the suggestions resolved this issue ?

Yoimer, 2021-04-09:
> @Yoimer
> Just checking back, did the suggestions resolved this issue ?

@ravibhagavandas Hello. Thanks for replying. Yes, all good. Thanks for your contribution.

closed 2021-04-09 by ravibhagavandas