aws/amazon-freertos #3090: Issue while Consuming Freertos in CMake project

NaitikSanas, 2021-04-07:
I am trying to create the main.c outside the amazon freertos directory and link it as executable. I am following the getting started guide from amazon https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project
where it provides example on how to consume freertos in own CMake project.

when i copy and paste the example cmakelists.txt provided by the guide and try to compile the project with command 
cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja

i get error that says following stuff : 
  add_executable cannot create target "user_app" because another target with
  the same name already exists.  The existing target is an executable created
  in source directory "C:/Users/Naitik/Documents/cmake_proj".  See
  documentation for policy CMP0002 for more details.

I think that there is issue in CMakeLists.txt provided in guide since when I change the Line set(IDF_PROJECT_EXECUTABLE user_app)
set(IDF_PROJECT_EXECUTABLE user_app2)
it creates the build directory successfully however when i run command
cmake --build build 
if get following errors in the end of compiling : 

[1721/1742] Linking CXX executable user_app
FAILED: user_app
cmd.exe /C "cd . && C:\Users\Naitik\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls  -Wl,--gc-sections -Wl,--cref -Wl,--Map=aws_demos.map -Wl,--undefined=uxTopUsedPriority @CMakeFiles\user_app.rsp  -o user_app  && cd ."
c:/users/naitik/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.project.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
[1722/1742] Generating ld/esp32.project.ld
ninja: build stopped: subcommand failed.

Can Someone please Point out where the issue is ? Do i need to add something else in cmake lists ?

CMakeLists I use looks this which is from the guide provided by amazon :
cmake_minimum_required(VERSION 3.13)

project(freertos_examples)

add_executable(user_app src/main.c)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE user_app2)

# Add FreeRTOS as a subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory(amazon-freertos)

# Link against the mqtt library so that we can use it. Dependencies are transitively
# linked.
target_link_libraries(user_app PRIVATE AFR::mqtt)

Yoimer, 2021-04-07:
@NaitikSanas Check this link https://github.com/tgsong/amazon-freertos-examples

It's explained there how to use `AFR` (Amazon FreeRTOS ESP32) as a CMake project for ESP32.

NaitikSanas, 2021-04-08:
@Yoimer  I tried the example from link you provided but I get same error where it says that 

CMake Error at freertos/vendors/espressif/boards/esp32/CMakeLists.txt:455 (add_executable):
  add_executable cannot create target "afr_demo" because another target with
  the same name already exists.  The existing target is an executable created
  in source directory "C:/Users/Naitik/Documents/gh_demo".  See documentation
  for policy CMP0002 for more details.

I never worked with CMake can you please tell why this happens? I mean I do understand what error is saying is that name afr_demo is used multiple times in CMakeLists.txt 
So when I rename Line : set(IDF_PROJECT_EXECUTABLE afr_demo) to set(IDF_PROJECT_EXECUTABLE afr_demo2)
things starts compiling but I get error about missing headerfiles. 
However I also expect that Cmake list provided in Example to work without any changes so i am unable to understand why it doesnt work as intended my end?

AniruddhaKanhere, 2021-04-14:
Hello @NaitikSanas,
Can you try to do a clean build? What I mean by that is remove all the source code and the `build` directory. Remove everything in the folder you might be working on. After you have done that, clone the source again and try the cmake command.

After that, you should not need to modify the CMakeLists.txt. I generally remove the build directory before running CMake again myself.
If you still get any missing header file errors, could you post them here so that we may have a look?

Thanks!

NaitikSanas, 2021-04-19:
hello! sorry For Late response as you said i deleted build directory and src/main.c only contains 
``` c
app_main(){
}
```
and Cmake looks like this :

```
cmake_minimum_required(VERSION 3.13)

project(freertos_examples)

add_executable(my_app src/main.c)
#Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE my_app)

#Add FreeRTOS as a subdirectory. AFR_BOARD tells which board to target.

set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL “”)

add_subdirectory(freertos)

#Link against the mqtt library so that we can use it. Dependencies are transitively
#linked.

target_link_libraries(my_app PRIVATE AFR::mqtt)

target_link_libraries(my_app PRIVATE AFR::freertos)

target_link_libraries(my_app PRIVATE AFR::kernel)
```

When I work with this cmake I get same error as i mentioned previously : 
add_executable cannot create target "xyz" because another target with the
  same name already exists.  The existing target is an executable created in
  source directory "C:/Users/Naitik/Documents/cmake_proj".  See documentation
  for policy CMP0002 for more details.

if I rename Line : set(IDF_PROJECT_EXECUTABLE my_app) to set(IDF_PROJECT_EXECUTABLE my_app2)
things starts compiling but I get following issue now

[1727/1748] Linking CXX executable my_app
FAILED: my_app
cmd.exe /C "cd . && C:\Users\Naitik\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls  -Wl,--gc-sections -Wl,--cref -Wl,--Map=aws_demos.map -Wl,--undefined=uxTopUsedPriority @CMakeFiles\my_app.rsp  -o my_app  && cd ."
c:/users/naitik/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.project.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
[1728/1748] Generating ld/esp32.project.ld
ninja: build stopped: subcommand failed.

I use following commands : 
freertos\Vendors\espressif\esp-idf\export.bat  //to export idf.py
cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja // Creating build directory
cmake --build build //Compile the project 

Thank you for your Response.

motumboe, 2021-05-25:
Hi @NaitikSanas 
did you find out how to solve the issue?
Same problem here.
Cheers
 Marco

NaitikSanas, 2021-05-27:
Hello @motumboe 
No Unfortunately I still could not resolve this issue. I also raised issue in freertos forum which is [here](https://forums.freertos.org/t/creating-cmake-project-with-amazon-freertos/12190)  but got no useful responses.

yanjos-dev, 2021-06-30:
Hi @NaitikSanas, I apologize for the delay. Were you able to find a solution or would you still like assistance with this?

laszlopataki, 2021-07-15:
I had the same issue. [This](https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2) example helped me to find the solution. 
Make sure you check out the _feature/idf_v4.2_ branch.

dFohlen, 2021-07-20:
I ran into the same problems. Which version are you using? 

Version [202107.00](https://github.com/aws/amazon-freertos/releases/tag/202107.00) uses the newer ESP-IDF v4.2 instead of ESP-IDF v3.3 in version [202012.00](https://github.com/aws/amazon-freertos/releases/tag/202012.00).

However, the "AWS -> Documentation -> FreeRTOS -> User Guide" seems to be inconsistent. In the [English documentation](https://docs.aws.amazon.com/en_en/freertos/latest/userguide/getting_started_espressif.html), ESP-IDF v3.3 and v4.2 contents are mixed, while on the [German side](https://docs.aws.amazon.com/de_de/freertos/latest/userguide/getting_started_espressif.html) you can switch between the two versions.

ESP-IDF v3.3:
```
cmake_minimum_required(VERSION 3.13)

project(freertos_examples)

add_executable(my_app src/main.c)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE my_app)

# Add FreeRTOS as a subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory(freertos)

# Link against the mqtt library so that we can use it. Dependencies are transitively
# linked.
target_link_libraries(my_app PRIVATE AFR::mqtt)
```

ESP-IDF v4.2
```
cmake_minimum_required(VERSION 3.13)

project(freertos_examples)

# Pull in ESP-IDF defined CMake build infrastructure logic to configure the project, discover all the components, etc
set(esp_idf_dir "${CMAKE_CURRENT_LIST_DIR}/freertos/vendors/espressif/esp-idf")
include(${esp_idf_dir}/tools/cmake/idf.cmake)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE my_app)
get_filename_component(
    IDF_EXECUTABLE_SRCS
    "src/main.c" ABSOLUTE
    ) 
```

Yoimer, 2021-07-20:
> I ran into the same problems. Which version are you using?
>
> Version [202107.00](https://github.com/aws/amazon-freertos/releases/tag/202107.00) uses the newer ESP-IDF v4.2 instead of ESP-IDF v3.3 in [...]

@dFohlen To put it simple. Versions 202007 and 202012 , zip files downloaded from AWS Console use ESP-IDF V3.3(Only the toolchain, not the whole IDF).
Versions 202007 and 202012 downloaded from Github uses ESP-IDF V4.2 (The whole IDF).
I haven't tested the latest version 202107, but it is quite likely it also uses the ESP-IDF V4.2 (The whole IDF)
Hope that clarifies a bit.

Developer-error, 2021-08-21:
@Yoimer i tried your update, now getting this error. Please help .

Linking CXX executable freertos_examples.elf
FAILED: freertos_examples.elf

NaitikSanas, 2021-08-23:
Sorry for late response but unfortunately I could not manage to make the Amazon-Freertos work my end with all solution provided by here or forums so i just chose to implement own APIs for OTA and used AWS IoT SDK.

yanjos-dev, 2021-10-04:
I'm going to close this issue due to inactivity. Please open a new ticket if there are any further issues creating a project.

closed 2021-10-04 by yanjos-dev