aws/amazon-freertos #3091: aws_demos.elf is not created when building AWS FreeRTOS V202002.00 for ESP32

Yoimer, 2021-04-07:
Hello there.

I want to debug the demos from this repository [AFR_ESP32_FreeRTOS_V202002.00](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00). You might clone it and download at any time if needed, please.

When I execute `cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DCMAKE_BUILD_TYPE=Debug -GNinja -S . -B demos/mqtt/build` and 

`python vendors/espressif/esp-idf/tools/idf.py build -B demos/mqtt/build`

from `FreeRTOS` folder, it does not create the `aws_demos.elf` on `demos/mqtt/build`

On the other hand, when I use this repo [extra_demos](https://github.com/Yoimer/aws_freertos_esp32_components/tree/extra_demos) which is using  `FreeRTOS V202012.00`, it generates the `aws_demo.elf` on `demos/coreMQTT/build` when I execute 

`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DCMAKE_BUILD_TYPE=Debug -GNinja -S . -B demos/coreMQTT/build`

`python vendors/espressif/esp-idf/tools/idf.py build -B demos/coreMQTT/build`

My questions are:

How could get the `aws_demos.elf` file when using `V202002.00`? This happens when I try any available demo, not only `mqtt`.

I was able to debug and generate breakpoints using the `ESP-PROG` hardware on `V202012.00` using the [projects](https://github.com/Yoimer/aws_freertos_esp32_components/tree/extra_demos/projects/espressif/esp32/make) folder from this repository.

Looking forward to hearing from the team.

Thanks in advance.

Yoimer, 2021-04-13:
Hello everyone. Have you had the chance of reading, this? Thanks a lot

paulbartell, 2021-04-28:
@Yoimer : Apologies for the late response. In more recent versions of amazon-freertos for esp32, the generated binary is named aws_demos.bin rather than aws_demos.elf.

closed 2021-04-28 by paulbartell

Yoimer, 2021-04-28:
@paulbartell Hi, thanks for replying. Yes, I figured out that the `aws_demos.elf` file is not present on` V202002.00` giving me no change to debugging over there.
However on more later versions such as `V202007.00` and `V202012.00` the `aws_demos.elf` file does appear after building so I can use the `ESP-PROG` probe for the ESP32 and begging stepping through the code and even make breakpoints.

Thanks again, and have a great day.