aws/amazon-freertos #3092: Error when I try to install the espressif toolchain for my MacOS

Mitsutaka80, 2021-04-08:
**Describe the bug**
I get an error, when I run "vendors/espressif/esp-idf/install.sh" after downloading the FreeRTOS package from FreeRTOS console as instructed at the getting started document for [ESP32 AWS FreeRTOS](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf42)

Error
-bash: vendors/espressif/esp-idf/install.sh: Permission denied

I tried to use sudo as well and I got this 
vendors/espressif/esp-idf/install.sh: command not found

**System information**
- Which hardware board or part numbers?   ESP32-WROOM-32U
- IDE used   VS Code 
- Operating System [Windows|Linux|MacOS]   MacOS
- Version of FreeRTOS (run ``git describe --tags`` to find it)   202012.00 (Downloaded from the FreeRTOS console)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
As in the document, I would expect that I could download and install the espressif toolchain for my platform. 

Thank you!

paulbartell, 2021-04-08:
@Mitsutaka80 : As a workaround, can you try the following command to make the script executable. It looks like somewhere the executable permission is getting dropped.

`chmod +x vendors/espressif/esp-idf/install.sh`

You may need to do the same for other shell scripts in your download package. Something like this should suffice:

`find . -name "*.sh" | xargs chmod +x`

n9wxu, 2021-04-08:
Additionally, many of these commands assume bash and MacOS switched to ZSH.  I have had more success "sourcing" the different scripts.
`source vendoers/espressif/esp-idf/install.sh`

n9wxu, 2021-04-09:
Further analysis shows that the install.sh is loosing its permissions when it is extracted from a zip file.  zip files are unable to store unix permissions.
the most correct fix is
`chmod +x vendors/espressif/esp-idf/install.sh`
The other scripts are intended to be 'sourced' so no execute permission change is required.

We will be updating the getting started documentation to teach these changes.

closed 2021-04-09 by n9wxu

paulbartell, 2021-04-09:
You may run into some other issues with recent versions of MAC OSX and the revision of esp-idf included the latest FreeRTOS release from the console (esp-idf v3.3). 

Please follow the instructions for this revision of esp-idf in the [AWS documentation](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#setup-espressif-idf33) and on [espressif's website](https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/macos-setup.html).