aws/amazon-freertos #3093: ESP32 does not keep sending data to AWS after reconnecting to Access Point. Wifi credentials were saved previously using BLE mobile app

Yoimer, 2021-04-09:
Hello there. I am still testing the `V202002` of AWS FreeRTOS ESP32 from this repo [V202002.00](AFR_ESP32_FreeRTOS_V202002.00), which is just the zip file generated from `AWS Console` The demo I'm running is this [aws_iot_ota_update_demo_and_mqtt.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/ota/V202002.00_aws_iot_ota_update_demo_and_mqtt.c)  . I am using our custom BLE mobile app, which is linked to our AWS account using `Cognito`, our `IoT device` and `IoT Rule` based on this [freertos-ble-mobile](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ble-mobile.html).

I am testing all the `critical` scenarios of the whole solution. And of those is when (for any reason) the router has to get restarted (someone has to unplug it from the outlet and plug it back). During the whole test, I am `pinging` (`ping 192.168.5.11 -t`) the local IP address of the ESP32 (it is generated with `DHCP` from the router). The firmware is able to reconnect by itself to the router. However, it does not continue sending data to AWS. What it shows is the next log:

```
I (161706) WIFI: SYSTEM_EVENT_STA_CONNECTED
539 16155 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
540 16155 [iot_thread] [INFO ][MQTT][161550] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
541 16155 [iot_thread] [INFO ][MQTT][161550] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
542 16156 [iot_thread] [ERROR][NET][161560] Error -27648 while sending data.
543 16157 [iot_thread] [INFO ][MQTT][161570] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

544 16253 [IP-task] vDHCPProcess: offer c0a8050bip
545 16257 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
546 16257 [iot_thread] [INFO ][MQTT][162570] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
547 16257 [iot_thread] [INFO ][MQTT][162570] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
548 16257 [iot_thread] [ERROR][NET][162570] Error -27648 while sending data.
549 16258 [iot_thread] [INFO ][MQTT][162580] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

I (163456) event: sta ip: 192.168.5.11, mask: 255.255.255.0, gw: 192.168.5.1
I (163456) WIFI: SYSTEM_EVENT_STA_GOT_IP
550 16314 [IP-task] vDHCPProcess: offer c0a8050bip
551 16358 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
552 16358 [iot_thread] [INFO ][MQTT][163580] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
553 16358 [iot_thread] [INFO ][MQTT][163580] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
554 16359 [iot_thread] [ERROR][NET][163590] Error -27648 while sending data.
555 16360 [iot_thread] [INFO ][MQTT][163600] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

556 16460 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
557 16460 [iot_thread] [INFO ][MQTT][164600] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
558 16460 [iot_thread] [INFO ][MQTT][164600] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
559 16461 [iot_thread] [ERROR][NET][164610] Error -27648 while sending data.
560 16462 [iot_thread] [INFO ][MQTT][164620] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

561 16562 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
562 16562 [iot_thread] [INFO ][MQTT][165620] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
563 16562 [iot_thread] [INFO ][MQTT][165620] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
564 16563 [iot_thread] [ERROR][NET][165630] Error -27648 while sending data.
565 16564 [iot_thread] [INFO ][MQTT][165640] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

566 16664 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
567 16664 [iot_thread] [INFO ][MQTT][166640] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
568 16664 [iot_thread] [INFO ][MQTT][166640] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
569 16665 [iot_thread] [ERROR][NET][166650] Error -27648 while sending data.
570 16666 [iot_thread] [INFO ][MQTT][166660] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

571 16766 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
572 16766 [iot_thread] [INFO ][MQTT][167660] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
573 16766 [iot_thread] [INFO ][MQTT][167660] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
574 16767 [iot_thread] [ERROR][NET][167670] Error -27648 while sending data.
575 16768 [iot_thread] [INFO ][MQTT][167680] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.
E (168426) i2cdev: Could not write to device [0x77 at 0]: -1

576 16868 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
577 16868 [iot_thread] [INFO ][MQTT][168680] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
578 16868 [iot_thread] [INFO ][MQTT][168680] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
579 16868 [iot_thread] [ERROR][NET][168680] Error -27648 while sending data.
580 16868 [iot_thread] [INFO ][MQTT][168680] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

581 16968 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
582 16968 [iot_thread] [INFO ][MQTT][169680] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
583 16968 [iot_thread] [INFO ][MQTT][169680] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
584 16969 [iot_thread] [ERROR][NET][169690] Error -27648 while sending data.
585 16970 [iot_thread] [INFO ][MQTT][169690] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

586 17070 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
587 17070 [iot_thread] [INFO ][MQTT][170700] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
588 17070 [iot_thread] [INFO ][MQTT][170700] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
589 17071 [iot_thread] [ERROR][NET][170710] Error -27648 while sending data.
590 17071 [iot_thread] [INFO ][MQTT][170710] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.
[... 8 lines cut ...]
597 17273 [iot_thread] [INFO ][MQTT][172730] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
598 17273 [iot_thread] [INFO ][MQTT][172730] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
599 17274 [iot_thread] [ERROR][NET][172740] Error -27648 while sending data.
600 17275 [iot_thread] [INFO ][MQTT][172750] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

601 17375 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
602 17375 [iot_thread] [INFO ][MQTT][173750] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
603 17375 [iot_thread] [INFO ][MQTT][173750] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
604 17376 [iot_thread] [ERROR][NET][173760] Error -27648 while sending data.
605 17377 [iot_thread] [INFO ][MQTT][173770] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.

606 17477 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
607 17477 [iot_thread] [INFO ][MQTT][174770] (MQTT connection 0x3fff5014) MQTT PUBLISH operation queued.
608 17477 [iot_thread] [INFO ][MQTT][174770] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Waiting for operation completion.
609 17478 [iot_thread] [ERROR][NET][174780] Error -27648 while sending data.
610 17479 [iot_thread] [INFO ][MQTT][174790] (MQTT connection 0x3fff5014, PUBLISH operation 0x3ffe6a3c) Wait complete with result NETWORK ERROR.
```

After this, I have to manually reboot the ESP32 and begin its process flawlessly.

Is there any way that the ESP32 can continue sending data to AWS after reconnecting to the access point?

Looking forward to hearing from you soon.

Thanks in advance.

abhidixi11, 2021-04-09:
Hello @Yoimer ,

Thanks for reaching out. 
A few clarifying questions:
- You mentioned that you are using BLE app, but you are waiting to  get IP from the router.  I'm assuming here that you are trying to connect to broker over WIFI. Is that the correct assumption ? Or the connection goes over your mobile ?
- Are you using the OTA demo without any change or have you modified it ?
- From the logs, I do not see your application attempting to connect to the broker anywhere. I do not see "Connecting to broker..." statement from the demo. Are you sure that IoTMqtt_Connect () is being called  before attempting to send PUBLISH?

Yoimer, 2021-04-09:
> Hello @Yoimer ,
> 
> Thanks for reaching out.
> A few clarifying questions:
> 
> * You mentioned that you are using BLE app, but you are waiting to  get IP from the router.  I'm assuming here that you are trying to connect to broker over WIFI. Is that the correct assumption ? Or the connection goes over your mobile ?
> * Are you using the OTA demo without any change or have you modified it ?
> * From the logs, I do not see your application attempting to connect to the broker anywhere. I do not see "Connecting to broker..." statement from the demo. Are you sure that IoTMqtt_Connect () is being called  before attempting to send PUBLISH?
> 
> Thanks

@abhidixi11 Thanks a lot for replying. I will respond each question, precisely: 

- _You mentioned that you are using BLE app, but you are waiting to get IP from the router. I'm assuming here that you are trying to connect to broker over WIFI. Is that the correct assumption ? Or the connection goes over your mobile ?_

We are using Wifi_provisioning over BLE mobile app on Android. So you're right on your assumption. The ESP32 needs a wifi router with internet access in order to connect to the broker.

- _Are you using the OTA demo without any change or have you modified it ?_

We are using the modified `ota` demo from `AWS FreeRTOS 202002`. The person who modified is [pvyawaha ](https://github.com/aws/amazon-freertos/issues/2569#issuecomment-708716699) I think he works for AWS. That demo works great!. The very same code is here on my repository [V202002.00_aws_iot_ota_update_demo_and_mqtt.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/ota/V202002.00_aws_iot_ota_update_demo_and_mqtt.c)

- _From the logs, I do not see your application attempting to connect to the broker anywhere. I do not see "Connecting to broker..." statement from the demo. Are you sure that IoTMqtt_Connect () is being called before attempting to send PUBLISH?_

I am attaching the whole log over here. This is after we provision the Wifi credentials using the ble mobile app.

Please, ignore the message:

```
E (5446) i2cdev: Could not write to device [0x77 at 0]: -1
Could not read data from bmp 180 sensor: -1
E (5466) dht: Initialization error, problem in phase 'C'
Could not read data from dht11 sensor
```

We have two sensors attached to the ESP32. When they're not physically connected, we see these messages on the console, but the code keeps running flawlessly. This is merely for debugging. 

Looking forward to seeing your comments.

Thanks in advance.

[ota_mqtt_ble_logs.txt](https://github.com/aws/amazon-freertos/files/6287528/ota_mqtt_ble_logs.txt)

abhidixi11, 2021-04-10:
Thanks for detailed explanation !
I think when you physically turn off the router and turn back on , the demo is not not calling MQTT connect again. 
After the [network is connected](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/demos/ota/V202002.00_aws_iot_ota_update_demo_and_mqtt.c#L253) , it looks like the code does not check for failed network connection and keeps publishing. The application should detect if either of following has happened :
1. Publish has failed 
2. Keep-alive has failed. ( through disconnect call back)
If yes, disconnect the MQTT connection and then reconnect. 

Let me get back to you I'll sync up with @pvyawaha. 
Thanks!

Yoimer, 2021-04-10:
> Thanks for detailed explanation !
> I think when you physically turn off the router and turn back on , the demo is not not calling MQTT connect again.
> After the [network is [...]

@abhidixi11 Sure. Thanks a lot for your kind support on this!

pvyawaha, 2021-04-16:
Hello Yoimer,

OTA V202002

Thank you for sharing details. When a connection issues happens the OTA Agent will try to publish number of times before stopping and reconnecting. The default OTA config for number of attempts is defined in [aws_ota_agent_config.h](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h) with configuration parameter `otaconfigMAX_NUM_REQUEST_MOMENTUM ` to 32. You can reduce this to lower value to even 1 to make the OTA Agent shutdown and reconnect sooner. Once the OTA Agent stops after number of publish attempts are tried it will  shutdown and reconnect as implemented in the demo. 

I will recommend to move to latest release as that supports resuming the OTA download when a reconnect happens. 

Please let me know if you have further questions.

Yoimer, 2021-04-16:
> Hello Yoimer,
>
> OTA V202002
>
> Thank you for sharing details. When a connection issues happens the OTA Agent will try to publish number of times before stopping and reconnecting. The default OTA config [...]

Dear @pvyawaha, 

Thanks a lot for your support on this, this is quite kind from you. I will try as you recommended. At the moment I can not move to the latest  OTA demo on AWS FreeRTOS ESP32, since it is not that easy to use `ota` and `coreMQTT` together. What it is making me keeps this current version (`202002`) is the fact that it is possible to have `mqtt` and `ota` running together. I've made quite a few otas from AWS (having the `.bin` file on an `S3`) and all working good.

Thanks a lot again, for your support.

Yoimer, 2021-04-19:
@pvyawaha Hello. I really hope you're doing fine.

I followed your instructions as per requested, and although ESP32 reconnects to the router (I verify this by pinging the ESP32 ip address), it does not reconnect to AWS, the console keeps showing the same error (infinitely):

```
E (222589) i2cdev: Could not write to device [0x77 at 0]: -1
Could not read data from bmp 180 sensor: -1
E (222609) dht: Initialization error, problem in phase 'C'
Could not read data from dht11 sensor
898 22238 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
899 22238 [iot_thread] [INFO ][MQTT][222380] (MQTT connection 0x3fff4940) MQTT PUBLISH operation queued.
900 22238 [iot_thread] [INFO ][MQTT][222380] (MQTT connection 0x3fff4940, PUBLISH operation 0x3ffe60f8) Waiting for operation completion.
901 22239 [iot_thread] [ERROR][NET][222390] Error -27648 while sending data.
902 22240 [iot_thread] [INFO ][MQTT][222400] (MQTT connection 0x3fff4940, PUBLISH operation 0x3ffe60f8) Wait complete with result NETWORK ERROR.
```

Here I changed the `otaconfigMAX_NUM_REQUEST_MOMENTUM` to `1U` [aws_ota_agent_config.h](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/14d6b91573690a82d58fee54521fdf10fc17fc4e/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L94) as you suggested.

I even also tried adding the `esp_wifi_connect()` function on this specific section of [iot_wifi.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202002.00/blob/14d6b91573690a82d58fee54521fdf10fc17fc4e/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c#L110), but no success so far.

Is there any way to reboot the ESP32 after, let's say 5 minutes of failing the attempts to send data to AWS after reconnecting to the router?. Since all the wifi credentials are saved on the EEPROM, the ota demo cicle would start from the beginning and the process should be able to continue as usual.

I am asking this because we're evaluating all the scenarios on the field. At the moment we're using the mobile app for `BLE WiFi Provisioning` based on the AWS SDK with Cognito in our AWS account.

So imagine for a moment that the client has to physically reboot the wifi router, or even the IT department does that from the UI of the own router. The ESP32 won't send any to data to AWS, and can't receive any OTA  automatically with this current configuration. 

If by any chance the wifi router is damaged, broken or malfunctioning, definitely  we need to send a technician to the location and ask them to activate a new router and pass the new wifi credentials using the `BLE WiFi Provisioning` mobile app.

I asked about the self-rebooting-logic after 5 minutes in the ESP32 (after reconnected to the local wifi router) because if I force to reboot it immediately, the ESP32 would be in an endless reboot loop, not giving me any chance the add new wifi credentials from the mobile app. It would try using the ones saved previously on the eeprom, but as the router is out of service, that won't simply work.

Apart of my idea, what would you suggest based on your experience?

Looking forward to hearing from you soon.

Respectfully, 

Yoimer.

pvyawaha, 2021-04-21:
Hello Yoimer,

Regarding your logs after changing the otaconfigMAX_NUM_REQUEST_MOMENTUM to 1U , can you please share complete log? If you can not share it here you can share with the support channel. It seems the network connection is not re-established before continuing OTA.

Also, the https://github.com/aws/amazon-freertos/releases/tag/202007.00 release is based on the pre-lts ( OTA and previous MQTT library not coreMQTT) and the reconnection is handled in the demo along with resuming the OTA. 

This should be handled in the application instead of forcing a reset on every disconnect. If still application logic needs a reset then you can call prvPAL_ResetDevice function from Espressif OTA Pal port which basically calls prvPAL_ResetDevice.

Yoimer, 2021-04-21:
@pvyawaha Hello. Thanks a lot for replying and assisting. Yes, for sure. I will pass you the whole log here.

Regarding the `202007.00` you are suggesting, would I be able to handle the `ota` and `mqtt` together in just one code as you created it here [aws_iot_ota_update_demo.c](https://github.com/aws/amazon-freertos/blob/mchp/otaapp/demos/ota/aws_iot_ota_update_demo.c) with version `V202002.00`?

The only thing that is keeping me using the `V202002.00` version is that it happens to be quite stable. In fact we're using the `BLE Wifi Provisioning` and 2 sensors and even now we were able to bring a GPS and it works flawlessly.

Of course, if all works OK on `202007.00` and even, we are able to add the `reconnection` and `resuming the OTA` features, that would be quite great also. That adds even more value to our current solution for our clients.

In a few hours I'm proving the whole log as per suggested.

Thanks a lot for your support. Looking forward to hearing from your comments.

Yoimer, 2021-04-21:
@pvyawaha Hello,

I am attaching the whole log as suggested. The ESP32 connects to the local wifi router after passing wifi credentials using the ble mobile app.  It begins sending data to AWS IoT Core. 

After a few minutes, I rebooted the wifi router. ESP32, reconnects to the router by itself (again, I see it on my network and I can even ping t). But it can no longer send data to AWS by itself.

For security reasons I just edited the endpoint string on the log. 

The rest is exactly the same as esp32 console prompted. Thanks in advance for your cooperation.

[aws-freertos-v202002-connect-disconnect-log.txt](https://github.com/aws/amazon-freertos/files/6354337/aws-freertos-v202002-connect-disconnect-log.txt)

Yoimer, 2021-04-22:
@pvyawaha Hello. I hope you're doing quite well there.

We made some progress regarding the reconnect issue:

1. We downloaded the `FreeRTOS V202007.00` zip file from AWS Console for the `OTA` demo and created a repository here [AFR_ESP32_FreeRTOS_V202007](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007) with the unzipped files on the `main branch`.
2. We were able to implement the `ota+mqtt` on `V202007.00` with a little modification on this `integration branch`  [aws_iot_ota_update_demo.c](https://github.com/Yoimer/AFR_ESP32_FreeRTOS_V202007/blob/integration/demos/ota/aws_iot_ota_update_demo.c)    , it is almost what you did for the `V202002` here [aws_iot_ota_update_demo.c](https://github.com/aws/amazon-freertos/blob/mchp/otaapp/demos/ota/aws_iot_ota_update_demo.c).
3. We only tested the connect and disconnect functionality on this new firmware. We observed that the ESP32 was able to reconnect to local wifi router and begin transmitting data to AWS after some `Retrying network connection` delays, which it is quite good for us, we were looking for something like this since while ago :) 
4. I am attaching the whole log for better understanding of the whole procedure.
5. Now we are going to perform some OTAS from AWS Console having the new `.bin` files on our `S3` bucket.
6. If `5` responds OK (likely it will), we are going to begin migrating the sensors and gps we successfully implemented on `V202002` to this  current `V202007.00`.

I will keep you posted on this. Thanks a lot for advancing using `V202007.00` yesterday.

Looking forward to hearing any comments or recommendation on this. 

Have a great day. 

[ota_mqtt_ble_202007_OK.txt](https://github.com/aws/amazon-freertos/files/6360177/ota_mqtt_ble_202007_OK.txt)

pvyawaha, 2021-04-22:
Hello Yoimer, 

Good to know that you migrated to later release. This release has number of important features like resume as I mentioned earlier, exponential backoff and retries and stability fixes. 

Let me know how your testing goes and if you have any further questions.

pvyawaha, 2021-05-03:
Hello ,

Closing this ticket now , please reopen for same topic or create new issue for other topics.

closed 2021-05-03 by pvyawaha

Yoimer, 2021-05-05:
> Hello ,
> 
> Closing this ticket now , please reopen for same topic or create new issue for other topics.

@pvyawaha Hello. Thanks for your support. Sure, I will re-open in case it's needed. Have a great day