aws/amazon-freertos #3096: TLS handshake failed, after updating shadow.

Raghav3107, 2021-04-13:
Hello 
I was working with the shadow demos example in the latest amazon-freertos vendor esp-idf v4.2. I want to make the flow of shadow is like, 
In the first run, the program should delete the existing shadow and update the initial shadow and from the second run, it will check if there any changes in the shadow in the cloud, If there are any changes in the cloud it should update the desired and report shadow in the cloud.
Here is my main function,

```
int RunDeviceShadowDemo(bool awsIotMqttMode,
                        const char *pIdentifier,
                        void *pNetworkServerInfo,
                        void *pNetworkCredentialInfo,
                        const void *pNetworkInterface)
{
    BaseType_t xDemoStatus = pdFAIL;
    BaseType_t xDemoRunCount = 0UL;
    BaseType_t xDeleteResponseLoopCount = 0UL;

    /* A buffer containing the update document. It has static duration to prevent
     * it from being placed on the call stack. */
    static char pcUpdateDocument[SHADOW_REPORTED_JSON_LENGTH + 1] = {0};

    static char pcdesiredUpdateDocument[SHADOW_DESIRED_JSON_LENGTH + 1] = {0};

    /* Remove compiler warnings about unused parameters. */
    (void)awsIotMqttMode;
    (void)pIdentifier;
    (void)pNetworkServerInfo;
    (void)pNetworkCredentialInfo;
    (void)pNetworkInterface;

    /* Reset the shadow delete status flags. */
    xDeleteResponseReceived = pdFALSE;
    xShadowDeleted = pdFALSE;
connectToMQTT:
    while (xDemoStatus == pdFAIL)
    {
        xDemoStatus = EstablishMqttSession(&xMqttContext,
                                           &xNetworkContext,
                                           &xBuffer,
                                           prvEventCallback);
    }

    /* First of all, try to delete any Shadow document in the cloud.
             * Try to subscribe to `delete/accepted` and `delete/rejected` topics. */
    xDemoStatus = SubscribeToTopic(&xMqttContext,
                                   SHADOW_TOPIC_STRING_DELETE_ACCEPTED(THING_NAME),
                                   SHADOW_TOPIC_LENGTH_DELETE_ACCEPTED(THING_NAME_LENGTH));

    if (xDemoStatus == pdPASS)
    {
        /* Try to subscribe to `delete/rejected` topic. */
        xDemoStatus = SubscribeToTopic(&xMqttContext,
                                       SHADOW_TOPIC_STRING_DELETE_REJECTED(THING_NAME),
                                       SHADOW_TOPIC_LENGTH_DELETE_REJECTED(THING_NAME_LENGTH));
    }

    if (xDemoStatus == pdPASS)
    {
        /* Publish to Shadow `delete` topic to attempt to delete the
                 * Shadow document if exists. */
        xDemoStatus = PublishToTopic(&xMqttContext,
                                     SHADOW_TOPIC_STRING_DELETE(THING_NAME),
                                     SHADOW_TOPIC_LENGTH_DELETE(THING_NAME_LENGTH),
                                     pcUpdateDocument,
                                     0U);
    }

    /* Wait for an incoming publish on `delete/accepted` or `delete/rejected`
             * topics, if not already received a publish. */
    if ((xDemoStatus == pdPASS) && (xDeleteResponseReceived != pdTRUE))
    {
[... 219 lines cut ...]
748 8223 [iot_thread] [shadow_demo_main.c:883] [ERROR] [ShadowDemo] Failed to connect to MQTT broker.

749 8223 [iot_thread] [mqtt_demo_helpers.c:453] [INFO] [MQTT] Create a TCP connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883.

E (83011) PKCS11: failed nvs get file size 4354 0
750 8442 [iot_thread] [core_pkcs11_mbedtls.c:3790] [ERROR] [PKCS11] Failed sign operation. mbedtls_pk_sign failed: mbed TLS error = RSA - The private key operation failed : BIGNUM - Memory al751 8442 [iot_thread] ERROR: Failure in signing callback: 6 

752 8442 [iot_thread] ERROR: TLS handshake failed trying to connect. <No-High-Level-Code> : <No-Low-Level-Code> 

753 8442 [iot_thread] TLS_Connect fail (0x7d3, a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com)

754 8442 [iot_thread] [transport_secure_sockets.c:342] [ERROR] [Transport_Secure_Sockets] Failed to establish new connection. secureSocketStatus=-1.

755 8442 [iot_thread] [mqtt_demo_helpers.c:460] [WARN] [MQTT] Connection to the broker failed. Attempting connection retry after backoff delay.

756 8455 [iot_thread] [mqtt_demo_helpers.c:402] [INFO] [MQTT] Retry attempt 2 out of maximum retry attempts 5.

757 8455 [iot_thread] [mqtt_demo_helpers.c:453] [INFO] [MQTT] Create a TCP connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883.

E (85331) PKCS11: failed nvs get file size 4354 0

```

can you suggest where I m doing wrong?
Also, can anyone explain the working flow of the shadow, I want to check my understanding is right or not.

muneebahmed10, 2021-04-13:
HI @Raghav3107,

I don't see where in your code you disconnect from MQTT, or even where MQTT returns an error. I see you have this block:
```
            if ((xUpdateAcceptedReturn != pdPASS))
            {
                LogError(("Callback function failed. [ xUpdateAcceptedReturn ]"));
                xDemoStatus = pdFAIL;
            }
```
which sets `xDemoStatus` to `pdFAIL`, and in the next iteration of your `do-while`, this block is triggered:
```
        if (xDemoStatus == pdFAIL)
        {
            /* Log error to indicate connection failure. */
            LogError(("Failed to connect to MQTT broker."));
            goto connectToMQTT;
        }
```
resulting in an incorrect error message. It then tries to establish a new MQTT session without first disconnecting the old one, which may result in the errors you are getting. You should first disconnect MQTT before creating a new connection.

Also, it seems like your issue is that `xUpdateAcceptedReturn` is erroneously set to `pdFAIL` somewhere, since I see the following line in your logs:
```
746 8063 [iot_thread] [shadow_demo_main.c:657] [INFO] [ShadowDemo] Received response from the device shadow. Previously published update with clientToken=8055 has been accepted. 
```
You should either set `xUpdateAcceptedReturn` to `pdPASS` where that log statement is, or look through your code for places where it may be set to false.

Raghav3107, 2021-04-14:
Hello @muneebahmed10 
Thanks a lot for replying and assisting me.
I made changes in the code as per your suggestion.
```

            if ((xUpdateAcceptedReturn != pdPASS))
            {
                LogError(("Callback function failed. [ xUpdateAcceptedReturn ]"));

                /* The MQTT session is always disconnected, even if there were prior failures. */
                xDemoStatus = pdFAIL;
                while (xDemoStatus == pdFAIL)
                {
                    /* The MQTT session is always disconnected, even if there were prior failures. */
                    xDemoStatus = DisconnectMqttSession(&xMqttContext, &xNetworkContext);
                    /* Log error to indicate connection failure. */
                    LogError(("%s", xDemoStatus ? "Disconnected from MQTT broker." :"Failed to disconnect from MQTT broker."));
                    vTaskDelay(1000);
                }

                xDemoStatus = pdFAIL;
                goto connectToMQTT;
            }
```
It working as per my requirement. But sometimes when I change the desired state,it not able to publish the report shadow. And giving the error. 
`438 9241 [iot_thread] [mqtt_demo_helpers.c:1058] [ERROR] [MQTT] Unable to find a free spot for outgoing PUBLISH message.`

**What does this mean?** 
Do we need to delete the previous shadow before publishing? I think we do not need that.
More log:

```
430 9116 [iot_thread] [mqtt_demo_helpers.c:568] [INFO] [MQTT] SUBSCRIBE sent for topic $aws/things/rgv_newAWStest/shadow/update/rejected to broker.

431 9120 [iot_thread] [core_mqtt.c:886] [INFO] [MQTT] Packet received. ReceivedBytes=3.

432 9120 [iot_thread] [mqtt_demo_helpers.c:694] [INFO] [MQTT] MQTT_PACKET_TYPE_SUBACK.

433 9120 [iot_thread] [mqtt_demo_helpers.c:715] [INFO] [MQTT] Subscribed to the topic $aws/things/rgv_newAWStest/shadow/update/rejected with maximum QoS 1.

434 9180 [iot_thread] [shadow_demo_main.c:943] [INFO] [ShadowDemo] Send desired power state with 0.

435 9180 [iot_thread] [mqtt_demo_helpers.c:1062] [INFO] [MQTT] the published payload:{"state":{"desired":{"powerOn":0}},"clientToken":"009180"} 
 

436 9181 [iot_thread] [mqtt_demo_helpers.c:1090] [INFO] [MQTT] PUBLISH sent for topic $aws/things/rgv_newAWStest/shadow/update to broker with packet ID 39.

437 9241 [iot_thread] [shadow_demo_main.c:973] [INFO] [ShadowDemo] Report to the state change: 0

438 9241 [iot_thread] [mqtt_demo_helpers.c:1058] [ERROR] [MQTT] Unable to find a free spot for outgoing PUBLISH message.

439 9241 [iot_thread] [shadow_demo_main.c:994] [INFO] [ShadowDemo]  Fail To Publish the Report Shadow

440 10241 [iot_thread] [shadow_demo_main.c:882] [ERROR] [ShadowDemo] Error in do-while loop

```

muneebahmed10, 2021-04-15:
Hi @Raghav3107,

If you're getting an error message saying there is no space to store the outgoing publish packet, you can increase the size of the [outgoing packet array](https://github.com/aws/amazon-freertos/blob/master/demos/common/mqtt_demo_helpers/mqtt_demo_helpers.c#L250) by increasing the value of [this macro](https://github.com/aws/amazon-freertos/blob/master/demos/common/mqtt_demo_helpers/mqtt_demo_helpers.c#L148).

However, is there a need for your application to have more than one outstanding publish at a time? If not, you can also just call MQTT_ProcessLoop() in a loop until you receive the publish acknowledgment, before proceeding with the next publish.

Raghav3107, 2021-04-15:
Thank you so much @muneebahmed10 for your time and advice. It working now.
I call MQTT_ProcessLoop() after publishing.

closed 2021-04-15 by Raghav3107