aws/amazon-freertos #3099: Avoid using  __DATE__ and __TIME__ macros.

DavidJurajdaNXP, 2021-04-14:
**Is your feature request related to a problem? Please describe.**  
Macros __DATE__ and __TIME__ prevent effective binary comparison during CI testing.
Specific location: https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/standard/tls/src/iot_tls.c#L281

**Describe the solution you would like.**  
Introduce conditional compilation of this macros or alternative functions without them.

Thank you!

n9wxu, 2021-04-14:
DATE & TIME certainly break binary comparisons.  Of course there is no promise that two otherwise identical builds will result in the same binary output.  We will look at alternatives.

yanjos-dev, 2021-06-30:
We have added an alternative solution in PR #3139 so I'm closing this feature request. Please reopen this issue if this feature does not meet your needs.

closed 2021-07-06 by yourslab