aws/amazon-freertos #3101: LoadProhibited at _IotMqtt_ScheduleOperation

samuelbles07, 2021-04-15:
Hi,

I am using amazon-freertos as library in my esp32 project, and currently developing OTA feature using OTA agent. Basically i am reproduce the `aws_iot_ota_update_demo.c` in my main code everything good (wifi connection, initialize tls) until the code arrive at `_createMqttConnection` in `iot_mqtt_api.c` as log below:

```
I (512) boot: Loaded app from partition at offset 0x20000
I (513) boot: Disabling RNG early entropy source...
I (513) cpu_start: Pro cpu up.
I (517) cpu_start: Application information:
I (521) cpu_start: Project name:     demo
I (527) cpu_start: App version:      1
I (532) cpu_start: Compile time:     Apr 15 2021 13:39:16
I (538) cpu_start: ELF file SHA256:  cdddc9be6ad56c09...
I (544) cpu_start: ESP-IDF:          v4.2-12-g3388b409c-dirty
I (550) cpu_start: Single core mode
I (555) heap_init: Initializing. RAM available for dynamic allocation:
I (561) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (568) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (574) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (580) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (586) heap_init: At 3FFC84B8 len 00017B48 (94 KiB): DRAM
I (592) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (599) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (605) heap_init: At 40095C28 len 0000A3D8 (40 KiB): IRAM
I (611) cpu_start: Pro cpu start user code
I (629) spi_flash: detected chip: generic
I (629) spi_flash: flash io: dio
I (630) cpu_start: Starting scheduler on PRO CPU.
0 1 [main] Initializing lwIP TCP stack
1 1 [main] [INFO ][MAINAPP][10] Firmware version: v0.1.0
I (658) PKCS11: Initializing NVS partition: "storage"
E (668) PKCS11: failed nvs get file size 4354 0
2 3 [main] Write certificate...
9 [main] Success Created task
I (748) wifi:wifi driver task: 3ffb962c, prio:23, stack:6656, core=0
I (748) system_api: Base MAC address is not set
I (748) system_api: read default base MAC address from EFUSE
I (758) wifi:wifi firmware version: 1865b55
I (758) wifi:wifi certification version: v7.0
I (758) wifi:config NVS flash: enabled
I (758) wifi:config nano formating: disabled
I (768) wifi:Init data frame dynamic rx buffer num: 32
I (768) wifi:Init management frame dynamic rx buffer num: 32
I (778) wifi:Init management short buffer num: 32
I (778) wifi:Init dynamic tx buffer num: 32
I (788) wifi:Init static rx buffer size: 1600
I (788) wifi:Init static rx buffer num: 10
I (798) wifi:Init dynamic rx buffer num: 32
I (798) wifi_init: rx ba win: 6
I (798) wifi_init: tcpip mbox: 32
I (808) wifi_init: udp mbox: 6
I (808) wifi_init: tcp mbox: 6
I (808) wifi_init: tcp tx win: 5744
I (818) wifi_init: tcp rx win: 5744
I (818) wifi_init: tcp mss: 1440
I (828) wifi_init: WiFi IRAM OP enabled
I (828) wifi_init: WiFi RX IRAM OP enabled
5 19 [iot_thread] [INFO ][WIFI_MANAGER][190] Credential OK, connecting...
I (928) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (928) wifi:mode : sta (3c:61:05:3d:de:7c)
I (928) WIFI: SYSTEM_EVENT_STA_START
I (1058) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1598) wifi:state: init -> auth (b0)
I (1608) wifi:state: auth -> assoc (0)
I (1618) wifi:state: assoc -> run (10)
I (1688) wifi:connected with samuelbles.com, aid = 3, channel 1, BW20, bssid = 3c:84:6a:3a:9a:7a
I (1698) wifi:security: WPA2-PSK, phy: bgn, rssi: -67
I (1698) wifi:pm start, type: 1

I (1698) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (1798) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2658) esp_netif_handlers: sta ip: 192.168.0.200, mask: 255.255.255.0, gw: 192.168.0.1
I (2658) WIFI: SYSTEM_EVENT_STA_GOT_IP
6 201 [sys_evt] [INFO ][MAINAPP][2010] _wifiEventHandler called!
7 201 [sys_evt] [INFO ][MAINAPP][2010] Connected to WiFi access point, ip address: 255.255.255.0.
31 1252 [iot_thread] [INFO ][MAINAPP][12520] starting ota manager
32 1252 [iot_thread] [INFO ][OTA_MANAGER][12520] Start initialize OTA_MANAGER connection
33 1252 [iot_thread] [INFO ][MQTT][12520] MQTT library successfully initialized.
34 1252 [iot_thread] [INFO ][OTA_MANAGER][12520] Attempt connection to mqtt broker
35 1252 [iot_thread] [INFO ][OTA_MANAGER][12520] MQTT client identifier is TEST (length 11).
E (13678) PKCS11: failed nvs get file size 4354 0
40 1601 [iot_thread] [INFO ][MQTT][16010] Establishing new MQTT connection.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008a9ba  PS      : 0x00060030  A0      : 0x80089929  A1      : 0x3ffb4c10  
0x4008a9ba: vListInsert at /Users/inventory-1/Project/build/../freertos/freertos_kernel/list.c:163 (discriminator 1)

A2      : 0x3ffc33a0  A3      : 0x3ffb8060  A4      : 0x3ffbd924  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffc33a8  
A10     : 0x00000000  A11     : 0x00000014  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffbd998  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace:0x4008a9b7:0x3ffb4c10 0x40089926:0x3ffb4c30 0x4008917f:0x3ffb4c50 0x400e17a6:0x3ffb4c90 0x400e17c1:0x3ffb4cb0 0x400dd989:0x3ffb4cd0 0x400dd9dd:0x3ffb4cf0 0x400d75dc:0x3ffb4d20 0x400d66f2:0x3ffb4d50 0x400d599e:0x3ffb4da0 0x400d5a33:0x3ffb4e30 0x400d3481:0x3ffb4e50 0x400e1699:0x3ffb4ed0 0x40087e01:0x3ffb4ef0
0x4008a9b7: vListInsert at /Users/inventory-1/Project/build/../freertos/freertos_kernel/list.c:163

0x40089926: vTaskPlaceOnEventList at /Users/inventory-1/Project/build/../freertos/freertos_kernel/tasks.c:3098 (discriminator 2)

0x4008917f: xQueueSemaphoreTake at /Users/inventory-1/Project/build/../freertos/freertos_kernel/queue.c:1612

0x400e17a6: prIotMutexTimedLock at /Users/inventory-1/Project/build/../freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:197

0x400e17c1: IotMutex_Lock at /Users/inventory-1/Project/build/../freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:207

0x400dd989: IotTaskPool_Schedule at /Users/inventory-1/Project/build/../freertos/libraries/c_sdk/standard/common/taskpool/iot_taskpool.c:708 (discriminator 2)

0x400dd9dd: IotTaskPool_ScheduleDeferred at /Users/inventory-1/Project/build/../freertos/libraries/c_sdk/standard/common/taskpool/iot_taskpool.c:748

0x400d75dc: _IotMqtt_ScheduleOperation at /Users/inventory-1/Project/build/../freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c:1278 (discriminator 2)

0x400d66f2: IotMqtt_Connect at /Users/inventory-1/Project/build/../freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:1304 (discriminator 2)

0x400d599e: _establishMqttConnection at /Users/inventory-1/Project/build/../components/ota_manager/src/ota_manager.c:127

0x400d5a33: OTA_MANAGER_init_connection at /Users/inventory-1/Project/build/../components/ota_manager/src/ota_manager.c:171

0x400d3481: MainAppTask at /Users/inventory-1/Project/build/../src/main.c:220

0x400e1699: _threadRoutineWrapper at /Users/inventory-1/Project/build/../freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:85

0x40087e01: vPortTaskWrapper at /Users/inventory-1/Project/build/../freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:154
```

I am confused, what would be the problem?

Thank you!

paulbartell, 2021-04-21:
@samuelbles07 This error usually indicates some type of memory corruption. 

Please look over the [FAQ](https://www.freertos.org/FAQHelp.html) (item 12 specifically). 

Inside the vListInsert function, there is also a block comment, which I've pasted below:
```
		If you find your application is crashing here then likely causes are:
			1) Stack overflow -
			   see http://www.freertos.org/Stacks-and-stack-overflow-checking.html
			2) Incorrect interrupt priority assignment, especially on Cortex-M3
			   parts where numerically high priority values denote low actual
			   interrupt priories, which can seem counter intuitive.  See
			   configMAX_SYSCALL_INTERRUPT_PRIORITY on http://www.freertos.org/a00110.html
			3) Calling an API function from within a critical section or when
			   the scheduler is suspended.
			4) Using a queue or semaphore before it has been initialised or
			   before the scheduler has been started (are interrupts firing
			   before vTaskStartScheduler() has been called?).
		See http://www.freertos.org/FAQHelp.html for more tips.
```

paulbartell, 2021-04-28:
@samuelbles07 Were you able the issue in this post?

closed 2021-05-17 by samuelbles07