aws/amazon-freertos #3102: problem in iot_secure_sockets.c : lwip_dns_found_callback

maa1370, 2021-04-15:
**Describe the bug**
according to https://www.nongnu.org/lwip/2_1_x/dns_8h.html#ab5a9dec5b22802f91876c53e99f427ae , for dns_found_callback, 
paddr: pointer  to an ip_addr_t containing the IP address of the hostname, or NULL if  the name could not be found (or on any other error).
but in iot_secure_sockets.c : lwip_dns_found_callback always calls *addr = *( ( uint32_t * ) ipaddr ); without checking if ipaddr is NULL (error state) or not.

**System information**
- hardware: ESP32
- note: using lwip
- f1cb0fbf1 (master) branch

**Expected behavior**
I think just checking against null should be enough. as SOCKETS_GetHostByName describes: uint32_t addr = 0; /* 0 indicates failure to caller */, addr is set to 0 by default, so i think changing to this should solve the issue:
    if(ipaddr!=NULL){
        *addr = *( ( uint32_t * ) ipaddr ); /* NOTE: IPv4 addresses only */
    }
see section below

**console output**
before change:
Creating a TLS connection to *.amazonaws.com:8883.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
lwip_dns_found_callback at C:/ESP32AWS/amazon-freertos-master/libraries/abstractions/secure_sockets/lwip/iot_secure_sockets.c:946
after change:
Creating a TLS connection to *.amazonaws.com:8883.
12 2280 [iot_thread] Unable to resolve (*.amazonaws.com) within (20l) seconds
13 2280 [iot_thread] [transport_secure_sockets.c:331] [ERROR] [Transport_Secure_Sockets] Failed to connect to server: DNS resolution failed: Server=*.amazonaws.com

**To reproduce**
connect to wifi with no internet access (tested by windows created hotspot wifi, disconnected from wifi after turning hotspot on, so hotspot wifi had no internet access)

AniruddhaKanhere, 2021-04-15:
Hello @maa1370, thank you for bringing this issue to our attention. We will get back to you shortly after looking at it in more detail.

AniruddhaKanhere, 2021-04-15:
Hello @maa1370, please see PR 3104.

maa1370, 2021-04-16:
> Hello @maa1370, please see PR 3104.

Thanks. Feel free to close the issue when done.

AniruddhaKanhere, 2021-04-16:
The PR was merged! I shall be closing this PR.

Thank you again @maa1370 for bringing this to our attention! Please open a new issue (or this one) if you want to report something. If you have questions, you may also use (FreeRTOS-Forums)[https://forums.freertos.org] where our very helpful community is ready to answer any questions you may have! :)

Cheers!

closed 2021-04-16 by AniruddhaKanhere