aws/amazon-freertos #3105: TypeError: cannot use a string pattern on a bytes-like object

haziqharuka, 2021-04-19:
hello, i'm try to flash to the esp32 but got some error,can someone help me because im beginner to this. 

these are the error that i got: 

Traceback (most recent call last):
  File "/home/haziqharuka/esp1/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/idf_monitor.py", line 743, in <module>
    main()
  File "/home/haziqharuka/esp1/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/idf_monitor.py", line 668, in main
    monitor.main_loop()
  File "/home/haziqharuka/esp1/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/idf_monitor.py", line 337, in main_loop
    self.handle_serial_input(data)
  File "/home/haziqharuka/esp1/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/idf_monitor.py", line 391, in handle_serial_input
    if self._output_enabled and (self._force_line_print or self._line_matcher.match(line)):
  File "/home/haziqharuka/esp1/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/idf_monitor.py", line 256, in match
    m = self._re.search(line)
TypeError: cannot use a string pattern on a bytes-like object
make: *** [/home/haziqharuka/esp1/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1

paulbartell, 2021-04-28:
@haziqharuka Were you able to resolve your build issue?

pvyawaha, 2021-05-17:
Closing this issue now, please reopen for same topic or create a new one for different topic.

closed 2021-05-17 by pvyawaha