aws/amazon-freertos #3107: deadlock in _mtuChangedCb when BLE MTU value is changed

bgklika, 2021-04-21:
**Describe the bug**

Definition of the problem
BLE connection can’t be established with iPhone 6. 
BLE connection stops responding after MTU change request. 

Technical details
Example of debug log after MTU change request:
GATT procedure initiated: notify; att_handle=71
34 43169 [ble] [INFO ][BLE_TXRX][43169] MTU changed, changing transmit length for data transfer service from 509 to 18

Ble task is blocked on xQueueSemaphoreTake call caused by recursive _BTInterface.threadSafetyMutex locking in functions _getServiceListElemFromHandle and _mtuChangedCb which are parts of Amazon FreeRTOS c_sdk (c_sdk/standart/ble/src/iot_ble_gatt.c).

Ble task callstack after MTU change request:
Thread #16 1073454412 (Name: ble) (Suspended : Container)	
	xQueueSemaphoreTake() at queue.c:1 614 0x4008f64c	
	prIotMutexTimedLock() at iot_threads_freertos.c:197 0x4013993a	
	IotMutex_Lock() at iot_threads_freertos.c:207 0x40139954	
	_getServiceListElemFromHandle() at iot_ble_gatt.c:171 0x401312e3	
	_getCallbackFromHandle() at iot_ble_gatt.c:202 0x40131318	
	_indicationSentCb() at iot_ble_gatt.c:605 0x4013135e	
	prvGAPeventHandler() at iot_ble_hal_common_gap.c:376 0x40134abe	
	ble_gap_call_event_cb() at ble_gap.c:607 0x4017eb90	
	ble_gap_call_conn_event_cb() at ble_gap.c:633 0x4017ed1a	
	ble_gap_notify_tx_event() at ble_gap.c:4 696 0x40180249	
	ble_gattc_notify_custom() at ble_gattc.c:4 190 0x4017d314	
	prvBTSendIndication() at iot_ble_hal_gatt_server.c:689 0x40135459	
	IotBle_SendIndication() at iot_ble_gatt.c:853 0x4013197d	
	_MTUChangedCallback() at iot_ble_device_information.c:447 0x40131a5a	
	_mtuChangedCb() at iot_ble_gatt.c:570 0x401312c7	
	prvGAPeventHandler() at iot_ble_hal_common_gap.c:444 0x40134b6e	
	ble_gap_call_event_cb() at ble_gap.c:607 0x4017eb90	
	ble_gap_call_conn_event_cb() at ble_gap.c:633 0x4017ed1a	
	ble_gap_mtu_event() at ble_gap.c:4 747 0x40180300	
	ble_att_svr_rx_mtu() at ble_att_svr.c:764 0x40186cac	
	ble_att_rx() at ble_att.c:503 0x4017bdcb	
	ble_hs_hci_evt_acl_process() at ble_hs_hci_evt.c:883 0x4017baf5	
	ble_hs_process_rx_data_queue() at ble_hs.c:238 0x4017b08c	
	ble_hs_event_rx_data() at ble_hs.c:532 0x4017b09e	
	ble_npl_event_run() at nimble_npl_os.h:121 0x40184a85	
	nimble_port_run() at nimble_port.c:82 0x40184a85	
	ble_host_task() at iot_ble_hal_common_gap.c:552 0x4013455e	

Solutions
a) Remove _BTInterface.threadSafetyMutex lock in _mtuChangedCb. That approach is checked and works.

b) Make _BTInterface.threadSafetyMutex recursive. That approach is not checked.

**System information**
- Which hardware board or part numbers?
 ESP32 DevKitC V4
- IDE used 
ESP-IDF 3.3

- Operating System [Windows|Linux|MacOS]
Windows 10
- Version of FreeRTOS (run ``git describe --tags`` to find it)
202012.00
ESP-IDF 3.3
In our application we use NimBLE, but looks like it will be reproducible on Bluedroid (This is not tested).

- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
Custom project

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 
Bug is inside amazon-freertos/libraries/c_sdk/standard/ble/src/iot_ble_gatt.c

**Expected behavior**
A clear and concise description of what you expected to happen.

We are expecting MTU is changed and ble task continue execution.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 
Not applicable, see log and call stack above.

**To reproduce**
Steps to reproduce the behavior:

1. Build 202012.00 with BLE Wi-Fi credentials provisioning
2. Using amazon demo application on IPhone and try to set Wi-Fi credentials

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:
No special code is required.

bgklika, 2021-04-21:
Patch file uploaded here https://gist.github.com/bgklika/8a453e877a51893970707e91d94d5f9a

ravibhagavandas, 2021-04-22:
@bgklika 

Thanks for reporting this. We are able to reproduce this issue when the MTU of the phone is less than the preferred MTU on the MCU and an MTU negotiation occurs.
We will fix this issue soon.

ravibhagavandas, 2021-04-30:
The fix for this issue is merged to master as part of PR #3121. I am closing this issue. Please feel free to open a new issue if you have further questions.

closed 2021-04-30 by ravibhagavandas