aws/amazon-freertos #3109: OTA Agent request next block twice when duplicate of block has been received

bgklika, 2021-04-21:
**Describe the bug**
OTA Agent execute  job and receive blocks of the OTA update.
Due to not stable network connection OTA agent can receive same block twice.
It is expected satiation and duplicated block is detected by bitmap and will not be written to the file/flash.
But even when block is duplication next block requested again.
It results in when once due to network issues block is duplicated  OTA Agent will request all remaining block twice and all remaining block will be received twice.

**System information**
- Which hardware board or part numbers?
ESP32_DevKitC_V4
- IDE used
-ESP-IDF 3.3

- Operating System [Windows|Linux|MacOS]
Windows10
- Version of FreeRTOS (run ``git describe --tags`` to find it)
202012.00

- Project
Custom Application

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 
Problem is inside prvProcessDataHandler() which is not make differences between eIngest_Result_Duplicate_Continue and eIngest_Result_Accepted_Continue  and always enqueue event eOTA_AgentEvent_RequestFileBlock to request next block twice.

**Expected behavior**
Even when block received twice (duplicated) we does not request next block twice.

**Screenshots or console output**
Output from the log:
3568 135458 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3569 135463 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3570 135463 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3571 135646 [OTA Agent Task] [prvIngestDataBlock] Received file block 469, size 4096
3572 135656 [OTA Agent Task] [prvIngestDataBlock] Remaining: 112
3573 135657 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3574 135657 [OTA Agent Task] [prvIngestDataBlock] Received file block 469, size 4096
3575 135657 [OTA Agent Task] [prvIngestDataBlock] block 469 is a DUPLICATE. 112 blocks remaining.
3576 135657 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3577 135662 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3578 135663 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3579 135665 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3580 135666 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3581 135852 [OTA Agent Task] [prvIngestDataBlock] Received file block 470, size 4096
3582 135862 [OTA Agent Task] [prvIngestDataBlock] Remaining: 111
3583 135862 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3584 135867 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3585 135868 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3586 135951 [OTA Agent Task] [prvIngestDataBlock] Received file block 470, size 4096
3587 135951 [OTA Agent Task] [prvIngestDataBlock] block 470 is a DUPLICATE. 111 blocks remaining.
3588 135951 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3589 135955 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3590 135955 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3591 136060 [OTA Agent Task] [prvIngestDataBlock] Received file block 471, size 4096
3592 136070 [OTA Agent Task] [prvIngestDataBlock] Remaining: 110
3593 136070 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3594 136074 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3595 136074 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3596 136157 [OTA Agent Task] [prvIngestDataBlock] Received file block 471, size 4096
3597 136158 [OTA Agent Task] [prvIngestDataBlock] block 471 is a DUPLICATE. 110 blocks remaining.
3598 136158 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3599 136161 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3600 136162 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3601 136261 [OTA Agent Task] [prvIngestDataBlock] Received file block 472, size 4096
3602 136271 [OTA Agent Task] [prvIngestDataBlock] Remaining: 109
3603 136272 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3604 136275 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3605 136275 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3606 136365 [OTA Agent Task] [prvIngestDataBlock] Received file block 472, size 4096
3607 136365 [OTA Agent Task] [prvIngestDataBlock] block 472 is a DUPLICATE. 109 blocks remaining.
3608 136366 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
[... 8 lines cut ...]
3617 136568 [OTA Agent Task] [prvIngestDataBlock] block 473 is a DUPLICATE. 108 blocks remaining.
3618 136568 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3619 136571 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3620 136572 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3621 136668 [OTA Agent Task] [prvIngestDataBlock] Received file block 474, size 4096
3622 136679 [OTA Agent Task] [prvIngestDataBlock] Remaining: 107
3623 136679 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3624 136683 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3625 136683 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3626 136776 [OTA Agent Task] [prvIngestDataBlock] Received file block 474, size 4096
3627 136777 [OTA Agent Task] [prvIngestDataBlock] block 474 is a DUPLICATE. 107 blocks remaining.
3628 136777 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3629 136780 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3630 136780 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]

**To reproduce**
Steps to reproduce the behavior:

1. Start OTA Update job
2. Waiting until device receive job description and starting receiving blocks
3. In a middle of receiving block disconnect internet cable (for example) from the router for a while (2-5 seconds is typically enough)
4. Look to the logs of device and find when duplicated block is receives once it becomes persistent even internet connection is becomes stable

**Code to reproduce the bug**
No need special code

pvyawaha, 2021-04-23:
Hello, 

The eIngest_Result_Accepted_Continue is return code when the block is received and successfully written to flash. The eIngest_Result_Duplicate_Continue return code is when the current block received is already received and written to flash so it is ignored. 

Can you please share the OTA configuration you are using ? What is otaconfigFILE_REQUEST_WAIT_MS set to?

bgklika, 2021-04-23:
Hello @pvyawaha 
I know the difference between eIngest_Result_Accepted_Continue  and eIngest_Result_Duplicate_Continue  as returned by prvIngestDataBlock. Problem in implementation of prvProcessDataHandler() which has no difference in processing of eIngest_Result_Accepted_Continue  and eIngest_Result_Duplicate_Continue.

Request for next block  is sent even on eIngest_Result_Duplicate_Continue.
That means when OTA Agent accidentally and ONCE receive duplicated block  it starts request all remaining blocks twice which Cloud send it twice, and all remaining blocks also will have duplicates.
For testing it I break internet connection for a short time. 
In other test I inject duplicate once for exact block by double calling of 'prvSendCallbackEvent( pvCallbackContext, pxPublishData, eOTA_AgentEvent_ReceivedFileBlock );'

I think no need to request next block on eIngest_Result_Duplicate_Continue. It will be requested anyway by timer with otaconfigFILE_REQUEST_WAIT_MS  interval

In our case otaconfigFILE_REQUEST_WAIT_MS  has value in 10 seconds by default.

lightblu, 2021-04-23:
I have no real log and not full certainty, but we observed similar behavior as described here I believe, where we wondered about receiving much too many duplicated blocks (especially in relation with 1 or multiple connection drops), so a careful +1.

pvyawaha, 2021-04-29:
Hello,

The duplicate blocks are received when the request is processed by the service but response is not yet received by the device and timer expires sending another request for same blocks. The response for the first request is now received and device marks those blocks as received. After that the response for the other request is received and those blocks will be considered as duplicate. This might happen if the requests are sent faster than the response and can happen sometime based on the network condition. The otaconfigFILE_REQUEST_WAIT_MS is the parameter that controls how fast these blocks are requested. Also this timeout is only used when the requested number of blocks are not received. So for example if 8 blocks are requested and they are received before the timer expires the next set is immediately requested but if less than 8 are received then the next set will be requested at next timeout.

pvyawaha, 2021-05-17:
Hello,

Did the otaconfigFILE_REQUEST_WAIT_MS  configuration helped in your issue?

bgklika, 2021-05-19:
> Hello,
> 
> Did the otaconfigFILE_REQUEST_WAIT_MS configuration helped in your issue?

Hello I do not tried because the problem does not related to value of otaconfigFILE_REQUEST_WAIT_MS setting.

It is related to implementation of prvProcessDataHandler() which send second request for next block even when prvIngestDataBlock() returns eIngest_Result_Duplicate_Continue. That makes sequence of duplication infinite.

pvyawaha, 2021-05-19:
Hello,

> It is related to implementation of prvProcessDataHandler() which send second request for next block even when prvIngestDataBlock() returns eIngest_Result_Duplicate_Continue. That makes sequence of duplication infinite.

The prvProcessDataHandler does not send the second request for next block when prvIngestDataBlock returns eIngest_Result_Duplicate_Continue. It means we have received a block that is already processed and written to flash. So the request was already sent. 

The duplicate error happens when a response to previous request was in flight so not yet written to flash/marked as received and a new request is sent. The speed at which the blocks are requested is controlled by otaconfigFILE_REQUEST_WAIT_MS. Also, this timeout is only used when the number of blocks requested are not received and when network is better and all blocks are received the next blocks are immediately requested without slowing down the process. 

Can you please try setting otaconfigFILE_REQUEST_WAIT_MS to much higher value ? I think that will solve your issue.

bgklika, 2021-05-19:
Hello,
Once occasionally duplicate occurs - duplicates are never finished until last block of OTA update.

Please check the implementation of prvProcessDataHandler(). Even when prvIngestDataBlock() returns eIngest_Result_Duplicate_Continue  prvProcessDataHandler() will execute lines:
`
                xEventMsg.xEventId = eOTA_AgentEvent_RequestFileBlock;

                if( !OTA_SignalEvent( &xEventMsg ) )
                {
                    OTA_LOG_L2( "[%s] Failed to signal OTA agent to close file.", OTA_METHOD_NAME );
                }
`
which cause request next block second time, and cloud responds with duplicate block.

ActoryOu, 2022-06-13:
Fixed in #3495 .

closed 2022-06-13 by ActoryOu