aws/amazon-freertos #3110: Crash in OTA agent in function prvPublishStatusMessage()

bgklika, 2021-04-21:
**Describe the bug**
When we doing receiving of OTA job block
if is possible to use NULL pointer from pxAgentCtx->pcOTA_Singleton_ActiveJobName in prvPublishStatusMessage()

When couple of blocks are received (even when duplicate) but writing to file/flash failed in function prvProcessDataHandler() code block 
`        /* Free any remaining string memory holding the job name since this job is done. */
        if( xOTA_Agent.pcOTA_Singleton_ActiveJobName != NULL )
        {
            vPortFree( xOTA_Agent.pcOTA_Singleton_ActiveJobName );
            xOTA_Agent.pcOTA_Singleton_ActiveJobName = NULL;
        }
`
is called. 
Same time next block can be already received and next task cycle will be processed by same prvProcessDataHandler() and code 
`xErr = xOTA_ControlInterface.prvUpdateJobStatus( &xOTA_Agent, eJobStatus_FailedWithVal, ( int32_t ) xCloseResult, ( int32_t ) xResult );`
will called.
Unfortunately prvPublishStatusMessage() called from  prvUpdateJobStatus_Mqtt() do not check pxAgentCtx->pcOTA_Singleton_ActiveJobName (and pxAgentCtx->pcThingName too )  before execute code block:
`    /* Try to build the dynamic job status topic . */
    ulTopicLen = ( uint32_t ) snprintf( pcTopicBuffer, /*lint -e586 Intentionally using snprintf. */
                                        sizeof( pcTopicBuffer ),
                                        pcOTA_JobStatus_TopicTemplate,
                                        pxAgentCtx->pcThingName,
                                        pxAgentCtx->pcOTA_Singleton_ActiveJobName );

`
In result exception is occured.

**System information**
- Which hardware board or part numbers?
ESP32_DevKitC_V4
- IDE used
ESP-IDF 3.3
- Operating System [Windows|Linux|MacOS]
Window10

- Version of FreeRTOS (run ``git describe --tags`` to find it)
202012.00

- Project
Custom Application

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 
No need special code

**Expected behavior**
Even OTA image is bad, or network connection is not stable application should not crash.

**Screenshots or console output**
Log output:
`3671 137397 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3672 137397 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3673 137494 [OTA Agent Task] [prvIngestDataBlock] Received file block 478, size 4096
3674 137505 [OTA Agent Task] [prvIngestDataBlock] Remaining: 103
3675 137505 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3676 137508 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3677 137508 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3678 137596 [OTA Agent Task] [prvIngestDataBlock] Received file block 478, size 4096
3679 137597 [OTA Agent Task] [prvIngestDataBlock] block 478 is a DUPLICATE. 103 blocks remaining.
3680 137597 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3681 137600 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3682 137600 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3683 137694 [OTA Agent Task] [prvIngestDataBlock] Received file block 479, size 4096
3684 137704 [OTA Agent Task] [prvIngestDataBlock] Remaining: 102
3685 137705 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3686 137708 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
3687 137708 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
3688 137795 [OTA Agent Task] [prvIngestDataBlock] Received file block 479, size 4096
3689 137795 [OTA Agent Task] [prvIngestDataBlock] block 479 is a DUPLICATE. 102 blocks remaining.
3690 137796 [OTA Agent Task] [prvExecuteHandler] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
3691 137799 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ac:67:b2:6e:81:4c/streams/AFR_OTA-61976ca5-1a73-4568-aebb-85536f0be3f1/get/cbor
[... 46 lines cut ...]

0x4011abec: prvPublishStatusMessage at H:/Work/Klika-tech/Cubic/sources/crrr-cbcq/esp32_firmware/src/amazon-freertos/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c:395

0x4011aed3: prvUpdateJobStatus_Mqtt at H:/Work/Klika-tech/Cubic/sources/crrr-cbcq/esp32_firmware/src/amazon-freertos/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c:768

0x40118f85: prvProcessDataHandler at H:/Work/Klika-tech/Cubic/sources/crrr-cbcq/esp32_firmware/src/amazon-freertos/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c:3331

0x40117b11: prvExecuteHandler at H:/Work/Klika-tech/Cubic/sources/crrr-cbcq/esp32_firmware/src/amazon-freertos/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c:3331

0x40118d76: prvOTAAgentTask at H:/Work/Klika-tech/Cubic/sources/crrr-cbcq/esp32_firmware/src/amazon-freertos/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c:3331

Rebooting...
`

**To reproduce**
Steps to reproduce the behavior:

1. Create OTA image which is bigger than flash bank side
2.  Start the OTA job for the device
3. Waiting until block starts to downloaded
4. Interrupt network connection for a while for example by unplug/plug internet cable from Wi-Fi router. 
5. Looking in device log for block duplicates and after that (when block tried to be written outside flash bank) - crash.

**Code to FIX the bug**
Please find the patch here https://gist.github.com/bgklika/ceeaa6fa35c1bdfa31a9e67e220b9c21

pvyawaha, 2021-04-23:
Thank you for reporting this. We are working on reproducing and fixing it.

dFohlen, 2021-05-27:
Hello @pvyawaha ,

I don't know if it's the same problem but the OTA agent keeps crashing after starting up. We are using AWS FreeRTOS (v202012.00) with an ESP32 board and the [aws_iot_ota_agent.h](https://github.com/aws/amazon-freertos/blob/202012.00/libraries/freertos_plus/aws/ota/include/aws_iot_ota_agent.h).

```
I (6636) MAIN: <main> start ota update task
I (6636) MQTT: <constructor> creating mqtt agent
I (6636) MQTT: <connect> connecting to broker
6 907 [iot_thread] [INFO ][MQTT][9070] Establishing new MQTT connection.
7 907 [iot_thread] [INFO ][MQTT][9070] (MQTT connection 0x3ffdb920, CONNECT operation 0x3ffdc6a8) Waiting for operation completion.
8 922 [NetRecv] [INFO] [MQTT] [core_mqtt_serializer.c:970]
9 922 [NetRecv] CONNACK session present bit not set.10 922 [NetRecv]
11 922 [NetRecv] [INFO] [MQTT] [core_mqtt_serializer.c:912] 
12 922 [NetRecv] Connection accepted.13 922 [NetRecv]
14 923 [iot_thread] [INFO ][MQTT][9230] (MQTT connection 0x3ffdb920, CONNECT operation 0x3ffdc6a8) Wait complete with result SUCCESS.
15 925 [iot_thread] [INFO ][MQTT][9250] New MQTT connection 0x3ffd941c established.
I (9326) MQTT: <connect> connected!
I (9336) OTA_UPDATE: <run ota agent> init ota agent
16 928 [iot_thread] [OTA_AgentInit_internal] OTA Task is Ready.
I (9356) ota_pal: prvPAL_GetPlatformImageState
I (9356) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (9356) esp_ota_ops: [0] aflags/seq:0x2/0x1, pflags/seq:0xffffffff/0x0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008b453  PS      : 0x00060f33  A0      : 0x8008a372  A1      : 0x3ffea260
0x4008b453: uxListRemove at /Volumes/Workspace/amazon-freertos/freertos_kernel/list.c:187

A2      : 0x0000001d  A3      : 0x3ffea510  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x3ffea3e0  A8      : 0x8008899f  A9      : 0x3ffea240
A10     : 0x00000000  A11     : 0x3ffd941c  A12     : 0x3ffea510  A13     : 0x00000001
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000002d  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff6

ELF file SHA256: 7ead7b3026c3d43b

Backtrace: 0x4008b453:0x3ffea260 0x4008a36f:0x3ffea280 0x40089a7c:0x3ffea2a0 0x40184793:0x3ffea2e0 0x401847ad:0x3ffea300 0x40104787:0x3ffea320 0x401062bd:0x3ffea350 0x40104d8a:0x3ffea370 0x40104f99:0x3ffea3b0 0x4010554c:0x3ffea3e0 0x40103641:0x3ffea410 0x401039c9:0x3ffea550 0x40101e68:0x3ffea6e0 0x401007ce:0x3ffea710 0x40101aa2:0x3ffea730
0x4008b453: uxListRemove at /Volumes/Workspace/amazon-freertos/freertos_kernel/list.c:187

0x4008a36f: xTaskRemoveFromEventList at /Volumes/Workspace/amazon-freertos/freertos_kernel/tasks.c:5248 (discriminator 4)

0x40089a7c: xQueueSemaphoreTake at /Volumes/Workspace/amazon-freertos/freertos_kernel/queue.c:2429

0x40184793: prIotMutexTimedLock at /Volumes/Workspace/amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x401847ad: IotMutex_Lock at /Volumes/Workspace/amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c:318

0x40104787: _IotMqtt_IncrementConnectionReferences at /Volumes/Workspace/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:758

0x401062bd: _IotMqtt_CreateOperation at /Volumes/Workspace/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_operation.c:355

0x40104d8a: _subscriptionCommon at /Volumes/Workspace/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:635

0x40104f99: IotMqtt_Subscribe at /Volumes/Workspace/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:1491

0x4010554c: IotMqtt_TimedSubscribe at /Volumes/Workspace/amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c:1515

0x40103641: prvSubscribeToJobNotificationTopics at /Volumes/Workspace/amazon-freertos/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c:173

0x401039c9: prvRequestJob_Mqtt at /Volumes/Workspace/amazon-freertos/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c:659

0x40101e68: prvRequestJobHandler at /Volumes/Workspace/amazon-freertos/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c:3179

0x401007ce: prvExecuteHandler at /Volumes/Workspace/amazon-freertos/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c:3179

0x40101aa2: prvOTAAgentTask at /Volumes/Workspace/amazon-freertos/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c:3179

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x32 (SPI_FAST_FLASH_BOOT)
[... 30 lines cut ...]
0x40080000: _WindowOverflow4 at /Volumes/Workspace/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (253) esp_image: segment 3: paddr=0x0006ef9c vaddr=0x40080400 size=0x01074 (  4212) load
I (263) esp_image: segment 4: paddr=0x00070018 vaddr=0x400d0018 size=0xd93e8 (889832) map
0x400d0018: _flash_cache_start at ??:?

I (582) esp_image: segment 5: paddr=0x00149408 vaddr=0x40081474 size=0x147dc ( 83932) load
0x40081474: _xt_user_exc at /Volumes/Workspace/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:670
```
**System information**

* Which hardware board or part numbers?  
  ESP32-WROOM-32D

* IDE used   
  ESP-IDF 3.3

* Operating System [Windows|Linux|MacOS]  
  MacOS

* Version of FreeRTOS (run git describe --tags to find it)  
  202012.00

* Project  
  Custom Application

pvyawaha, 2021-06-14:
Hello @dFohlen,

Are you running the OTA demo as it is or integrated with your application ? Are you also sharing the connection for other MQTT operations?

pvyawaha, 2021-06-14:
Hello @bgklika ,

We are working on integrating the LTS OTA library in this repo and will be verifying the issue.

dFohlen, 2021-06-14:
Hi @pvyawaha,

Thanks for the answer

> Are you running the OTA demo as it is or integrated with your application ?

Integrated, but it's mainly copied from the test sources:

```
static OTA_ConnectionContext_t xOTAConnContext = { nullptr, nullptr, nullptr };

static auto CompleteCallback(OTA_JobEvent_t aEvent) -> void
{
  ESP_LOGI(TAG, "<complete callback> event: %i", aEvent);
}

auto RunOtaAgent(std::string const& aThingName) -> bool
{
  OTA_State_t eOtaStatus = eOTA_AgentState_Init;
  TickType_t xTicksToWait = pdMS_TO_TICKS(otatestAGENT_INIT_WAIT);
  xOTAConnContext.pvControlClient = MQTT::Get()->mMqttClientHandle;

  ESP_LOGI(TAG, "<run ota agent> init agent");
  if (eOtaStatus = OTA_AgentInit(&xOTAConnContext, (const uint8_t*)aThingName.c_str(), CompleteCallback, xTicksToWait); eOtaStatus != eOTA_AgentState_Ready)
  {
    ESP_LOGE(TAG, "<run ota agent> failed to init ota agent");
    return false;
  }

  // Wait for OTA agent to transit to eOTA_AgentState_WaitingForJob state.
  while ((xTicksToWait > 0U) && (eOtaStatus != eOTA_AgentState_WaitingForJob))
  {
    ESP_LOGI(TAG, "<run ota agent> agent state: %i", eOtaStatus);
    eOtaStatus = OTA_GetAgentState();
    vTaskDelay(1);
    xTicksToWait--;
  }

  return true;
}
```

> Are you also sharing the connection for other MQTT operations?

Not currently (in this example), but we want to use the OTA agent for that.

pvyawaha, 2021-12-16:
Hello , 

Do you still see the issues with latest version of the library or can this ticket closed?

bgklika, 2021-12-16:
I have fixed that bug locally via patch before reporting the issue. If bug actually resolve in main repo you could close the issue.

dFohlen, 2021-12-16:
I've fixed it too. After updating to ESP-IDF 4.2 and using the MQTT + OTA agent, the problem no longer occurred.

pvyawaha, 2021-12-16:
Thank you , closing this issue.

closed 2021-12-16 by pvyawaha