aws/amazon-freertos #3111: Strange IotMqttConnection_t implementation

bgklika, 2021-04-21:
**Describe the bug**
A clear and concise description of what the bug is.

Technical details:

functions like IotMqtt_Connect(), IotMqtt_Disconnect() and so one operate with object-like structure  IotMqttConnection_t. That is very strange object which looks like self-destroyed object. Due to memory limits and implementation of Amazon IoT Core which allow only one MQTT connection per-device we have to share single IotMqttConnection_t for at least 2 clients - Sparkplug and OTA Agent. Unfortunately  these clients can’t be run in single RTOS task due to OTA agent is implemented in separate task.

The first argument of function IotMqttError_t IotMqtt_Connect() has member named disconnectCallback.function which is pointer to function. That callback will be called from 3 different cases.

IotMqtt_Disconnect() has been explicitly called. In that case callback function is called outside of internal IotMqttConnection_t lock. After callback has been called IotMqttConnection_t will be dereferenced and if reference counter becomes 0 object will be deleted. Reference counter is internally maintained by implementation via internal API.

When Internal Amazon iot_core task tried to send keep-alive request and it failed. In that case callback called under locking of internal IotMqttConnection_t lock. Number of references to the object will be decremented once and object potentially deleted.

When invalid MQTT payload which can’t be parsed has been received. In that case callback is called outside locking and object will NOT be dereferences.

 

Actually behavior of the code after calling of the callback function is undefined. In some cases we should drop all references to the IotMqttConnection_t object due it will be deleted. In other cases we have to call IotMqtt_Disconnect or some memory will be leaked.

And we can’t call any destructor functions which call other function like IotMqtt_UnsubscribeSync() for example due to callback can be called with lock acquired.

Another possible problem - simultaneous calls of IotMqtt_Disconnect() in one task and callback from other for example from iot_core task due to keep-alive failed.

 
Due to all that usage of IotMqttConnection_t becomes very tricky.

**System information**
- Which hardware board or part numbers?
ESP32_DevkitC_V4
- IDE used
ESP-IDF 3.3

- Operating System [Windows|Linux|MacOS]
Windows10
- Version of FreeRTOS (run ``git describe --tags`` to find it)
202012.00

- Project
Custom Application

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 
Use IotMqttConnection_t for 2 clients at the same time.

**Expected behavior**
Usage of IotMqttConnection_t  must be clear.

**Screenshots or console output**

paulbartell, 2021-04-22:
@bgklika : Please check out the documentation on the coreMQTT agent project and it's git repository, which is meant to address this type of problem.

https://www.freertos.org/mqtt/mqtt-agent-demo.html
https://github.com/FreeRTOS/coreMQTT-Agent

bgklika, 2021-04-23:
Thank you @paulbartell 

I check it, and understand that method.
I think root problems described in that issue not only about thread safely. 

Seems coreMQTT Agent v1.0.0 uses API of coreMQTT library like MQTT_Publish() but same time most interested components of Amazon FreeRTOS 202012.00 like aws_iot_ota_agent.c  for example directly uses other library and API - "FreeRTOS MQTT V2.3.1"  from iot_mqtt_api.c 

Same time iot_mqtt_agent.c  there is, looks like components should be switched from iot_mqtt_api.c to iot_mqtt_agent.c

aggarw13, 2021-04-27:
Hi @bgklika, to address the problems you have highlighted about the "MQTT V2.3.1" stack, we will be deprecating the library and providing examples of sharing an MQTT connection across multiple tasks using the [coreMQTT-Agent](https://github.com/FreeRTOS/coreMQTT-Agent) library in a future release.
For now, you can refer to the following places for example on using the coreMQTT Agent library in a multi-task use case:
* [**FreeRTOS/FreeRTOS**](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo/coreMQTT_Windows_Simulator/MQTT_Multitask) - This represents a simple demo of multiple Publish-Subscribe tasks sharing the same MQTT connection
* [**FreeRTOS/coreMQTT-Agent-Demos**](https://github.com/FreeRTOS/coreMQTT-Agent-Demos) - This repository contains a more complex demo example of sharing the same MQTT connection across multiple tasks to communicate with AWS services like OTA, AWS IoT Device Defender and AWS IoT Device Shadow

Hope the reference examples can help you migrate to using the coreMQTT-Agent library 🙂

lightblu, 2021-05-03:
@aggarw13 , is there any timeline estimate on the release of the coreMQTT-Agent library, or would you already deem it stable enough to start migrating to it?

muneebahmed10, 2021-05-03:
Hi @lightblu,

We [released version 1.0.0](https://github.com/FreeRTOS/coreMQTT-Agent/releases) of the coreMQTT Agent library 4 days ago, on 4/29. You can find demo projects for it [here](https://github.com/FreeRTOS/coreMQTT-Agent-Demos) and [here](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo/coreMQTT_Windows_Simulator/MQTT_Multitask), with a demo for this repository to be added soon.

samuelbles07, 2021-05-17:
hi @muneebahmed10 , 

I see that coreMQTT Agent demo already in this repository, but OTA agent library still using mqtt v2 library. Is OTA agent library update in the timeline? Is there any estimate date?

Thank you!

muneebahmed10, 2021-05-17:
@samuelbles07 

The OTA library in this repository will be updated in a future release. You can track the current progress of the new OTA library integration in the [ota_v3_integration](https://github.com/aws/amazon-freertos/tree/feature/ota_v3_integration) branch.

aggarw13, 2021-05-22:
I will be closing the issue as the original author's query has been answered. 
@lightblu and @samuelbles07, if you have any further questions, please feel free to create new issues.

closed 2021-05-22 by aggarw13