aws/amazon-freertos #3112: BLE advertisement is turned off after 15 min with NimBLE

bgklika, 2021-04-21:
**Describe the bug**
Definition of the problem
By default NimBLE each 900 seconds (15 minutes) generate new RPA address and in addition stop BLE advertising. That means after 15 minutes device disappears from BLE scan list.

**System information**
- Which hardware board or part numbers?
ESP32_DevkitC_V4
- IDE used
ESP-IDF 3.3
- Operating System [Windows|Linux|MacOS]
Windows10

- Version of FreeRTOS (run ``git describe --tags`` to find it)
202012.00

- Project
Custom Application

**Expected behavior**
We expecting after RPA is changed advertisement will be restarted.

Problem is caused by missing function for set member _BTInterface.pStopAdvCb.
That callback is never called.

**Screenshots or console output**
None

**To reproduce**

1. Build amazon freertos with nimBLE Bluetooth stack 
2. Run the device and check GATT services are advertised
3. Wait for 15 minutes and check again

**Code to FIX the bug**
Patch is here https://gist.github.com/bgklika/e4b922b252824f0ac67d0277291e4ce3

In addition some support from the application is required for 
1) set callback
2) restart advertisement when it was turned off by RPA change

paulbartell, 2021-04-23:
@bgklika My understanding is that this is an intentional time limit due to security concerns. @ravibhagavandas can share more of the reasoning behind this choice.

ravibhagavandas, 2021-04-24:
@bgklika 
Thanks for providing a patch. As mentioned above, its recommended to change the RPA address after a duration to make the device non traceable. Hence the advertising needs to be stopped and restarted. However it makes sense to have the callback for application to know when the advertising is stopped. We will evaluate the patch and provide a fix soon.

ravibhagavandas, 2021-04-30:
The PR #3121 merges the patch to add an API to set stop advertisement callback. I am closing this issue. Please feel free to open a new issue if you have further questions.

closed 2021-04-30 by ravibhagavandas