aws/amazon-freertos #3113: Wi-Fi credentials can't be stored after not existing credentials tried to delete

bgklika, 2021-04-21:
**Describe the bug**
We are use Wi-Fi credentials provisioning over BLE GATT service.
In case of ESP-IDF 3.3 after we tried to remove for example credentials with index 0 when no credentials have been stored at all  - we can't add any new credentials.

Source of the problem
Client send the request to add Wi-FI credentials. Due to 'connect' flag is set in request before saving to NVS _addNewNetwork() function of iot_ble_wifi_provisioning.c source file try to connect to AP. Because that connect has been failed _appendNetwork() does not called and `wifiProvisioning.numNetworks` still 0.

Unfortunately test app does not check result of request by waiting for response and before exit try to remove Wi-Fi credential with index 0 by send request.

Control path of handling that request is below:

static void _callback()
   _handleDeleteNetworkRequest( pBuffer, length );
        status = _deserializeDeleteNetworkRequest( pData, length, &wifiProvisioning.deleteNetworkRequest );
        	_deleteNetworkTask() async
                     ret = _popNetwork( wifiProvisioning.deleteNetworkRequest.idx, NULL );
                         ret = WIFI_NetworkDelete( STORAGE_INDEX( index ) );		 // !!! HERE  -1 will be passed
                         only on success wifiProvisioning.numNetworks--;
Possible resolutions

update WIFI_NetworkDelete() of ESP-IDF to return error when out of range index is passed to function

update _popNetwork() implementation to check index before passing

**System information**
- Which hardware board or part numbers?
ESP32_DevKitC_V4

- IDE used
ESP-IDF 3.3

- Operating System
Windows10
- Version of FreeRTOS (run ``git describe --tags`` to find it)
202012.00

- Project
Custom Application

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
Wi-Fi credentials provisioning GATT service will works after we tried to delete not existing credentials

**Screenshots or console output**
1829 110623 [iot_thread] [ERROR][BLE_WIFI_PROV][110623] Failed to add a new network, max networks limit (8) reached

**To reproduce**

1. Try to add new Wi-Fi network which is not exist via Wi-Fi credentials provisioning. 
2. Try to delete Wi-Fi credentials with index 0 which is not exist due to step 1. has been failed.
3. After that any addition of good, existing Wi-Fi network with valid credentials will be failed.

**Code to FIX the bug**
Uploaded as patch here https://gist.github.com/bgklika/2c2ed7b5969748b0d90cf84213a6bee8

ravibhagavandas, 2021-04-24:
@bgklika 
Thanks for bringing this into our attention. We are reproducing the issue and provide an update soon.

ravibhagavandas, 2021-04-30:
The PR #3121 merges the fix for the issue to master. I am closing this issue. Please feel free to open a new issue if you have further questions.