aws/amazon-freertos #3117: Crypto signature is not valid after OTA update PIC32MZ FreeRTOS V202002.00

scottnightingale, 2021-04-24:
We built out an application based on the OTA update demo just over a year ago, which has been working happily up until a few days ago when the self signed code signing certificate expired :-/  

I followed along with this documentation to create a new self signed cert: https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert-mchip.html

Copied the new cert to demos/ota/bootloader/utility/codesigner_cert_utility/aws_ota_codesigner_certificate.pem and updated demos/include/aws_ota_codesigner_certificate.h as per:
https://docs.aws.amazon.com/freertos/latest/userguide/download-ota-mchip.html

Created a new OTA job + Code Signing Profile targeting the new certificate

The update is picked up by the test device, but on booting the new image the crypto signature validation fails causing the OTA update to fail.

I have reverted the project back to the vanilla OTA demo and behaviour is still the same.

**System information**

PIC32MZ EF
MPLAB X IDE v5.40
Linux
FreeRTOS V202002.00
AWS OTA demo

**Expected behavior**

OTA update is download, image activated and successfully booted confirming the update as successful

**Screenshots or console output**

=============================================================

3564 1102704 [OTA Agent Task] [prvIngestDataBlock] Received final expected block of file.
3565 1102705 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
3566 1102705 [OTA Agent Task] [prvPAL_CloseFile] Authenticating and closing file.
3567 1102705 [OTA Agent Task] [prvPAL_CheckFileSignature] Started sig-sha256-ecdsa signature verification, file: aws_ota_codesi3568 1102705 [OTA Agent Task] [prvPAL_ReadAndAssumeCertificate] No such cert0
3570 1103122 [OTA Agent Task] [prvPAL_CloseFile] sig-sha256-ecdsa signature verification passed.
3571 1103122 [OTA Agent Task] [prvContextUpdateImageHeaderAndTrailer] OTA Sequence Number: 5
3572 1103122 [OTA Agent Task] [prvContextUpdateImageHeaderAndTrailer] Image - Start: 0xbd000008, End: 0xbd083890
3573 1103126 [OTA Agent Task] [prvContextUpdateImageHeaderAndTrailer] Writing Trailer at: 0x9d183890
                                                                                                    3574 1103126 [OTA Agent Task] [prvPAL_CloseFile] Image header updated.
3575 1103126 [OTA Agent Task] [prvIngestDataBlock] File receive complete and signature is valid.
3576 1103127 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
3577 1103127 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","upda3578 1103127 [OTA Agent Task] [INFO ][MQTT][1103127] (MQTT connection 0x8001.
3579 1103127 [OTA Agent Task] [INFO ][MQTT][1103127] (MQTT connection 0x80015ad0, PUBLISH operation 0x80016708) Waiting for ope3580 1103182 [OTA Agent Task] [INFO ][MQTT][1103182] (MQTT connection 0x8001.
3583 1103182 [OTA Agent Task] [INFO ][MQTT][1103182] (MQTT connection 0x80015ad0) Disconnecting connection.
3584 1103184 [OTA Agent Task] [INFO ][MQTT][1103184] (MQTT connection 0x80015ad0, DISCONNECT operation 0x80016248) Waiting for 3585 1103184 [OTA Agent Task] [INFO ][MQTT][1103184] (MQTT connection 0x8001.
3587 1103184 [OTA Agent Task] [INFO ][MQTT][1103184] (MQTT connection 0x80015ad0) Network connection closed.
3588 1103247 [OTA Agent Task] [prvPAL_ActivateNewImage] Activating the new MCU image.
3589 1103247 [OTA Agent Task] [prvPAL_ResetDevice] Resetting the device.

Bootloader version 00.09.00
[prvBOOT_Init] Watchdog timer initialized.
[prvBOOT_Init] Crypto initialized.

[prvValidateImage] Validating image at Bank : 0
[prvValidateImage] No application image or magic code present at: 0xbd000000
[prvBOOT_ValidateImages] Validation failed for image at 0xbd000000 

[prvValidateImage] Validating image at Bank : 1
[prvValidateImage] Valid magic code at: 0xbd100000
[prvValidateImage] Valid image flags: 0xff at: 0xbd100000
[prvValidateImage] Addresses are valid.
[prvValidateImage] Crypto signature is not valid.
[prvBOOT_ValidateImages] Validation failed for image at 0xbd100000 
[BOOT_FLASH_EraseBank] Bank erased at : 0xbd100000

[prvBOOT_ValidateImages] Booting default image. 

=============================================================

Thanks for your help!

Scott

scottnightingale, 2021-04-25:
I can confirm this issue with the latest Amazon FreeRTOS OTA demo code base.  Setting bootconfigENABLE_CRYPTO_SIGNATURE_VERIFICATION = 0U allows the update to succeed

pvyawaha, 2021-04-27:
Hello,

Thank you for sharing  the logs. Can you also share information where the public key/cert for the bootloader is stored. It seems the code signer certificate is not updated for the bootloader which is failing the crypto check.

scottnightingale, 2021-04-28:
thanks for your replay Prasad, do you mean: demos/ota/bootloader/utility/codesigner_cert_utility/aws_ota_codesigner_certificate.pem ?

I did not see any documentation on copying the code signing PK anywhere.  Please advise if I have missed this?

paulbartell, 2021-04-28:
@scottnightingale Typically the expiration date of a code signing certificate used by a bootloader is not enforced, since this would brick an end-product on a specific date. Additionally, many platforms do not have an RTC available, so would have no current date to compare the certificate expiration to.

For this reason, It's best practice to use separate keys for the image signature (checked on every bootup) and the upgrade package signature (checked during OTA upgrades).

paulbartell, 2021-04-28:
There is some documentation regarding setting up the bootloader key in the PIC32MZEF user guide at https://docs.aws.amazon.com/freertos/latest/userguide/download-ota-mchip.html.

scottnightingale, 2021-04-28:
Thanks Paul.  So I've been over that specific documentation a number of times but I must be missing something. Can I clarify a couple of things for me.

1) Are steps 3,4 and 7 referring to the code signing certificate used to sign updates created in these steps: https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert-mchip.html ?

2) From this documentation https://docs.aws.amazon.com/freertos/latest/userguide/burn-initial-firmware-microchip.html is this the command that signs the bootloader? `factory_image_generator.py -b mplab.production.bin -p MCHP-Curiosity-PIC32MZEF –k private_key.pem  -x aws_bootloader.X.production.hex` # 

3) If the answer to 2 is yes, is this what you mean by the bootloader should use a different key in that the private key referenced in this command should be different from the private key used to generate the signing certificate in 1?

paulbartell, 2021-04-29:
@scottnightingale My apologies. I was confused by the documentation as well and my last message was probably a bit misleading.

In the current implementation, only a single ECDSA key is used for both 
1. verifying the ota image when it is received by the application code and written to flash.
2. verification of the application image by the bootloader during startup

Only the ECDSA signature of the application image is verified by the bootloader. No x509 certificate is used by the bootloader. This means that you could re-use your existing ecdsa private key and generate a new certificate with a different expiration date. This should allow you to use the OTA service to sign your application images and satisfy whatever checks the OTA service performs on the certificate validity date.

Below is a summary of my current understanding of the bootloader implementation for the PIC32MZEF:

A few different types of images are built for the PIC32MZEF:
| Short name  | filename ext | Description |
| ------------- | ------------- | ------------- |
| bootloader | aws_bootloader.X.production.hex | Bootloader image (contains ECDSA public key). This is written at the factory and not meant to be updated over the air because it is not cryptographically signed. |
| app-image | .bin | raw application binary |
| ota-image | .ota.bin | application binary with the OTA descriptor prepended. Use this image when you want the OTA service to sign the image. |
| signed-ota-image | .initial.bin | ota-image + trailer (where the trailer includes some metadata for the bootloader and an ECDSA signature). Use this image as a "pre-signed" image with the OTA service. |
|  factory-image | .factory.unified.hex  | bootloader + signed-ota-image. This contains the bootloader and signed application image |

On bootup, the PIC32MZEF loads the bootloader which was originally flashed at the factory. This bootloader image is not cryptographically signed for verified on boot. This bootloader determines which flash slot to load an application image from and verifies the chosen application image against an ECDSA _public key_ that was defined when the bootloader was first built as part of the factory image.

The current OTA implementation does not support updating the bootloader and this seems like a wise design choice since the bootloader image is not cryptographically verified before it is loaded by the PIC32MZEF. You may be able to implement ota for the bootloader image, but you should carefully consider the effects of this on your security threat model.

A helpful Readme file describing the steps for using the bootloader: https://github.com/aws/amazon-freertos/blob/master/vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/README.md

Microchip has some lower level details over here that could be helpful: http://ww1.microchip.com/downloads/en/DeviceDoc/60001193B.pdf.

The following web page describes the format used by application images in flash and sent via OTA: https://docs.aws.amazon.com/freertos/latest/userguide/microchip-bootloader.html

The main code implementing the bootloader is located in this file: 
https://github.com/aws/amazon-freertos/blob/master/vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/loader/aws_boot_loader.c.

To answer your questions above directly:

> Are steps 3,4 and 7 referring to the code signing certificate used to sign updates created in these steps: https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert-mchip.html ?

The ECDSA key created in step 2 is used for signing application images which are included in both the factory image and any ota update images. The certificate from step 3 is only used by the ota service. You could repeat step 3 with the same ECDSA key you originally used for your factory image and generate a longer-lived certificate.

> From this documentation https://docs.aws.amazon.com/freertos/latest/userguide/burn-initial-firmware-microchip.html is this the command that signs the bootloader? factory_image_generator.py -b mplab.production.bin -p MCHP-Curiosity-PIC32MZEF –k private_key.pem -x aws_bootloader.X.production.hex #

The bootloader is not signed. Rather it contains a copy of the ECDSA public key which the bootloader uses to verify application images. factory_image_generator.py adds some metadata, signs the application image, and combines it with the bootloader image to form the factory image.

> If the answer to 2 is yes, is this what you mean by the bootloader should use a different key in that the private key referenced in this command should be different from the private key used to generate the signing certificate in 1?

On other chipsets, One-Time-Programmable memory (called eFUSE or OTP) is used to configure the chip and program a public key into the chip. Bootloader images can be cryptographically signed and then verified by the bootROM built into the chip on each boot. This allows all software in RW memory to be verified before it is run. In this case, updating the bootloader over the air would have a decreased effect on your product's security model.

For security reasons, it's best to use separate keys to sign the bootloader and application images so that the application key can be rotated frequently, while the bootloader key stays the same throughout the lifetime of the product and is more heavily guarded.

paulbartell, 2021-05-04:
@scottnightingale Were you able to resolve your OTA / code signing issue?

scottnightingale, 2021-05-05:
Hi @paulbartell, thanks for following this up.  Sorry for the slow reply, just working through this.  Will come back to you shortly.

scottnightingale, 2021-05-06:
@paulbartell I have been unable to resolve this issue.  It seems there are a few things that don't line up between the documentation and the code base and I'm obviously misinterpreting one of the jumps that is being made.  Can we take this convo offline and I can share more information?

closed 2021-05-06 by scottnightingale

paulbartell, 2021-05-10:
Note: 

Two separate header files in the repository contain the public key / certificate information. One for the ECC public key (in the bootloader image and another containing the OTA code signing certificate (in the application image).

The aws_boot_codesigner_public_key.h file is generated automatically using the codesigner_cert_utility.py script located in ```vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/utility/codesigner_cert_utility/codesigner_cert_utility.py```. This header file contains only the ECDSA public key (pulled from the certificate file) and is used by the PIC32MZEF bootloader to verify the selected image on each bootup.

The aws_ota_codesigner_certificate.h file can be generated by hand (follow the instructions [here](https://docs.aws.amazon.com/freertos/latest/userguide/download-ota-mchip.html) ) or automatically using the html page located at ```tools/certificate_configuration/PEMfileToCString.html```. This header file is used by the OTA library built into the application image to validate the firmware image received from the AWS OTA service.