aws/amazon-freertos #3119: Remote logging / Cloudwatch integration

dFohlen, 2021-04-26:
Hello AWS FreeRTOS team 👋 

**Is your feature request related to a problem? Please describe.**  

We currently have several IoT devices based on the ESP32 with the FreeRTOS version 202012.00. Sometimes it is necessary to follow the software process on the devices in order to be able to find and correct possible errors. It would be helpful if the log output could optionally be streamed from the ESP32 via a Cloudwatch integration.

**Describe the solution you would like.**  
A Cloudwatch integration for writing messages from IoT devices. This could possibly also be implemented directly as an endpoint in `esplog.h` or` iot_logging.h`.

Best regards

yourslab, 2021-04-26:
Thanks for your feedback. One thing to consider is publishing all your logs to a certain MQTT topic. You can setup an [AWS IoT rule action](https://docs.aws.amazon.com/iot/latest/developerguide/iot-rule-actions.html) for storing messages published onto that topic into Cloudwatch. Does this fit your current use case? If not, what else do you think this Cloudwatch integration should contain?

dFohlen, 2021-05-05:
Hi @yourslab,

Thank you, that should be enough for our requirements 👍

closed 2021-05-05 by dFohlen