aws/amazon-freertos #3124: CMakeLists.txt linking error

Raghav3107, 2021-04-27:
Hello,
I m using the [personal-box-freertos-demo ](https://github.com/aws-samples/personal-box-freertos-demo)repository. In the src folder, I wrote a program that tries to use the iot_mqtt_api.c files API which is at _/amazon-freertos/libraries/c_sdk/standard/mqtt/src_ directory in starting program not able to find the file so I pass the directory after that is able to compile but at linking time its show error.
Also when I tried to use the FreeRTOS_IP.h which is at _/amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include/FreeRTOS_IP.h_, program not able to find the file or directory.

**Log-1:**
```
able -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -Wno-maybe-uninitialized -MD -MT CMakeFiles/afr_workshop.dir/src/connection_manager.c.obj -MF CMakeFiles/afr_workshop.dir/src/connection_manager.c.obj.d -o CMakeFiles/afr_workshop.dir/src/connection_manager.c.obj   -c ../src/connection_manager.c
../src/connection_manager.c:14:25: fatal error: FreeRTOS_IP.h: No such file or directory
compilation terminated.
[15/25] Building C object CMakeFiles/afr_workshop.dir/src/simple_pub_sub_task.c.obj
In file included from ../include/subscription_manager.h:55:0,
                 from ../src/simple_pub_sub_task.c:24:
../amazon-freertos/libraries/logging/include/logging_stack.h:95:0: warning: "LogError" redefined
         #define LogError( message )    SdkLog( ( "[ERROR] [%s] "LOG_METADATA_FORMAT, LIBRARY_LOG_NAME, LOG_METADATA_ARGS ) ); SdkLog( message ); SdkLog( ( "\r\n" ) )
 ^
In file included from ../amazon-freertos/libraries/coreMQTT/source/include/core_mqtt.h:40:0,
                 from ../src/simple_pub_sub_task.c:18:
../amazon-freertos/libraries/coreMQTT/source/include/core_mqtt_config_defaults.h:183:0: note: this is the location of the previous definition
     #define LogError( message )
 ^
../src/simple_pub_sub_task.c: In function 'prvSubscribeToTopic':
../src/simple_pub_sub_task.c:315:14: warning: variable 'ulSubscribeMessageID' set but not used [-Wunused-but-set-variable]
     uint32_t ulSubscribeMessageID;
              ^
../src/simple_pub_sub_task.c: In function 'prvSimpleSubscribePublishTask':
../src/simple_pub_sub_task.c:439:40: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     xCommandParams.cmdCompleteCallback = prvPublishCommandCallback;

```
Here is the **CMakeLists.txt** file at directory /_personal-box-freertos-demo_ remaining CMakeLists file is the same.

[CMakeLists.txt](https://github.com/aws/amazon-freertos/files/6386296/CMakeLists.txt)

paulbartell, 2021-04-28:
@Raghav3107

Please try adding an additional include_directories line to your CMakeLists.txt. Something like this should suffice:
```
include_directories(afr_workshop PRIVATE "amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include")
```

You may also need to add AFR:freertos_plus_tcp to target_link_libraries item.

Thanks,
Paul

Raghav3107, 2021-04-29:
Hello  @paulbartell 
Thanks for your suggestion. 
Now [following error](https://github.com/aws/amazon-freertos/issues/3124#issue-869057018) gone. But some file which is in directory `/amazon-freertos/libraries/c_sdk/standard/mqtt/src `when I use the API's of this directory file first it was unable to find the file so I include the directory and cmake able to compile the program but it not able to link the API.
showing` "undefine reference error".`
If you see the CMakeLists.txt at directory `/amazon-freertos/libraries/` it links the files of` /c_sdk/ `and /freertos-plus/. Can we link this CMake file and this file will link the other directory by default?

paulbartell, 2021-04-29:
@Raghav3107 It is usually best to link only the libraries needed for your project. Can you paste the full error message in this thread?

Likely adding the missing items to the target_link_libraries section of your CMakeLists.txt will fix the problem.