aws/amazon-freertos #3126: Triggering an OTA update from a mobile app and notifications on ESP32

Johnnyjax, 2021-04-28:
Hello. I'm currently trying to modify the ota demo so that notifications come into a shadow topic which the app is subscribed to whenever an update is available and the app can start the OTA update process if a user chooses to update. Is there a way to do this with the amazon OTA agent? Thanks.

sukhmanm, 2021-04-29:
Hi @Johnnyjax,

Just to clarify, are you looking to add publish/subscribe functionality for MQTT shadow topics within the OTA update demo? Also, could you please clarify which code version you are using?

To perform additional MQTT operations as required by your application (ex. shadow updates, pub/sub), you can share the MQTT connection initialized by the OTA demo with additional demo tasks as needed. [This demo](https://github.com/FreeRTOS/coreMQTT-Agent-Demos/blob/main/source/demo-tasks/ota_over_mqtt_demo.c) in our labs repository uses a MQTT agent (based on [coreMQTT](https://github.com/FreeRTOS/coreMQTT)) to manage thread safety of a shared MQTT connection, and subscribes to OTA job notifications with some registered callbacks. That repository also contains a [shadow demo](https://github.com/FreeRTOS/coreMQTT-Agent-Demos/blob/main/source/demo-tasks/shadow_demo.c), which uses the [Device Shadow Library](https://github.com/aws/Device-Shadow-for-AWS-IoT-embedded-sdk/tree/main) to maintain and update shadow documents. The OTA demo can be updated accordingly to include any of these relevant to your application's needs. 

Please let me know if I can provide further clarification on any topic. Thanks.

Johnnyjax, 2021-04-29:
Thanks for you response @sukhmanm. I'm using the 202012.00 version. I'm not just looking to add publish/subscibe functionality for MQTT shadow topics within the OTA update demo, but I want a way for the mobile app to be notified whenever a new OTA job is created for a new update, I'm thinking of using a shadow topic for this but If you have a better suggesion, that'd be great. I also don't want the OTA update to begin immediately the Job is created, but instead begin when the user wants the update by selecting the option through the mobile app. The way the [ota demo](https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c) that comes with amazon freertos currently works is there really isn't a way to make this kind of modification to the code. Also, the job begins immediately when the demo is run and no way to get notified when a Job is available. Thanks

pvyawaha, 2021-04-29:
Hello,

Thank you for sharing more details regarding the requirement. We have added events in the application callback when a valid job is received in the OTA LTS release version of the library. https://freertos.org/Documentation/api-ref/ota-for-aws-iot-embedded-sdk/docs/doxygen/output/html/group__ota__enum__types.html#ga96a2e1f5cfad783897e805196eee39a7. 
The library is released and work is in progress for the integration release and will be available in this repository. The application callback will receive event `OtaJobEventReceivedJob`  when a valid job is received. It also has all the information regarding the ota job so application can decide to continue, suspend or abort.  The previous release you are using i.e 202012.00 is having callback when the job is received and is successfully processed ( download is complete ) but not when the job notification is received. If you To support this on 202012.00  a new event will have to be added in the callback as present in the OTA LTS library.

Johnnyjax, 2021-05-06:
Hello @pvyawaha, thank you for your response. Do you have any idea how I can add this callback that will trigger whenever a job notification is received? I've been looking through the amazon-freertos ota library and there's no clear path on how to acheive this.

closed 2021-05-26 by Johnnyjax