aws/amazon-freertos #3130: Stack overflow at when the initialing the repository

Raghav3107, 2021-04-29:
Hello,
I was trying to run a [personal-box-freertos-demo](https://github.com/aws-samples/personal-box-freertos-demo) and testing the shadow demo in it. The program sometimes shows the stack overflow error when it tries to initialize the networks and some it runs very well. I was searching for a way to increase the stack size but I m not able to find any way or function to do that.
I m not sure why this issue is coming?

Also, the program not able to initialize the ble. You can see an error in the log "Failed to toggle BLE".

**Error Log:**

```
I (73) boot: Chip Revision: 1
I (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3.2-241-g1d2d93acd 2nd stage bootloader
I (41) boot: compile time 13:10:05
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (100) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (108) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (115) boot: End of partition table
I (120) boot: ota rollback check done
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2eab8 (191160) map
I (207) esp_image: segment 1: paddr=0x0004eae0 vaddr=0x3ffbdb60 size=0x01530 (  5424) load
I (210) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0xbac8c (765068) map
0x400d0018: _flash_cache_start at ??:?

I (482) esp_image: segment 3: paddr=0x0010acac vaddr=0x3ffbf090 size=0x02310 (  8976) load
I (486) esp_image: segment 4: paddr=0x0010cfc4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (490) esp_image: segment 5: paddr=0x0010d3cc vaddr=0x40080400 size=0x1a57c (107900) load
I (558) boot: Loaded app from partition at offset 0x20000
I (558) boot: Disabling RNG early entropy source...
I (559) cpu_start: Pro cpu up.
I (562) cpu_start: Application information:
I (567) cpu_start: Project name:     esp-idf
I (572) cpu_start: App version:      222eced
I (577) cpu_start: Compile time:     Apr 26 2021 13:41:35
I (583) cpu_start: ELF file SHA256:  92b32c9ec80fb04b...
I (589) cpu_start: ESP-IDF:          v3.3.2-241-g1d2d93acd
I (595) cpu_start: Single core mode
I (600) heap_init: Initializing. RAM available for dynamic allocation:
I (607) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (613) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (619) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (625) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (631) heap_init: At 3FFCE3D0 len 00011C30 (71 KiB): DRAM
I (637) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (644) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (650) heap_init: At 4009A97C len 00005684 (21 KiB): IRAM
I (656) cpu_start: Pro cpu start user code
I (3) cpu_start: Starting scheduler on PRO CPU.
0 1 [main] Initializing lwIP TCP stack
I (27) PKCS11: Initializing NVS partition: "storage"
E (67) PKCS11: failed nvs get file size 4354 0
1 6 [main] Write certificate...
2 12 [main] [INFO ][DEMO][120] ---------STARTING DEMO---------

3 12 [main] [INFO ][INIT][120] SDK successfully initialized.
4 12 [main] [INFO ][DEMO][120] AwsIotNetworkManager_Init
5 12 [main] [INFO ][DEMO][120] IotSemaphore_Create( &demoNetworkSemaphore, 0, 1 ) != true )
6 13 [main] [INFO ][DEMO][130] AwsIotNetworkManager_SubscribeForStateChange
7 14 [main] [INFO ][DEMO][140] AwsIotNetworkManager_EnableNetwork
8 14 [main] [ERROR][DEMO][140] Failed to toggle BLE on.
I (177) wifi:wifi driver task: 3ffb2230, prio:23, stack:3584, core=0
I (177) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (187) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (197) wifi:wifi firmware version: 1afdae0
I (197) wifi:config NVS flash: enabled
I (197) wifi:config nano formating: disabled
I (207) wifi:Init dynamic tx buffer num: 32
I (207) wifi:Init data frame dynamic rx buffer num: 32
I (207) wifi:Init management frame dynamic rx buffer num: 32
I (217) wifi:Init management short buffer num: 32
I (217) wifi:Init static rx buffer size: 1600
I (227) wifi:Init static rx buffer num: 10
I (227) wifi:Init dynamic rx buffer num: 32
I (327) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (327) wifi:mode : sta (fc:f5:c4:6e:89:3c)
I (327) WIFI: SYSTEM_EVENT_STA_START
I (447) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (1437) wifi:state: init -> auth (b0)
I (1467) wifi:state: auth -> assoc (0)
I (1527) wifi:state: assoc -> run (10)
I (3997) wifi:connected with Gryffindor, aid = 1, channel 1, 40U, bssid = c0:25:e9:42:30:a2
I (3997) wifi:security type: 3, phy: bgn, rssi: -59
I (4007) wifi:pm start, type: 1

I (4007) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (4007) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (14327) event: sta ip: 192.168.0.104, mask: 255.255.255.0, gw: 192.168.0.1
I (14327) WIFI: SYSTEM_EVENT_STA_GOT_IP
9 1431 [eventTask] [INFO ][DEMO][14310] Connected to WiFi access point, ip address: 192.168.0.104.
10 1432 [main] [INFO ][DEMO][14320] Successfully initialized the demo. Network type for the demo: 1
11 1433 [main] [INFO] [MQTT] [mqtt_demo_helpers.c:363] 12 1433 [main] Create a TCP connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883.13 1434 [main] 
E (20437) PKCS11: failed nvs get file size 4354 0
***ERROR*** A stack overflow in task main has been detected.
abort() was called at PC 0x4008cfa0 on core 0
0x4008cfa0: vApplicationStackOverflowHook at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

ELF file SHA256: 92b32c9ec80fb04b

Backtrace: 0x4008cd58:0x3ffb99d0 0x4008cf89:0x3ffb99f0 0x4008cfa0:0x3ffb9a10 0x4008ead5:0x3ffb9a30 0x4008d6c8:0x3ffb9a50 0x4008d67a:0x40080ff9
0x4008cd58: invoke_abort at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008cf89: abort at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008cfa0: vApplicationStackOverflowHook at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008ead5: vTaskSwitchContext at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/tasks.c:5248 (discriminator 4)

0x4008d6c8: _frxt_dispatch at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/portasm.S:406

0x4008d67a: _frxt_int_exit at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/portasm.S:206

0x40080ff9: heap_caps_free at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/heap/heap_caps.c:232
```

sukhmanm, 2021-04-30:
Hi @Raghav3107,

The files [aws_demo_config.h](https://github.com/aws-samples/personal-box-freertos-demo/blob/master/amazon-freertos-configs/aws_demo_config.h) and [iot_config.h](https://github.com/aws-samples/personal-box-freertos-demo/blob/master/amazon-freertos-configs/iot_config.h) contain some stack configurations that you might find useful. Also, keep in mind that the various tasks created in [this entry function](https://github.com/aws-samples/personal-box-freertos-demo/blob/222eced0d8aacb1bd6c864f5b94c08972ff9ddd9/src/controller.c#L78-L121) have size specifications as well, and may be contributing to your issue. Let me know if you are still facing errors after finetuning these values. Thanks.

Raghav3107, 2021-04-30:
Hello @sukhmanm 
Thanks for the reply.

I tried to increase the [stack size](https://github.com/Raghav3107/aws-freertos-workspace/blob/9442bef3e591d5cce88442b153313c0c0b9109d0/amazon-freertos-configs/iot_config.h#L59) but still same issue is occurring.
Also, I remove many tasks from the [repository ](https://github.com/Raghav3107/aws-freertos-workspace.git)but nothing changes.

sukhmanm, 2021-04-30:
@Raghav3107 Have you tried modifying the [demo task's size](https://github.com/Raghav3107/aws-freertos-workspace/blob/9442bef3e591d5cce88442b153313c0c0b9109d0/src/connection_manager.c#L893) specifically? If the error persists, I'll forward this to someone with greater familiarity to assist you on this issue.

Raghav3107, 2021-05-01:
Hello @sukhmanm  
Thanks for the reply and for assisting me.
I tried to increase the [demo stack size ](https://github.com/Raghav3107/aws-freertos-workspace/blob/2772931cb7a271a9d29aad9bc6c00b6d22831ef4/src/connection_manager.c#L895)and [task size](https://github.com/Raghav3107/aws-freertos-workspace/blob/2772931cb7a271a9d29aad9bc6c00b6d22831ef4/amazon-freertos-configs/aws_demo_config.h#L51) but still, the error is persisting and also I m getting Heap Corrupt Error.

**Stack Overflow Error:**
```
I (1845) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (1865) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3225) event: sta ip: 192.168.0.100, mask: 255.255.255.0, gw: 192.168.0.1
I (3225) WIFI: SYSTEM_EVENT_STA_GOT_IP
9 288 [eventTask] [INFO ][DEMO][2880] Connected to WiFi access point, ip address: 192.168.0.100.
10 289 [main] [INFO ][DEMO][2890] Successfully initialized the demo. Network type for the demo: 1
prvConnectAndCreateDemoTasks ==========
connection to broker 
300 prvSocketConnect........... 
337 prvSocketConnect........... 
Creating a TLS connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883. 
E (3305) PKCS11: failed nvs get file size 4354 0
***ERROR*** A stack overflow in task main has been detected.
abort() was called at PC 0x4008cfa8 on core 0
0x4008cfa8: vApplicationStackOverflowHook at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

ELF file SHA256: 37c06ad830a39f3c

Backtrace: 0x4008cd60:0x3ffb9a60 0x4008cf91:0x3ffb9a80 0x4008cfa8:0x3ffb9aa0 0x4008eadd:0x3ffb9ac0 0x4008d6d0:0x3ffb9ae0 0x4008d682:0x3ffb9b00 0x40080e44:0x00000073
0x4008cd60: invoke_abort at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008cf91: abort at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008cfa8: vApplicationStackOverflowHook at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008eadd: vTaskSwitchContext at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/tasks.c:5248 (discriminator 4)

0x4008d6d0: _frxt_dispatch at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/portasm.S:406

0x4008d682: _frxt_int_exit at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/portasm.S:206

0x40080e44: esp_mbedtls_mem_calloc at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_mem.c:25

```

**Corrupt Heap Error:**
```
337 prvSocketConnect........... 
Creating a TLS connection to a3ca3lp4kvh6ns-ats.iot.ap-south-1.amazonaws.com:8883. 
E (3315) PKCS11: failed nvs get file size 4354 0
CORRUPT HEAP: multi_heap.c:431 detected at 0x3ffb9a34
abort() was called at PC 0x400907a4 on core 0
0x400907a4: multi_heap_assert at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:431

ELF file SHA256: 37c06ad830a39f3c

Backtrace: 0x4008cd60:0x3ffb9ab0 0x4008cf91:0x3ffb9ad0 0x400907a4:0x3ffb9af0 0x40080f89:0x3ffb9b10 0x40081165:0x3ffb9b30 0x40080e44:0x3ffb9b50 0x40140ffd:0x3ffb9b70 0x40137ffe:0x3ffb9b90 0x401380dd:0x3ffb9bb0 0x4013893e:0x3ffb9bd0 0x401389f5:0x3ffb9c10 0x40138d34:0x3ffb9c40 0x40138eb7:0x3ffb9cc0 0x40139146:0x3ffb9ce0 0x40141d08:0x3ffba350 0x401426d8:0x3ffba400 0x40142765:0x3ffba420 0x4013f95d:0x3ffba450 0x4013f73d:0x3ffba480 0x4013f77a:0x3ffba4b0 0x4013503b:0x3ffba4e0 0x40136ab9:0x3ffba570 0x4013f73d:0x3ffba6a0 0x4014f08f:0x3ffba6d0 0x4014f2ac:0x3ffba740 0x40146005:0x3ffba760 0x40146031:0x3ffba780 0x40136fba:0x3ffba7a0 0x400ddb88:0x3ffba7c0 0x40136342:0x3ffba820 0x400dd4d4:0x3ffba840 0x400dd5db:0x3ffba870 0x400dd813:0x3ffba890 0x400d1549:0x3ffba8b0 0x400d15a3:0x3ffba920 0x400d1691:0x3ffba940 0x400d1928:0x3ffba960 0x400d1a03:0x3ffba980 0x400d6d4d:0x3ffba9a0
0x4008cd60: invoke_abort at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008cf91: abort at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x400907a4: multi_heap_assert at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/heap/multi_heap.c:4 [... cut ...]
0x400d1691: prvConnectAndCreateDemoTasks at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../src/connection_manager.c:857

0x400d1928: runDemoTask_custom at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../src/demoRunner.c:289

0x400d1a03: app_main at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../src/main.c:157

0x400d6d4d: main_task at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/cpu_start.c:574

```

shubhamkulkarni97, 2021-05-03:
@Raghav3107 You can increase stack size for main task by following these steps:

* Run `idf.py menuconfig`
* Goto `Component Config -> Common ESP-related`
* Update value for `Main task stack size` with required value. By default it is set to 3584.

Hope this helps!

Raghav3107, 2021-05-03:
Thank you so much @shubhamkulkarni97 @sukhmanm for assisting me and for your valuable suggestion. 
This issue is resolved now. I increase the main stack size from 3458 to 4500.

If possible can you help me to solve the following issue?

https://github.com/aws/amazon-freertos/issues/3051
https://github.com/FreeRTOS/coreMQTT-Agent-Demos/issues/46

closed 2021-05-03 by Raghav3107