aws/amazon-freertos #3131: how to change default source directory

Raghav3107, 2021-04-29:
Hello, 
I want to run the shadow demo files in a directory amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code instead of the default directory.
I put the shadow file directory of the application folder and try to compile but cmake not able to find the source file.

I think it is because the source directory is set to demos. I was searching for where it declares but I did not find anything like that. 
Is there another way we can run the program in the application folder? or Set the source directory to run the application folder?
[image]

sukhmanm, 2021-04-30:
Hi @Raghav3107,

It is likely the CMake files haven't been correctly updated to reflect this change. Were you able to run the demo successfully prior to the file relocation? Could you also further clarify what you mean by: "set the source directory to run the application folder?"

In the current CMake infrastructure, the board's `application_code/main.c` file is linked to the demo target in the board-specific [CMakeLists.txt file](https://github.com/aws/amazon-freertos/blob/176636b9cc60fdeab297a57798666105eac95946/vendors/espressif/boards/esp32/CMakeLists.txt#L466). Then, `main.c` is able to [run the demo](https://github.com/aws/amazon-freertos/blob/176636b9cc60fdeab297a57798666105eac95946/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L159) selected (in [`aws_demo_config.h`](https://github.com/aws/amazon-freertos/blob/176636b9cc60fdeab297a57798666105eac95946/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L51)). The shadow demo module, in this case, is responsible for [adding shadow_demo_main.c](https://github.com/aws/amazon-freertos/blob/176636b9cc60fdeab297a57798666105eac95946/demos/device_shadow_for_aws/CMakeLists.txt#L16) and its dependencies to the CMake build system. This is likely where the error is occurring, as the file can no longer be found in the CMake sub-directory. If the existing workflow is insufficient for your application, [ESP32's "Getting Started Guide"](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project) contains a few examples for modifying the CMake project that may be helpful. Let me know if I can clarify anything in particular. Thanks.

Raghav3107, 2021-04-30:
> Were you able to run the demo successfully prior to the file relocation?

Yes, I m able to the program in the default directory.

> Could you also further clarify what you mean by: "set the source directory to run the application folder?"

I thought there may be some declaration in CMake where they set the source directory to /demos/, that why cmake always searching the file in the demos folder.

Thanks for clarifying the CMake infrastructure and I made some in CMakeLists of shadow and set the directory tod shadow to the application folder. Now it's working now.

sukhmanm, 2021-04-30:
That's great, glad you were able to resolve the issue.