aws/amazon-freertos #3135: Connection issue over port 443

MichalChromecNXP, 2021-05-03:
Hello,
I am trying to run device_shadow_for_aws over port 443, I can make tcp connection but the mqtt connection fails:

```
12 9047 [iot_thread] [DEBUG] Length of serialized CONNECT packet is 28.
13 9048 [iot_thread] [DEBUG] BytesSent=28, BytesRemaining=0
14 9048 [iot_thread] [DEBUG] Successfully sent packet at time 9048.
15 9048 [iot_thread] [DEBUG] Sent 28 bytes of CONNECT packet.
16 9249 [iot_thread] [DEBUG] No data was received from the transport.
17 9450 [iot_thread] [DEBUG] No data was received from the transport.
18 9651 [iot_thread] [DEBUG] No data was received from the transport.
19 9852 [iot_thread] [DEBUG] No data was received from the transport.
20 10053 [iot_thread] [DEBUG] No data was received from the transport.
21 10053 [iot_thread] [ERROR] CONNACK recv failed with status = MQTTNoDataAvailable.
22 10053 [iot_thread] [ERROR] MQTT connection failed with status = MQTTNoDataAvailable.
23 10053 [iot_thread] [ERROR] Connection with MQTT broker failed with status MQTTNoDataAvailable.
24 10053 [iot_thread] [ERROR] Failed to connect to MQTT broker.
```

I have found the issue is caused by pAlpnProtos:
https://github.com/aws/amazon-freertos/blob/176636b9cc60fdeab297a57798666105eac95946/demos/common/mqtt_demo_helpers/mqtt_demo_helpers.c#L430

Probably it should be set in similar way as here:
https://github.com/aws/amazon-freertos/blob/e2a614f4d932c505ef091a5ca1e3393841c2601d/demos/network_manager/aws_iot_network_manager.c#L784

Could you please check it?

Thank you,
Michal Chromec

sukhmanm, 2021-05-03:
Hi @MichalChromecNXP,

You are correct, the ALPN extension is required for MQTT connections over port 443 (8883 does not have this requirement). You can configure the extension in the demo itself using `xSocketConfig.pAlpnProtos` and `xSocketConfig.alpnProtosLen`. AWS IoT Core currently only supports the “x-amzn-mqtt-ca” protocol name (you can find more information regarding this configuration at https://aws.amazon.com/blogs/iot/mqtt-with-tls-client-authentication-on-port-443-why-it-is-useful-and-how-it-works/).

Let me know if you are able to configure the connection successfully, or if any issues remain. Thanks.

MichalChromecNXP, 2021-05-05:
Hi @sukhmanm,
I have changed this line:
```
    xSocketConfig.pAlpnProtos = NULL;
```
to
```
    if( xServerInfo.port == 443 )
    {
        xSocketConfig.pAlpnProtos = socketsAWS_IOT_ALPN_MQTT;
    }
    else
    {
        xSocketConfig.pAlpnProtos = NULL;
    }
```
And it works for me.

Update of `xSocketConfig.alpnProtosLen` is not necessary, it is not used in the code.

Could you please update the code on github to fix the issues with connection over port 443?

I am confused why the configuration of connection parameters is at two places in the code, do you plan to update it to be at one place?

Thank you,
Michal Chromec

aggarw13, 2021-05-22:
Hi @MichalChromecNXP, thank you for pointing out the missing functionality from our MQTT demos. We will update the code to support connection over port 443. 

>I am confused why the configuration of connection parameters is at two places in the code, do you plan to update it to be at one place?

I understand the case of confusion about the presence of credentials at multiple places. The credentials' logic at amazon-freertos/demos/network_manager/aws_iot_network_manager.c represents an old demo runner framework of passing credentials to all demos through the [single function prototype-based entry function](https://github.com/aws/amazon-freertos/blob/main/demos/demo_runner/iot_demo_freertos.c#L311-L318) of all demos. The newer demos (like coreMQTT and AWS specific libraries that use coreMQTT) don't use this parameter (as can be seen for the MQTT Mutual Auth demo [here](https://github.com/aws/amazon-freertos/blob/main/demos/coreMQTT/mqtt_demo_mutual_auth.c#L502)) but instead use the credentials directly from the [`aws_clientcredential.h`](https://github.com/aws/amazon-freertos/blob/main/demos/include/aws_clientcredential.h) and [`aws_clientcredential_keys.h`](https://github.com/aws/amazon-freertos/blob/main/demos/include/aws_clientcredential_keys.h) files to make it easier for the reader to trace the place of defining these credentials from the demo files.

In the future, when none of the demos use the old demo runner framework, we will look into removing the duplicity of credential initialization logic.

aggarw13, 2021-05-25:
@MichalChromecNXP, we have added support for all MQTT based demos to connect with IoT Core on port 443 in #3167.

closed 2021-05-25 by aggarw13