aws/amazon-freertos #3143: How to run the build command in a docker container for the esp32

Johnnyjax, 2021-05-08:
I recently started working with docker and I was able to build my esp-idf projects in a docker container but I've been having issues doing the same for amazon freertos. I ran this command after pulling the espressif/idf docker image from docker hub

`docker run --rm -v ${PWD}:/project -w /project -e IDF_PATH=/project/freertos/vendors/espressif/esp-idf espressif/idf cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja`

When I ran this, it gave the following errors - 

`Adding ESP-IDF tools to PATH...
fatal: not a git repository: /project/freertos/vendors/espressif/esp-idf/../../../../.git/modules/freertos/modules/vendors/espressif/esp-idf
WARNING: Git describe was unsuccessul: Command '['git', 'describe']' returned non-zero exit status 128.
Using Python interpreter in /usr/bin/python
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
click>=5.0
pyserial>=3.0
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
reedsolo>=1.5.3,<=1.5.4
bitstring>=3.1.6
ecdsa>=0.16.0
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/opt/esp/tools/cmake/3.16.4/bin:/opt/esp/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin:/project/freertos/vendors/espressif/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin`

I'm using the amazon freertos as a subdirectory in my custom project with the instructions given [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project) and I have no problems running it without docker. The directory the docker command is called from is the directory of the custom amazon-freertos project.

lundinc2, 2021-05-12:
Hi @Johnnyjax,

I took a stab, was able to get it to start compiling, but hit a wall due to some build issues. Going to post here so you have something to work off in the meantime.

Here is what I did:
```sh
docker pull espressif/idf:release-v4.2
touch entrypoint.sh
chmod +x entrypoint.sh
docker run --rm --mount src="$(pwd)",target=/amazon-freertos,type=bind -w=/amazon-freertos --entrypoint=/amazon-freertos/entrypoint.sh espressif/idf:release-v4.2 cmake -S . -B build -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-e
sp32 -GNinja
docker run --rm --mount src="$(pwd)",target=/amazon-freertos,type=bind -w=/amazon-freertos --entrypoint=/amazon-freertos/entrypoint.sh espressif/idf:release-v4.2 idf.py build
```
The contents of `entrypoint.sh` is:
```
#!/usr/bin/env bash
set -e
. vendors/espressif/esp-idf/export.sh
exec "$@"
```

Johnnyjax, 2021-05-12:
hello @lundinc2. Thanks for your reply. When I ran this command - docker run --rm --mount src="$(pwd)",target=/amazon-freertos,type=bind -w=/amazon-freertos --entrypoint=/amazon-freertos/entrypoint.sh espressif/idf:release-v4.2 cmake -S . -B build -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-e
sp32 -GNinja.
I had this error

Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf4.2_py3.6_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.16.4/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
  /opt/esp/python_env/idf4.2_py3.6_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

CMake Warning at freertos/vendors/espressif/esp-idf/tools/cmake/idf.cmake:18 (message):
  IDF_PATH environment variable is different from inferred IDF_PATH.

                              Check if your project's top-level CMakeLists.txt includes the right
                              CMake files. Environment IDF_PATH will be used for the build:
                              /opt/esp/idf
Call Stack (most recent call first):
  CMakeLists.txt:4 (include)

-- Submodule update
fatal: not a git repository: /amazon-freertos/freertos/../.git/modules/freertos
CMake Error at freertos/tools/cmake/afr.cmake:45 (message):
  git submodule update --init failed with 128, please checkout submodules
Call Stack (most recent call first):
  freertos/CMakeLists.txt:14 (include)

-- Configuring incomplete, errors occurred!
See also "/amazon-freertos/build/CMakeFiles/CMakeOutput.log".

I also tried setting the environment variable for my esp-idf path in the docker run command by using this command - docker run --env IDF_PATH=/amazon-freertos/freertos/vendors/espressif/esp-idf --rm --mount src="${pwd}",target=/amazon-freertos,type=bind -w=/amazon-freertos --entrypoint=/amazon-freertos/entrypoint.sh espressif/idf:release-v4.2 cmake -S . -B build -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-e

and got this error

Adding ESP-IDF tools to PATH...
fatal: not a git repository: /amazon-freertos/freertos/vendors/espressif/esp-idf/../../../../.git/modules/freertos/modules/vendors/espressif/esp-idf
WARNING: Git describe was unsuccessul: Command '['git', 'describe']' returned non-zero exit status 128.
Using Python interpreter in /opt/esp/python_env/idf4.2_py3.6_env/bin/python
Checking if Python packages are up to date...
Python requirements from /amazon-freertos/freertos/vendors/espressif/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /amazon-freertos/freertos/vendors/espressif/esp-idf/components/esptool_py/esptool
  /amazon-freertos/freertos/vendors/espressif/esp-idf/components/espcoredump
  /amazon-freertos/freertos/vendors/espressif/esp-idf/components/partition_table
  /amazon-freertos/freertos/vendors/espressif/esp-idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.16.4/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
  /opt/esp/python_env/idf4.2_py3.6_env/bin
  /amazon-freertos/freertos/vendors/espressif/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

-- Component directory /amazon-freertos/freertos/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- Component directory /amazon-freertos/freertos/vendors/espressif/esp-idf/components/unity does not contain a CMakeLists.txt file. No component will be added
fatal: not a git repository: /amazon-freertos/freertos/vendors/espressif/esp-idf/../../../../.git/modules/freertos/modules/vendors/espressif/esp-idf
-- Submodule update
fatal: not a git repository: /amazon-freertos/freertos/../.git/modules/freertos
CMake Error at freertos/tools/cmake/afr.cmake:45 (message):
  git submodule update --init failed with 128, please checkout submodules
Call Stack (most recent call first):
  freertos/CMakeLists.txt:14 (include)

-- Configuring incomplete, errors occurred!
See also "/amazon-freertos/build/CMakeFiles/CMakeOutput.log".

lundinc2, 2021-05-12:
Have you run `git submodule update --init --recursive` or `git clone --recurse-submodules`?

muneebahmed10, 2021-10-20:
Hi @Johnnyjax,

We apologize for the late reply. If you are still experiencing this issue, these lines seem to be indicative of your problem:
> fatal: not a git repository: /amazon-freertos/freertos/vendors/espressif/esp-idf/../../../../.git/modules/freertos/modules/vendors/espressif/esp-idf

> fatal: not a git repository: /amazon-freertos/freertos/../.git/modules/freertos

I'm not sure where the `freertos` part of the path is coming from. I've tried the same commands as @lundinc2 [above](https://github.com/aws/amazon-freertos/issues/3143#issuecomment-839316109) and my paths looked like this:
```
Python requirements from /amazon-freertos/vendors/espressif/esp-idf/requirements.txt are satisfied.
```
I suspect either that you are not be running those commands from the root of the repository, you might not be working with the latest `main` branch, or your git repository became corrupted somehow. Can you try a fresh clone? I.e.

```shell
git clone https://github.com/aws/amazon-freertos.git
cd amazon-freertos
git submodule update --init --recursive
echo '#!/usr/bin/env bash\nset -e\n. vendors/espressif/esp-idf/export.sh\nexec "$@"' > entrypoint.sh
chmod +x entrypoint.sh
docker run --rm --mount src="$(pwd)",target=/amazon-freertos,type=bind -w=/amazon-freertos --entrypoint=/amazon-freertos/entrypoint.sh espressif/idf:release-v4.2 cmake -S . -B build -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja
```

Note that, like @lundinc2, I also encountered build issues when trying to run `docker run --rm --mount src="$(pwd)",target=/amazon-freertos,type=bind -w=/amazon-freertos --entrypoint=/amazon-freertos/entrypoint.sh espressif/idf:release-v4.2 idf.py build`, but the above shell commands should be enough to get you past the CMake configuration step.