aws/amazon-freertos #3144: [General] ESP-IDF compilation error when using freertos in custom project.

blakjak44, 2021-05-09:
**Briefly summarize the issue being raised**
Compilation was working fine earlier this year but since switching to ESP-IDF v4.2 and updating to the newest freertos version I am getting the following compilation error:

`CMake Error at freertos/vendors/espressif/boards/esp32/CMakeLists.txt:455 (add_executable):
  add_executable cannot create target "dev_app" because another target with
  the same name already exists.  The existing target is an executable created
  in source directory "~/afr-dev".`

**Describe the desired outcome**
I expect the project to compile so that I can flash the board.

**System information**
- Hardware board: ESP_32
- IDE used: No IDE, iTerm2
- Operating System: MacOS
- Code version: 202012.00-292-gda1da0611
- Project/Demo: Basic custom application from "Getting Started" guide
- Extensa version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

**References**
Compilation Command:
`./freertos/vendors/espressif/esp-idf/tools/idf.py -DCOMPILER=xtensa-esp32 build`

Compilation Log:
```
Setting IDF_PATH environment variable: /Users/blakjak/projects/personal/scratch/afr-dev/freertos/vendors/espressif/esp-idf
Executing action: all (aliases: build)
Running cmake in directory /Users/blakjak/projects/personal/scratch/afr-dev/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCOMPILER=xtensa-esp32 -DCCACHE_ENABLE=0 /Users/blakjak/projects/personal/scratch/afr-dev"...
-- The C compiler identification is AppleClang 11.0.3.11030032
-- The CXX compiler identification is AppleClang 11.0.3.11030032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.24.1 (Apple Git-126)")
-- Submodule update
Skipping submodule '../../../libraries/abstractions/backoff_algorithm/'
Skipping submodule '../../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
WARNING: IDF_PATH environment variable is not cleared.
[... 494 lines cut ...]
    system_event_t evt;

    if( eNetworkEvent == eNetworkUp )
    {
        /* Print out the network configuration, which may have come from a DHCP
         * server. */
        FreeRTOS_GetAddressConfiguration(
            &ulIPAddress,
            &ulNetMask,
            &ulGatewayAddress,
            &ulDNSServerAddress );

        evt.event_id = SYSTEM_EVENT_STA_GOT_IP;
        evt.event_info.got_ip.ip_changed = true;
        evt.event_info.got_ip.ip_info.ip.addr = ulIPAddress;
        evt.event_info.got_ip.ip_info.netmask.addr = ulNetMask;
        evt.event_info.got_ip.ip_info.gw.addr = ulGatewayAddress;
        esp_event_send( &evt );
    }
}
#endif
```

**Additional context**
If I run the identical compilation command inside freertos to build the demo project, the project compiles fine. This is only when building a custom project that consumes freertos.

Johnnyjax, 2021-05-11:
I think I've came across this issue when migrating to esp-idf 4.2 too. Check out the cmakelists file in [this](https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2) repository. It should help with fixing the issue

pvyawaha, 2021-05-17:
Hello , 

Were you able to fix the build issue?

blakjak44, 2021-05-22:
Just got around to this now. I tried @Johnnyjax's suggestion and tried copying the `CMakeList.txt` file from that repo but still got a compilation error. I just ran the test quickly and only modified a couple lines so it's possible that I need to configure it properly. I'm going to try just cloning that repo and building from that when I have chance.

I will say that I didn't get the error `add_executable cannot create target "dev_app" because another target with
  the same name already exists`, but that's likely because `add_executable` call has been removed.

blakjak44, 2021-05-23:
@pvyawaha I was able to get the build to work using the example repo mentioned by @Johnnyjax. Considering the cmakelists deviates quite a bit for idf v4.2, the docs in the Getting Started guide should be updated.

aggarw13, 2021-05-24:
@blakjak44, great that you have been able to get the custom CMake project building. Thank you for notifying about the documentation issues in the guide. We will look to update the instructions soon.

closed 2021-05-24 by aggarw13