aws/amazon-freertos #3145: NVS Encryption Fail

SimionMogildea, 2021-05-10:
**Briefly summarize the issue being raised**
NVS Encryption not working. Problably a bug or an incomplete procedure documentation.

**Describe the desired outcome**
The NVS keys CRC check fails. In particular the `nvs_flash_read_security_cfg()` function call is giving an `ESP_ERR_NVS_CORRUPT_KEY_PART` error.    
Am I missing something or it's an actual issue?   

The returned log messages of my app are shown here:
```
I (284) boot: Checking flash encryption...
I (284) flash_encrypt: flash encryption is enabled (2 plaintext flashes left)
...
W (399) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (401) spi_flash: detected chip: generic
I (401) spi_flash: flash io: dio
I (405) cpu_start: Starting scheduler on PRO CPU.
E (412) nvs: Failed to read nvs keys (rc=0x1117)
E (412) main: Failed to initialize nvs (rc=0x1117). Halting.
```

**System information**
- Hardware board? ESP32-WROOM-32E
- IDE used: VSCode
- Operating System: Windows 10
- Version of FreeRTOS: 202012.00-333-g39e6c6a0c
- Project: NVS Encryption

**References**
I created a demo project (https://github.com/SimionMogildea/AmazonFreertosTest) that uses `amazon-freertos` repo and highligts the reported issue. 

**Code/Steps to reproduce the current status**
First of all I enabled the Flash Encryption option as shown in the main Espressif guide (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html). The Flash encryption functionality seems to work as intended but as soon as I try to enable the Encryption for the NVS partition (as shown https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html) there is where the error occures.

Here are the steps I used to reproduce the behavior:
1. Clone the example code:
      - `git clone https://github.com/SimionMogildea/AmazonFreertosTest.git --recurse-submodules`
1. Configure project: 
      - `cmake -S . -B build -DBOARD=esp32_devkitc -DCMAKE_TOOLCHAIN_FILE='amazon-freertos/tools/cmake/toolchains/xtensa-esp32.cmake' -G Ninja`
2. Build project: 
      - `cmake --build build`
1. Generare encrypted NVS partition: 
      - `python nvs_partition_gen.py encrypt nvs.csv encrypted_nvs.bin 0x4000 --keygen --keyfile nvs_keys.bin`
1. Flash NVS partition: 
      - `esptool.py -p COMxx --before default_reset --after no_reset write_flash 0xa000 encrypted_nvs.bin`
1. Flash NVS Keys partition: 
      - `esptool.py -p COMxx --before default_reset --after no_reset write_flash 0x320000 nvs_keys.bin`
2. Flash app: 
      - `python ./amazon-freertos/vendors/espressif/esp-idf/tools/idf.py encrypted-flash -p COMxx -b 115200 monitor`

**Additional context**
* Partition table (AmazonFreertosTest/partition-table.csv):
```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,       ,  0x4000,
otadata,  data, ota,       ,  0x2000,
phy_init, data, phy,       ,  0x1000,
factory,  0,    0,         , 1M,
ota_0,    0,    ota_0,    , 1M,
ota_1,    0,    ota_1,    , 1M,
nvs_key,  data, nvs_keys,         , 0x1000, encrypted
```

* Nvs configuration file (AmazonFreertosTest/tools/nvs_partition_generator/nvs.csv):
```
key,type,encoding,value
PARAM_1,data,string,123123
TEST_FILE,file,string,./files/test.txt
```

* NVS partition initialization:
      - See https://github.com/SimionMogildea/AmazonFreertosTest/blob/master/main.c

* Efuse summary:
      - `espefuse.py --port COMxx summary` 
<details>
<summary>Details</summary>
<p>

```
Detecting chip type... ESP32
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1128 R/W (0b00100)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 50 R/W (0x32)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 128 R/W (0x0080)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 1 R/W (0x1)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = 40:f5:20:51:ff:20 (CRC 0x03 OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 3 R/W (0x03)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 7 R/W (0b0000111)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

</p>
</details>

mahavirj, 2021-05-11:
@SimionMogildea 

> E (412) nvs: Failed to read nvs keys (rc=0x1117)

This error indicates that "nvs_keys" partition is corrupted (error code `ESP_ERR_NVS_CORRUPT_KEY_PART`)

This may happen if you are flashing this partition post flash encryption is enabled on device. In that case, this partition will not be encrypted on (subsequent) device reset and hence contents will be treated as garbage (as its labelled with "encrypted" flag in partition table).

To overcome this:

1. Either you could try to flash this partition along with bootloader, application (single command) and then let bootloader perform encryption sequence.
2. Or you may keep this partition empty (erased) and allow encryption keys to be generated dynamically. Please see code snippet below for that:

https://github.com/aws/amazon-freertos/blob/cf36d8d31ccf4d7d072df67e57ea643df2b19616/vendors/espressif/boards/ports/pkcs11/core_pkcs11_pal.c#L80-L87

SimionMogildea, 2021-05-11:
Thank you for your reply @mahavirj!

> 1. Either you could try to flash this partition along with bootloader, application (single command) and then let bootloader perform encryption sequence.

Can you please tell me how can I do this? I'm not finding any single command I can flash multiple partitions with.    

I was using `idf.py encrypted-flash` but I don't really understand what partitions does it really flash and if there is a way to specify additional partition to flash along with bootloader and application binary.

AdityaHPatwardhan, 2021-05-11:
Hi @SimionMogildea Can you please erasing the flash once with command: `idf.py erase_flash`. I think that will solve your issue by keeping the key partition empty as @mahavirj has suggested.

mahavirj, 2021-05-12:
@SimionMogildea 

> I was using idf.py encrypted-flash but I don't really understand what partitions does it really flash

```
$ idf.py help
  encrypted-app-flash    Flash the encrypted app only.
  encrypted-flash        Flash the encrypted project.
```
 
Here project refers to core artifacts which includes "bootloader", "partition-table", and "application".

Again `idf.py` is wrapper script only, underlying it uses `esptool` to program flash on device. You can use command similar to following to flash multiple (custom) partitions in single go:

```
$ $IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --
before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size
detect 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table.bin 0x20000 build/app.bin
<nvs_keys_offset> encryption_keys.bin <nvs_partition_offset> nvs_partition.bin
```

If you add additional `--encrypt` argument in above command, then all partitions will be written in encrypted manner based on flash encryption settings on device. For more info, please look at `esptool.py write_flash --help`

Hope this helps!

SimionMogildea, 2021-05-12:
Thank you @mahavirj
I managed to solve my problem by using the esptool command you suggested.    
Here is the complete guide on how to correctly flash the partitions with NVS Encryption enabled:

1. Generate the nvs and `nvs_keys` partition:
      - `python nvs_partition_gen.py encrypt nvs.csv encrypted_nvs.bin 0x6000 --keygen --keyfile nvs_keys.bin`

2. Flash and encrypt `bootloader`, `partition_table`, `application`, `nvs_key` and other partitions (e.g. ota) to encrypt:
      - `python amazon-freertos\vendors\espressif\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 --port COMxx --baud 115200 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 build/bootloader/bootloader.bin 0x9000 build/partition_table/partition-table.bin 0x20000 build/app.bin 0x320000 tools/nvs_partition_generator/keys/nvs_keys.bin --encrypt`

3. Flash (no encrypt) the already encrypted nvs partition (`encrypted_nvs`):
      - `python amazon-freertos\vendors\espressif\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 --port COMxx --baud 115200 --before default_reset --after no_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0a000 tools/nvs_partition_generator/encrypted_nvs.bin`

4. Monitor and enjoy!

The reported offsets of 0x1000 `bootloader`, 0x9000 `partition_table`, 0x20000 `application` partitions can be read from `build/flasher_args`. The ones of `nvs_keys` and `nvs` can be discovered either by inspecting the esp32 serial output (where the partitions are printed out) or by using the gen_esp32part.py utility on the project's partition image (e.g. gen_esp32part.py build/partition_table/partition-table.bin).

Hope this can be useful to someone else too.

SimionMogildea, 2021-05-13:
I'm closing this issue for now. If something else is needed feel free to reopen it

closed 2021-05-13 by SimionMogildea