aws/amazon-freertos #3151: ESP32 BLE WIFI provisioning Demo is not able to connect back to STA (ESP32)->WiFi AP (router) once router rebooted

pintu-sakariya, 2021-05-13:
**Briefly summarize the issue being raised**
I am testing [BLE_WIFI_PROVISIONING](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-wifi)  provisioning demo as mentioned in the configuration, I have changed IOT_BLE_ENABLE_WIFI_PROVISIONING = 1.  
I  provision WiFi over BLE. Application is able to connect with a router and able to connect with AWS IoT.  
**ISSUE**
I have restarted my router and after that ESP32 is not able to connect with a router that is provisioned previously. The network is switch to BLE. 

**Describe the desired outcome**
I assume that once WiFi is provisioned over BLE. It should try to connect with that network instead of switching to BLE once the router (AP) restarted.
So either it should have some configuration like if WiFi credential already stored in Flash it should continue trying to connect with that or wait for new WiFi provisioning over BLE. Even after reboot device it shoud start with provision WiFi instead starting BLE connection.

**System information**
- Hardware board: [ ESP32 esp32_devkitc ]
- IDE used: [ ESP-IDF v4.2-15-g6abb2ca33 ]
- Operating System: [ Linux ]
- Code version: (202012.00-312-gdf48dbfa2)
- Project/Demo: [ IOT_BLE_ENABLE_WIFI_PROVISIONING + custom application (coreMQTT-Agent) ]
- If your project is a custom application, please add the relevant code snippet(s) in the section titled `"Code/Steps to reproduce the current status"`.

**References**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7728
ho 0 tail 12 room 4
load:0x40078000,len:13724
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-15-g6abb2ca33-dirty 2nd stage bootloader
I (30) boot: compile time 13:51:20
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (79) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (86) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (94) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (101) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (109) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (116) boot: End of partition table
I (121) boot: ota rollback check done
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x305a0 (198048) map
I (217) esp_image: segment 1: paddr=0x000505c8 vaddr=0x3ffbdb60 size=0x03ce8 ( 15592) load
I (223) esp_image: segment 2: paddr=0x000542b8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at amazon-freertos/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1851

I (224) esp_image: segment 3: paddr=0x000546c4 vaddr=0x40080404 size=0x0b954 ( 47444) load
I (253) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xc7474 (816244) map
0x400d0020: _stext at ??:?

I (564) esp_image: segment 5: paddr=0x0012749c vaddr=0x4008bd58 size=0x128d8 ( 75992) load
0x4008bd58: r_lld_evt_restart at ??:?

I (614) boot: Loaded app from partition at offset 0x20000
I (614) boot: Disabling RNG early entropy source...
I (615) cpu_start: Pro cpu up.
I (618) cpu_start: Application information:
I (623) cpu_start: Project name:     amazon-freertos
I (629) cpu_start: App version:      1
I (633) cpu_start: Compile time:     May 12 2021 18:41:08
I (639) cpu_start: ELF file SHA256:  c805e709fee6f461...
[... 317 lines cut ...]

116 12871 [iot_thread] [INFO ][DEMO][128710] _onNetworkStateChangeCallback:99

117 12871 [iot_thread] [INFO ][DEMO][128710] _onNetworkStateChangeCallback:131

118 12871 [iot_thread] [INFO ][DEMO][128710] _onNetworkStateChangeCallback:144

119 12871 [iot_thread] [INFO ][DEMO][128710] AwsIotNetworkManager_EnableNetwork:994

120 12871 [iot_thread] [INFO ][DEMO][128710] AwsIotNetworkManager_EnableNetwork:994

121 12871 [iot_thread] [INFO ][DEMO][128710] _bleEnable:330

122 12871 [iot_thread] [INFO ][DEMO][128710] _bleRegisterUnregisterCb:274

I (129459) BTDM_INIT: BT controller compile version [3723d5b]
failed to configure restored IRK
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
123 12910 [iot_thread] [INFO ][DEMO][129100] _getConnectedNetworkForDemo:55

124 12910 [iot_thread] [INFO ][DEMO][129100] AwsIotNetworkManager_GetConnectedNetworks:975

125 12910 [iot_thread] [INFO ][DEMO][129100] prvWiFiNetworkStateChangeCallback:82

126 12911 [WiFiConnectTask] [INFO ][DEMO][129110] prvWiFiConnectTask:103

ravibhagavandas, 2021-05-20:
Hi @pintu-sakariya 

From the last line of the the log, it seems the [task](https://github.com/aws/amazon-freertos/blob/main/demos/wifi_provisioning/aws_wifi_connect_task.c#L87) for wifi reconnection is triggered correctly, when the WIFI Access point is disconnected. The task will connect to the provisioned networks on the device.
`126 12911 [WiFiConnectTask] [INFO ][DEMO][129110] prvWiFiConnectTask:103`
Could you share additional logs after the wifi connect task is triggered ?  Also from the logs above I don't see the case where the WiFi does not connect to a stored network after reboot. Could you share the logs of reboot as well ?

pintu-sakariya, 2021-05-21:
Hi @ravibhagavandas,

I have debug the issue and figure out that code under "#if IOT_BLE_ENABLE_WIFI_PROVISIONING == 1" is not executing,

File: demos/wifi_provisioning/aws_wifi_connect_task.c
Function: void prvWiFiConnectTask( void * pvParams )

Issue: Even though IOT_BLE_ENABLE_WIFI_PROVISIONING  is set to 1 at vendors/espressif/boards/esp32/aws_demos/config_files/iot_ble_config.h, On calling to prvWiFiConnectTask() below part is not executing,
                 #if IOT_BLE_ENABLE_WIFI_PROVISIONING == 1
                     ulNumNetworks += IotBleWifiProv_GetNumNetworks();
                 #endif 

I have followed [ble-demo->WiFi Provisioning](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-wifi) but it seems this value is not resolved in demos/wifi_provisioning/aws_wifi_connect_task.c for ESP32.
[image]

To test this i just removed #if IOT_BLE_ENABLE_WIFI_PROVISIONING == 1 from  demos/wifi_provisioning/aws_wifi_connect_task.c and it is working as expected.

Case 1: It is connecting to provision WiFi after router reboot.
Case 2: It is connecting to provision WiFi after ESP32 reboot.

It seems like need to update demo documentation or need to set IOT_BLE_ENABLE_WIFI_PROVISIONING  at proper place.

Thanks,
Pintu

ravibhagavandas, 2021-05-22:
Hi @pintu-sakariya 

Thanks for catching that. I agree there is an inconsistency between the doc and where the flag needs to be placed. We will fix this soon.

closed 2021-05-22 by aggarw13