aws/amazon-freertos #3153: Failed to toggle BLE

Raghav3107, 2021-05-14:
Hello,

I m using [this repository](https://github.com/Raghav3107/aws-freertos-workspace). When esp start initially, BLE is not getting on. 
As per my knowledge, the parameter which enables the wifi and ble is already defined in [network_config](https://github.com/Raghav3107/aws-freertos-workspace/blob/a35cd5b568975763d2d4cc53e8877019d4e127a3/amazon-freertos-configs/aws_iot_network_config.h#L55) files.
But still, it failed to get on.

**Log:**
```
E (42) PKCS11: failed nvs get file size 4354 0
1 4 [main] Write certificate...
2 9 [main] [INFO ][DEMO][90] ---------STARTING DEMO---------

3 9 [main] [INFO ][INIT][90] SDK successfully initialized.
4 9 [main] [INFO ][DEMO][90] AwsIotNetworkManager_Init
5 9 [main] [INFO ][DEMO][90] IotSemaphore_Create( &demoNetworkSemaphore, 0, 1 ) != true )
6 10 [main] [INFO ][DEMO][100] AwsIotNetworkManager_SubscribeForStateChange
7 11 [main] [INFO ][DEMO][110] AwsIotNetworkManager_EnableNetwork
Ble on -> 346 
BLE_GAP-> BLE on fun 
BLE___FALSE 
 BLE____BTInterface.pBTInterface->pxBtManagerInit( &_BTManagerCb ) 
 BLE_GAP-Status -> 1 
8 13 [main] [ERROR][DEMO][130] Failed to toggle BLE on.                    //failed 
I (162) wifi:wifi driver task: 3ffb9854, prio:23, stack:3584, core=0
I (162) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (172) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (182) wifi:wifi firmware version: 1afdae0
```

ravibhagavandas, 2021-05-18:
Hi @Raghav3107 

Looks like the error is returned from Nimble which is the underlying BLE stack for ESP32. Could you enable ESP32 logging and send over the results.

To enable log: Run `make menuconfig` in build folder then goto `Component config` - > `Log output` -> `Default log verbosity` ->  `Select Debug` . Save configuration and then rebuild.

Raghav3107, 2021-05-20:
Hello @ravibagavandas
Sorry for the late reply.

**Here is the log:**
```
I (73) boot: Chip Revision: 1
I (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3.2-241-g1d2d93acd 2nd stage bootloader
I (41) boot: compile time 12:28:21
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (100) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (108) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (115) boot: End of partition table
I (119) boot: ota rollback check done
I (124) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x3049c (197788) map
I (209) esp_image: segment 1: paddr=0x000504c4 vaddr=0x3ffbdb60 size=0x03844 ( 14404) load
I (215) esp_image: segment 2: paddr=0x00053d10 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (217) esp_image: segment 3: paddr=0x00054118 vaddr=0x40080400 size=0x0bef8 ( 48888) load
I (245) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xc4180 (803200) map
0x400d0018: _flash_cache_start at ??:?

I (528) esp_image: segment 5: paddr=0x001241a0 vaddr=0x4008c2f8 size=0x0fbac ( 64428) load
0x4008c2f8: bootloader_flash_cs_timing_config at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/vendors/espressif/esp-idf/components/bootloader_support/src/bootloader_flash_config.c:40

I (571) boot: Loaded app from partition at offset 0x20000
I (571) boot: Disabling RNG early entropy source...
I (571) cpu_start: Pro cpu up.
I (575) cpu_start: Application information:
I (580) cpu_start: Project name:     esp-idf
I (585) cpu_start: App version:      222eced
I (589) cpu_start: Compile time:     May 19 2021 12:28:18
I (596) cpu_start: ELF file SHA256:  9620c664d3981266...
I (602) cpu_start: ESP-IDF:          v3.3.2-241-g1d2d93acd
I (608) cpu_start: Single core mode
D (612) memory_layout: Checking 9 reserved memory ranges:
D (617) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (624) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (630) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (637) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (643) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (649) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffceef8
D (656) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (662) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (669) memory_layout: Reserved memory range 0x40080000 - 0x4009bea4
0x40080000: _WindowOverflow4 at /home/horsemann/Desktop/WorkSpace/aws_latest_repository/personal-box-freertos-demo/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

D (675) memory_layout: Building list of available memory regions:
D (681) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
D (688) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000
D (695) memory_layout: Available memory region 0x3ffb9a20 - 0x3ffbdb28
D (701) memory_layout: Available memory region 0x3ffbdb5c - 0x3ffbdb60
D (708) memory_layout: Available memory region 0x3ffceef8 - 0x3ffd0000
D (714) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (721) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (727) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (734) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (741) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (747) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
[... 230 lines cut ...]
D (1172) efuse: coding scheme 0
D (1172) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1182) efuse: coding scheme 0
D (1182) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1192) efuse: coding scheme 0
D (1192) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1202) efuse: coding scheme 0
D (1202) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (1212) nvs: nvs_get_str_or_blob cal_data
D (1212) nvs: nvs_close 22
D (1222) efuse: coding scheme 0
D (1222) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (1232) efuse: coding scheme 0
D (1232) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (1242) efuse: coding scheme 0
D (1242) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (1252) efuse: coding scheme 0
D (1252) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (1262) efuse: coding scheme 0
D (1262) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (1272) efuse: coding scheme 0
D (1272) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (1282) efuse: coding scheme 0
D (1282) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 
```

ravibhagavandas, 2021-05-21:
Hi @Raghav3107 

Looking at the `main()` function in your repository, it releases the RAM preallocated for bluetooth classic and BLE. For NimBLE stack to function, BLE memory still needs to be preallocated.

Removing the following line should fix that issue: https://github.com/Raghav3107/aws-freertos-workspace/blob/master/src/main.c#L154

Raghav3107, 2021-05-21:
Hello @ravibhagavandas
I tried to remove the following and ble getting started but another issue is coming in network configuration.
After this other network configuration is not getting started.

```
I (2204) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (2294) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (2924) event: sta ip: 192.168.43.99, mask: 255.255.255.0, gw: 192.168.43.1
I (2924) WIFI: SYSTEM_EVENT_STA_GOT_IP
8 258 [eventTask] [INFO ][DEMO][2580] Connected to WiFi access point, ip address: 192.168.43.99.
GAP procedure initiated: stop advertising.
9 259 [main] [ERROR][DEMO][2590] Failed to initialize all the networks configured for the device.
GAP procedure initiated: stop advertising.
10 261 [iot_thread] [DEBUG][TASKPOOL][2610] Worker thread exiting because shutdown condition was set.
11 262 [iot_thread] [ERROR][DEMO][2620] Failed to create subscription task for network: 2, state: 0, error: 4
12 262 [iot_thread] [DEBUG][TASKPOOL][2620] Worker thread exiting because shutdown condition was set.
13 263 [main] [INFO ][INIT][2630] SDK cleanup done.
14 264 [main] [ERROR][DEMO][2640] Failed to initialize the demo. exiting...
15 264 [main] [ERROR][DEMO][2640] Error running demo.
16 265 [main] [INFO ][DEMO][2650] -------DEMO FINISHED-------

```

ravibhagavandas, 2021-05-22:
I think the issue is with this line of code: https://github.com/Raghav3107/aws-freertos-workspace/blob/master/src/demoRunner.c#L209
The line tries to initialize both WiFi and BLE networks, and since both networks are initialized successfully, API returns a mask as both networks enabled `( AWSIOT_NETWORK_TYPE_WIFI | AWSIOT_NETWORK_TYPE_BLE )`
The return values can be checked as follows:

```
 uint32_t enabledNetworks = AwsIotNetworkManager_EnableNetwork( configENABLED_NETWORKS );
 if( enabledNetworks == configENABLED_NETWORKS )
 {
       // All networks initialized.
 }
 else if( ( enabledNetworks & AWSIOT_NETWORK_TYPE_WIFI )  == 0 )
 {
     // WiFi not initialized.
 }
 else if( ( enabledNetworks & AWSIOT_NETWORK_TYPE_BLE )  == 0 )
 {
     // BLE not initialized.
 }
 
```

Please let us know if you have questions.

Raghav3107, 2021-05-22:
Thank you so much, @ravibhagavandas 
It got it. I changed  `if (AwsIotNetworkManager_EnableNetwork(configENABLED_NETWORKS) != AWSIOT_NETWORK_TYPE_WIFI)` to `if (AwsIotNetworkManager_EnableNetwork(configENABLED_NETWORKS) != configENABLED_NETWORKS).`

closed 2021-05-22 by Raghav3107