aws/amazon-freertos #3156: [BUG] ESP32 BLE Connection disconnects after connection since freertos update

thomasklaush, 2021-05-17:
**Describe the bug**
I used as basis the BLE GATT demo and modified it.
Since pulling from master, BLE connection is disconnected after connection.
I assume the change of the esp-idf to 4.2.1 is the reason, but cannot find the bug.

**System information**
- Hardware board: ESP32 WROOM
- Update from 39e6c6a0c6703fac8d004cd1474ddb31a81919f7 to current 80e473913f873406c4d0ad833511e0782f22fa1b

**Expected behavior**
BLE Connection should work as before the update

**Screenshots or console output**

73 6298 [ble] [INFO ][DEMO][62980] BLE Connected to remote device, connId = 0

GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
74 6872 [ble] [INFO ][DEMO][68720]  Disconnected from BLE device. Stopping the counter update 

75 6872 [ble] [INFO ][DEMO][68720] BLE disconnected with remote device, connId = 0

ravibhagavandas, 2021-05-18:
Hi @thomasklaush 
Could you provide following details:
1. What type of device is used as BLE central, is it an IOS or Android and which version of freertos  IOS/Android sdk is used?
2. What type of connection is made ? Is it just works or pairing with numeric comparison ? Is bonding enable for the connection ?

thomasklaush, 2021-05-18:
Hi @ravibhagavandas 
1. I use a flutter app on android with the flutter_ble_lib.
    I don't use the kit, since it's not on flutter, I'm currently programming it myself.
2. It's connecting BLE, nothing else.
   No pairing with the device.
   It's the most basic connection, everything deactivated.
   #define IOT_BLE_ENABLE_NUMERIC_COMPARISON        ( 0 )
   #define IOT_BLE_ENABLE_SECURE_CONNECTION         ( 0 )
   #define IOT_BLE_INPUT_OUTPUT                     ( eBTIONone )
   #define IOT_BLE_ENCRYPTION_REQUIRED               ( 0 )

shubhamkulkarni97, 2021-05-19:
@thomasklaush Thanks for reporting this issue. Please try changes in https://github.com/aws/amazon-freertos/pull/3159

thomasklaush, 2021-05-19:
@shubhamkulkarni97 Thanks for the fast change, your changes solved the issue.
I will do some further tests, but I don't assume a problem anymore.
Best regards,
Thomas

ravibhagavandas, 2021-05-21:
Thanks for the fix @shubhamkulkarni97 
The changes are merged to main. I am closing this issue. Please feel free to open an issue if you have additional questions.

closed 2021-05-21 by ravibhagavandas