aws/amazon-freertos #3163: How to change the latest release branch of Amazon freertos to use esp-idf v4.2 instead of v3.3

Johnnyjax, 2021-05-20:
I had been using the latest release branch of amazon-freertos but that uses the esp-idf v3.3 and I had to change to the main branch before I could use the esp-idf v4.2. Is there a way to still use the release branch but with esp-idf v4.2? There are some features of esp-Idf v4.2 that we need for our application that v3.3 doesn't have and we'll also like to use the release branch since it has the fully tested released source code instead of using the main branch. Thanks

aggarw13, 2021-05-21:
Hi @Johnnyjaxm, you can refer to [this PR](https://github.com/aws/amazon-freertos/pull/2893) that updated the ESP-IDF version from `v3.3` to `v4.2` on Dec 30 2020 which was close to the [last release](https://github.com/aws/amazon-freertos/tree/202012.00) made on Dec 15 2020.

shubhamkulkarni97, 2021-05-24:
@Johnnyjax Alternately, you can use [esp32-s2](https://github.com/aws/amazon-freertos/tree/esp32-s2) branch which supports ESP-IDF v4.2 on top of latest release. Please note that some fixes related to ESP-IDF v4.2 are merged in master branch. I recommend to cherry-pick these fixes in esp32-s2 branch.

Thanks,
Shubham

Johnnyjax, 2021-05-25:
Thanks a lot for your response @aggarw13 and @shubhamkulkarni97. I use the PR @aggarw13 suggested.

closed 2021-05-25 by Johnnyjax