aws/amazon-freertos #3169: Issue in BLE addvertisment Name.

Raghav3107, 2021-05-24:
Hello everyone,

I was trying to change the [BLE advertisement name](https://github.com/aws/amazon-freertos/blob/c2388cfcd27bb0efd7c21361f630dea1a11dcdfa/vendors/espressif/boards/esp32/aws_demos/config_files/iot_ble_config.h#L36) from `ESP32` to something like `"MY_BLE_NAME"`. But when I m changing the name the initial network configuration is getting the failed.
```
I (161) BTDM_INIT: BT controller compile version [9891bc6]
2 2 [iot_thread] [INFO ][INIT][20] SDK successfully initialized.
I (171) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (181) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (311) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
GAP procedure initiated: stop advertising.
IOT_BLE_ADD_CUSTOM_SERVICES -> IotBle_AddCustomServicesCb 
3 37 [iot_thread] [ERROR][DEMO][370] Failed to toggle BLE on.
I (521) wifi: wifi driver task: 3ffdccdc, prio:23, stack:3584, core=0
I (521) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (521) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (541) wifi: wifi firmware version: c94f3e6
I (541) wifi: config NVS flash: enabled
I (541) wifi: config nano formating: disabled
I (541) wifi: Init dynamic tx buffer num: 32
I (551) wifi: Init data frame dynamic rx buffer num: 16
I (551) wifi: Init management frame dynamic rx buffer num: 16
I (561) wifi: Init management short buffer num: 32
I (561) wifi: Init static rx buffer size: 1600
I (561) wifi: Init static rx buffer num: 10
I (571) wifi: Init dynamic rx buffer num: 16
No of saved  WIFI Profle:[ 0 ]
No of saved  WIFI Profle:[ 0 ]
4 44 [iot_thread] [ERROR][DEMO][430] Failed to intialize all the networks configured for the device.
5 45 [iot_thread] [INFO ][INIT][450] SDK cleanup done.
6 45 [iot_thread] [ERROR][DEMO][450] Failed to initialize the demo. exiting...
7 45 [iot_thread] [INFO ][DEMO][450] -------DEMO FINISHED-------
```
I searched for the solution and I find some [comment ](https://github.com/aws/amazon-freertos/blob/c2388cfcd27bb0efd7c21361f630dea1a11dcdfa/libraries/c_sdk/standard/ble/src/iot_ble_gap.c#L115) which say's "total available data size in the advertisement is 31 bytes. Parameters are chosen below such that the overall size does not exceed 31 bytes". And the struct store the name of ble. So, I thought when I m giving the big name to `IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE` it increases the advertisement size and it causes this error.

But after that, I found in [aws_test](https://github.com/aws/amazon-freertos/blob/c2388cfcd27bb0efd7c21361f630dea1a11dcdfa/vendors/espressif/boards/esp32/aws_tests/config_files/iot_ble_config.h#L36) where it uses other names which increase size.

So can we change the ble addv name as per our requirement?

Also, I want to create the dynamic name of ble adv. For that, I changed ` #define IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE    ESP32` to 

```
extern const char * getBLE_AdvName( void );
#define IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE   getBLE_AdvName() 
```
But it giving me the error 
```
CC build/amazon-freertos-common/libraries/c_sdk/standard/ble/src/services/wifi_provisioning/iot_ble_wifi_provisioning.o
/home/horsemann/Desktop/WorkSpace/IR/libraries/c_sdk/standard/ble/src/iot_ble_gap.c:138:18: error: initializer element is not constant
         .pvVal = ( void * ) IOT_BLE_DEVICE_COMPLETE_LOCAL_NAME
                  ^
/home/horsemann/Desktop/WorkSpace/IR/libraries/c_sdk/standard/ble/src/iot_ble_gap.c:138:18: note: (near initialization for '_deviceProperties[0].pvVal')
```
And the same thing I m doing to get the thing name that time I m not getting any error.

aggarw13, 2021-05-24:
Hi @Raghav3107, thank you for the detailed explanation of the issue you're facing. 

For the build issue you're facing with defining a longer value for `IOT_BLE_DEVICE_COMPLETE_LOCAL_NAME` configuration, the library will support using a longer name as well as larger value of `IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE` as long as you can update other members of the advertisement parameters, like UUIDs, (in the [`IotBleAdvertisementParams_t`](https://github.com/aws/amazon-freertos/blob/c2388cfcd27bb0efd7c21361f630dea1a11dcdfa/libraries/c_sdk/standard/ble/include/iot_ble.h#L50-L64) struct) to keep the total data to under 31 bytes. 
Can you whether your issue gets resolved after increasing the value of the `IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE` to a value larger than the default value of 4 bytes, and decreasing the UUID value length ([here](https://github.com/aws/amazon-freertos/blob/c2388cfcd27bb0efd7c21361f630dea1a11dcdfa/libraries/c_sdk/standard/ble/src/iot_ble_gap.c#L72-L76)) to use 16 bits instead of 128 bits?

Regarding the error that you see when defining as `IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE` as a function, the code will not currently support dynamic configuration of the name as the advertisement parameters are statically defined in the library [here](https://github.com/aws/amazon-freertos/blob/c2388cfcd27bb0efd7c21361f630dea1a11dcdfa/libraries/c_sdk/standard/ble/src/iot_ble_gap.c#L126) for the `IotBleAdvertisementParams_t ` struct that only takes an integer for the field where the `IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE` macro is assigned as the value.

Raghav3107, 2021-05-25:
Thanks, @aggarw13 I will try to do that.

One more thing, when pairing the device with android app I seen the advertisement name "ESP32" and MAC address of the Device.
[image]

I search from where its adding the MAC address of the device in advertisement parameter. But I did not find anything.
Can you tell me from where it adding the MAC address in advertisement parameter? and Can we send other thing instead of MAC address, dynamically? I think we should.

Thanks for your help and advices.

Raghav3107, 2021-05-30:
Hello 

I was trying Ble-Wifi Provision, in my [custom repository](https://github.com/Raghav3107/amazon-freertos-workspace-v202012). I added the parameter to use custom service callback in iot_ble_config.h
`#define IOT_BLE_ADD_CUSTOM_SERVICES                     (1)`

But some linking issue I'm getting at build time I added the file path in [CmakeLists.txt](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/blob/4d84217ca7e7b29efc59ff4ca23d22ea9d9e78e0/CMakeLists.txt#L18) and also [AFR::ble_wifi_provisioning](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/blob/4d84217ca7e7b29efc59ff4ca23d22ea9d9e78e0/CMakeLists.txt#L62).
**Can you tell me what I need to add?**

**Error Log:**
                 ^
```
[1007/1008] Linking CXX executable afr_workshop
FAILED: afr_workshop 
: && /home/horsemann/Desktop/WorkSpace/esp_toolchain/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls  -nostdlib   -Wl,--gc-sections -Wl,--cref -Wl,--Map=aws_demos.map -Wl,--undefined=uxTopUsedPriority CMakeFiles/afr_workshop.dir/src/bme680_sensor.c.obj CMakeFiles/afr_workshop.dir/src/demoRunner.c.obj CMakeFiles/afr_workshop.dir/src/main.c.obj CMakeFiles/afr_workshop.dir/src/mqtt_connection_manager.c.obj CMakeFiles/afr_workshop.dir/src/mqtt_pub_sub.c.obj CMakeFiles/afr_workshop.dir/src/ota_over_mqtt_demo.c.obj CMakeFiles/afr_workshop.dir/src/shadow_task.c.obj CMakeFiles/afr_workshop.dir/src/subscription_manager.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_bignum.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_mem.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_sha256.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_hardware.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_sha1.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_sha512.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/mbedtls_debug.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/wifi_provisioning/aws_wifi_connect_task.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/aws_demo_version.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/aws_demo.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/network_manager/aws_iot_network_manager.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/network_manager/aws_iot_demo_network.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/iot_demo_freertos.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/iot_demo_runner.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/libraries/freertos_plus/standard/utils/src/iot_system_init.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/ble/numeric_comparison/iot_ble_numericComparison.c.obj  -o afr_workshop  -Wl,--gc-sections  -Wl,--start-group  amazon-freertos/esp-idf/soc/libsoc.a  amazon-freertos/esp-idf/log/liblog.a  amazon-freertos/esp-idf/heap/libheap.a  amazon-freertos/esp-idf/freertos/libfreertos.a  amazon-freertos/esp-idf/vfs/libvfs.a  amazon-freertos/esp-idf/newlib/libnewlib.a  amazon-freertos/esp-idf/esp_ringbuf/libesp_ringbuf.a  amazon-freertos/esp-idf/driver/libdriver.a  amazon-freertos/esp-idf/esp_event/libesp_event.a  amazon-freertos/esp-idf/ethernet/libethernet.a  amazon-freertos/esp-idf/micro-ecc/libmicro-ecc.a  amazon-freertos/esp-idf/efuse/libefuse.a  amazon-freertos/esp-idf/bootloader_support/libbootloader_support.a  amazon-freertos/esp-idf/app_update/libapp_update.a  amazon-freertos/esp-idf/spi_flash/libspi_flash.a  amazon-freertos/esp-idf/nvs_flash/libnvs_flash.a  amazon-freertos/esp-idf/lwip/liblwip.a  amazon-freertos/esp-idf/tcpip_adapter/libtcpip_adapter.a  amazon-freertos/esp-idf/xtensa-debug-module/libxtensa-debug-module.a  amazon-freertos/esp-idf/app_trace/libapp_trace.a  amazon-freertos/esp-idf/pthread/libpthread.a  amazon-freertos/esp-idf/smartconfig_ack/libsmartconfig_ack.a  amazon-freertos/esp-idf/wpa_supplicant/libwpa_supplicant.a  amazon-freertos/esp-idf/espcoredump/libespcoredump.a  amazon-freertos/esp-idf/esp32/libesp32.a  amazon-freertos/esp-idf/cxx/libcxx.a  amazon-freertos/esp-idf/OTA/libOTA.a  amazon-freertos/esp-idf/bme680/libbme680.a  amazon-freertos/esp-idf/mqtt/libmqtt.a  amazon-freertos/esp-idf/nimble/libnimble.a  amazon-freertos/esp-idf/bt/libbt.a  amazon-freertos/esp-idf/console/libconsole.a  amazon-freertos/esp-idf/esp_adc_cal/libesp_adc_cal.a  amazon-freertos/esp-idf/expat/libexpat.a  amazon-freertos/esp-idf/wear_levelling/libwear_levelling.a  amazon-freertos/esp-idf/sdmmc/libsdmmc.a  amazon-freertos/esp-idf/fatfs/libfatfs.a  amazon-freertos/esp-idf/freemodbus/libfreemodbus.a  amazon-freertos/esp-idf/json/libjson.a  amazon-freertos/esp-idf/mdns/libmdns.a  amazon-freertos/esp-idf/openssl/libopenssl.a  amazon-freertos/esp-idf/protobuf-c/libprotobuf-c.a  amazon-freertos/esp-idf/spiffs/libspiffs.a  amazon-freertos/esp-idf/ulp/libulp.a  amazon-freertos/afr_core_mqtt.a  amazon-freertos/afr_backoff_algorithm.a  amazon-freertos/afr_core_json.a  amazon-freertos/afr_transport_interface_secure_sockets.a  amazon-freertos/afr_secure_sockets.a  amazon-freertos/afr_pkcs11_helpers.a  amazon-freertos/afr_mqtt_demo_helpers.a  amazon [... cut ...]
amazon-freertos/afr_ble.a(iot_ble_gap.c.obj):(.literal._startGATTServices+0x0): undefined reference to `IotBle_AddCustomServicesCb'
amazon-freertos/afr_ble.a(iot_ble_gap.c.obj): In function `_startGATTServices':
/home/horsemann/Desktop/WorkSpace/aws_latest_repository/amazon-freertos-workspace-v202012/build/../amazon-freertos/libraries/c_sdk/standard/ble/src/iot_ble_gap.c:960: undefined reference to `IotBle_AddCustomServicesCb'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

ravibhagavandas, 2021-06-01:
Hi @Raghav3107 

The flag `IOT_BLE_ADD_CUSTOM_SERVICES` is used if you are  creating a new custom GATT service for your use case. If the flag is enabled, you need to provide the implementation for `IotBle_AddCustomServicesCb ` hook,  with the init function of your custom GATT service. A demo  custom GATT service is provided in the repository [here](https://github.com/aws/amazon-freertos/blob/main/demos/ble/gatt_server/iot_ble_gatt_server_demo.c#L207).

From the  post it seems you need to only enable WiFi provisioning over BLE (the GATT service for this is already implemented in the repository). To enable WiFi provisioning over BLE you need to set `IOT_BLE_ENABLE_WIFI_PROVISIONING` and use the sample [here](https://github.com/aws/amazon-freertos/blob/main/demos/wifi_provisioning/aws_wifi_connect_task.c) to wait for a network to be provisioned.

Raghav3107, 2021-06-01:
Thanks, @ravibhagavandas  for your reply.

> you are creating a new custom GATT service for your use case

I m creating some new custom GATT service where I m reading the mac address and other data from nvs and sending it to the app.
This is working in an older version of `amazon-freertos(201912)` when I m running the program under the `aws_demos/application `folder.
But I m getting the linking issue on my [custom repository.](https://github.com/Raghav3107/amazon-freertos-workspace-v202012).
```
[1007/1008] Linking CXX executable afr_workshop
FAILED: afr_workshop 
: && /home/horsemann/Desktop/WorkSpace/esp_toolchain/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -mlongcalls  -nostdlib   -Wl,--gc-sections -Wl,--cref -Wl,--Map=aws_demos.map -Wl,--undefined=uxTopUsedPriority CMakeFiles/afr_workshop.dir/src/bme680_sensor.c.obj CMakeFiles/afr_workshop.dir/src/demoRunner.c.obj CMakeFiles/afr_workshop.dir/src/main.c.obj CMakeFiles/afr_workshop.dir/src/mqtt_connection_manager.c.obj CMakeFiles/afr_workshop.dir/src/mqtt_pub_sub.c.obj CMakeFiles/afr_workshop.dir/src/ota_over_mqtt_demo.c.obj CMakeFiles/afr_workshop.dir/src/shadow_task.c.obj CMakeFiles/afr_workshop.dir/src/subscription_manager.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_bignum.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_mem.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_sha256.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_hardware.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_sha1.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/esp_sha512.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/mbedtls_debug.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/wifi_provisioning/aws_wifi_connect_task.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/aws_demo_version.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/aws_demo.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/network_manager/aws_iot_network_manager.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/network_manager/aws_iot_demo_network.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/iot_demo_freertos.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/demo_runner/iot_demo_runner.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/libraries/freertos_plus/standard/utils/src/iot_system_init.c.obj CMakeFiles/afr_workshop.dir/amazon-freertos/demos/ble/numeric_comparison/iot_ble_numericComparison.c.obj  -o afr_workshop  -Wl,--gc-sections  -Wl,--start-group  amazon-freertos/esp-idf/soc/libsoc.a  amazon-freertos/esp-idf/log/liblog.a  amazon-freertos/esp-idf/heap/libheap.a  amazon-freertos/esp-idf/freertos/libfreertos.a  amazon-freertos/esp-idf/vfs/libvfs.a  amazon-freertos/esp-idf/newlib/libnewlib.a  amazon-freertos/esp-idf/esp_ringbuf/libesp_ringbuf.a  amazon-freertos/esp-idf/driver/libdriver.a  amazon-freertos/esp-idf/esp_event/libesp_event.a  amazon-freertos/esp-idf/ethernet/libethernet.a  amazon-freertos/esp-idf/micro-ecc/libmicro-ecc.a  amazon-freertos/esp-idf/efuse/libefuse.a  amazon-freertos/esp-idf/bootloader_support/libbootloader_support.a  amazon-freertos/esp-idf/app_update/libapp_update.a  amazon-freertos/esp-idf/spi_flash/libspi_flash.a  amazon-freertos/esp-idf/nvs_flash/libnvs_flash.a  amazon-freertos/esp-idf/lwip/liblwip.a  amazon-freertos/esp-idf/tcpip_adapter/libtcpip_adapter.a  amazon-freertos/esp-idf/xtensa-debug-module/libxtensa-debug-module.a  amazon-freertos/esp-idf/app_trace/libapp_trace.a  amazon-freertos/esp-idf/pthread/libpthread.a  amazon-freertos/esp-idf/smartconfig_ack/libsmartconfig_ack.a  amazon-freertos/esp-idf/wpa_supplicant/libwpa_supplicant.a  amazon-freertos/esp-idf/espcoredump/libespcoredump.a  amazon-freertos/esp-idf/esp32/libesp32.a  amazon-freertos/esp-idf/cxx/libcxx.a  amazon-freertos/esp-idf/OTA/libOTA.a  amazon-freertos/esp-idf/bme680/libbme680.a  amazon-freertos/esp-idf/mqtt/libmqtt.a  amazon-freertos/esp-idf/nimble/libnimble.a  amazon-freertos/esp-idf/bt/libbt.a  amazon-freertos/esp-idf/console/libconsole.a  amazon-freertos/esp-idf/esp_adc_cal/libesp_adc_cal.a  amazon-freertos/esp-idf/expat/libexpat.a  amazon-freertos/esp-idf/wear_levelling/libwear_levelling.a  amazon-freertos/esp-idf/sdmmc/libsdmmc.a  amazon-freertos/esp-idf/fatfs/libfatfs.a  amazon-freertos/esp-idf/freemodbus/libfreemodbus.a  amazon-freertos/esp-idf/json/libjson.a  amazon-freertos/esp-idf/mdns/libmdns.a  amazon-freertos/esp-idf/openssl/libopenssl.a  amazon-freertos/esp-idf/protobuf-c/libprotobuf-c.a  amazon-freertos/esp-idf/spiffs/libspiffs.a  amazon-freertos/esp-idf/ulp/libulp.a  amazon-freertos/afr_core_mqtt.a  amazon-freertos/afr_backoff_algorithm.a  amazon-freertos/afr_core_json.a  amazon-freertos/afr_transport_interface_secure_sockets.a  amazon-freertos/afr_secure_sockets.a  amazon-freertos/afr_pkcs11_helpers.a  amazon-freertos/afr_mqtt_demo_helpers.a  amazon-freertos/afr_device_shadow.a  amazon-freertos/afr_ble_wifi_provisioning.a  ../components/bme680/algo/normal_version/bin/esp/esp32/libalgobsec.a  ../amazon-freertos/vendors/espressif/esp-idf/components/newlib/lib/libm.a  -L /home/horsema [... cut ...]
ninja: build stopped: subcommand failed.

```
I did not understand why every time linking issue is coming. While I added the [path](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/blob/ad7710cb1c2c62bac88e98b03bf4482c0024b040/CMakeLists.txt#L18) and [AFR](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/blob/ad7710cb1c2c62bac88e98b03bf4482c0024b040/CMakeLists.txt#L62) in the [CmakeList.txt](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/blob/master/CMakeLists.txt#L62) file.

ravibhagavandas, 2021-06-01:
The default implementation for the hook can be found in file `aws_ble_gatt_server_demo.c` under folder [gatt_server](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/tree/ad7710cb1c2c62bac88e98b03bf4482c0024b040/amazon-freertos/demos/ble/gatt_server). 

So to link to the implementation, the file's cmake module needs to be added in [CMakeLists.txt here](https://github.com/Raghav3107/amazon-freertos-workspace-v202012/blob/ad7710cb1c2c62bac88e98b03bf4482c0024b040/CMakeLists.txt#L50), by linking as `AFR::demo_gatt_server` dependency

Raghav3107, 2021-06-02:
Thank you so much @ravibhagavandas  
I will try to do that. Till then, can you answer [this question](https://github.com/aws/amazon-freertos/issues/3169#issuecomment-848049631) which asked?

ravibhagavandas, 2021-06-07:
Hi @Raghav3107 

`AFR::demo_gatt_server` target is created as part of executing  [CMakeLists.txt](https://github.com/aws/amazon-freertos/blob/main/demos/ble/gatt_server/CMakeLists.txt#L3) from `demos/ble/gatt_server` folder. 

>I search from where it's adding the MAC address of the device in the advertisement parameter. But I did not found anything.
Can you tell me from where it adding the MAC address in the advertisement parameter? 

The mac address referred to here is the BLE address by which a central device scans and lists the device. Its a 48 bit address as well and is assigned only by the underlying (BLE) stack. In this cases its assigned by NIMBLE which is ESP32 BLE stack.

> Can we send other things instead of MAC address, dynamically?

BLE peripheral should always be configured with an address. Address can be of different types such as public or random (as referenced [here]( https://www.novelbits.io/bluetooth-address-privacy-ble/)).

> I was trying to add a second UUID of 128bit at the BLE advertisement parameter but I didn't see the other UUID instead It shows some message "Incomplete Lists of 128bit Service UUID". What does this mean?

It could be very well possible that the UUID list size is exceeding the advertisement limit which is 31 bytes. Adding two 128 bit UUIDs will lead to 32 bytes which itself is greater than limit of the advertisement payload, not considering the other fields in advertisement. Instead of sending two UUIDs in advertisement, you can:
1. Choose and send only one of the UUID as the primary 128 bit UUID. The current 128 bit UUID in advertisement can be overridden with your UUID by setting the following config:
```
 #define IOT_BLE_ADVERTISING_UUID   {  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }

Replace 0x00 with each bytes of your 128 bit UUID in little endian order.
```
The other service UUIDs will be discovered after connection is made.
2.  Add 16 bit UUIDs. For this set:
```
#define IOT_BLE_ADVERTISING_UUID   {  0x00, 0x00 }
Replace 0x00 with each bytes of your 16 bit UUID in little endian order.

 #define IOT_BLE_ADVERTISING_UUID_SIZE    2
```
Then you can set the 2nd UUID to either 16 bit or 128 bit UUID based on available space in advertisement.

Raghav3107, 2021-06-08:
Thanks, @ravibhagavandas  For answering the questions.
But my problem is not solved.
My problem is I want to send some dynamic parameter based on the thing name using ble before even ble is connected to the app. Like if the thing name is` A I will send 0x00 if the thing name is B I will send 0x01.` Based on the thing name I will change the `UUID`.
So, I thought ble advertisement parameter will be best for that. Now seeing yours and I think we cannot set a dynamic parameter to the ble advertisement. Can be?

closed 2021-06-09 by Raghav3107