aws/amazon-freertos #3174: Modifying the ota_agent on amazon freertos for esp32

Johnnyjax, 2021-05-25:
Hello, Amazon-freertos ota agent starts a job immediately the job is created on aws iot. I'm trying to modify the ota_agent on amazon-freertos so that it doesn't start the pending job until it receives an incoming message from a specific mqtt topic. Is there a way to achieve this? Thanks.

pvyawaha, 2021-05-25:
Hello, 

Are you trying to hold starting the job on the device on some application event on the device or the requirement is to send another command from the service to start the job on device?

Johnnyjax, 2021-05-26:
Hello @pvyawaha. There are two options I'm considering. The first is to hold starting the Job on the the device until a mobile application publishes  to a particular mqtt topic which the device would be subscribed to. The second is to hold starting the Job on the device  until a mobile app starts the Job by publishing to the **$aws/things/thingName/jobs/start-next** reserved topic to start the Job so that the Job can move to the "IN_PROGRESS" state. Which of these options is better and do you have any suggestions on how I can achieve this?

pvyawaha, 2021-06-07:
Hello,

You can shutdown and restart the OTA Agent based on some event and that can be a message on MQTTR topic in your application. You can use init and shutdown APIs for this. If you want to suspend the OTA Agent for short time you can use the suspend and resume APIs. Please refer the APIS here - https://freertos.org/Documentation/api-ref/ota-for-aws-iot-embedded-sdk/docs/doxygen/output/html/ota_design.html#ota_design_agent_api

closed 2021-06-09 by Johnnyjax