aws/amazon-freertos #3176: How to change a thing attribute from amazon-freertos

Johnnyjax, 2021-05-26:
Hello. Is it possible to change a Thing attribute directly from the amazon-freertos embedded code and is there any suggestions on how I can do that? Thanks

aggarw13, 2021-05-26:
Hi @Johnnyjax, can you please clarify about what you functionality about the IoT Thing you want to achieve from the application? 
If you want to just configure the IoT Thing name of your registered device in AWS IoT Core within the example demos of this repository, you can do so by defining the `clientcredentialIOT_THING_NAME` macro [here](https://github.com/aws/amazon-freertos/blob/main/demos/include/aws_clientcredential.h#L37-L46)?

Johnnyjax, 2021-05-26:
Hi @aggarw13. What I mean is changing certain thing "attributes" like searchable thing attributes for instance but in run time. Not defining the thing attribute directly in the macro before the device boots up.  So that if for instance, a specific message comes into an mqtt topic that the device is subscribed to, it changes the thing attribute to something else. Just want to know if there's a function available in amazon-freertos for doing that.

aggarw13, 2021-05-26:
Thanks for explaining your desired functionality of modifying IoT Thing attributes during runtime. 

We do not have an example of modifying Thing attributes at runtime as such an operation classifies as a "Control Plane" operation which are operations for administrative tasks like setting up AWS account resources like certificates, policies, things (along with thing attributes), etc. These operations are not typically performed by the embedded/IoT devices but instead performed by an administrator of the AWS account(s). These operations are usually performed with AWS CLI or AWS SDK. For example, this is the `update-thing` AWS CLI command that supports modifying attributes of a Thing: https://docs.aws.amazon.com/cli/latest/reference/iot/update-thing.html

(On the other hand, operations like publishing sensor data to IoT Core or communicating with AWS IoT Services like Device Shadow, Device Defender, etc, from the connected devices classify as "Data Plane" operations as these typically represent tasks for data ingested by IoT devices from field to IoT Core.)

It is still possible to modify IoT Thing attribute without using AWS SDK or AWS CLI by making an HTTP REST call to the [`UpdateThing`](https://docs.aws.amazon.com/iot/latest/apireference/API_UpdateThing.html) Control Plane API of IoT Core. The authentication mechanism for Control Plane API calls requires signing of the HTTP request with SigV4. This is automatically handled when using AWS CLI or AWS SDK tools, but when creating your own HTTP request to Control Plane APIs, you will need to sign the HTTP request by adding the SigV4 Authentication Header to the HTTP request. You can find more information on the signing process here: https://docs.aws.amazon.com/general/latest/gr/signing_aws_api_requests.html

Johnnyjax, 2021-05-26:
Thanks @aggarw13. I will try taking a look at making http requests with sigv4

closed 2021-05-26 by Johnnyjax