aws/amazon-freertos #3180: Receiving a very large incoming publish on Amazon-freertos

Johnnyjax, 2021-05-29:
Hello, I'm currently working on fleet provisioning on amazon freertos and I'm publishing to $aws/certificates/create/json topic and I'm supposed to receive a response on the $aws/certificates/create/payload-format/accepted topic with the certificate and key data. I got this error

`69 1046 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:872] 70 1046 [iot_thread] Incoming packet will be dumped: Packet length exceeds network buffer size.PacketSize=3603, NetworkBufferSize=1024.71 1046 [iot_thread]
72 1046 [iot_thread] [ERROR] [MQTT] [core_mqtt.c:845] 73 1046 [iot_thread] Dumped packet. DumpedBytes=3603.74 1046 [iot_thread]`

I went to the demo config file and changed the `#define democonfigNETWORK_BUFFER_SIZE    ( 1024U )` to `#define democonfigNETWORK_BUFFER_SIZE    ( 5000U )`

After I made that change, It no longer shows the error but I can't receive all the full information. Only a little bit of the certificate information comes in and it shows this

`81 1088 [iot_thread] [INFO] [ShadowDemo] [shadow_demo_main.c:57] 82 1088 [iot_thread] Incoming Publish Topic Name: $aws/certificates/create/json/accepted matches subscribed topic.Incoming Publish Message : {"certificateId":"7d69a84f1bd1089f2a28940d2668498283 1088 [iot_thread]`

Is there something I need to change to be able to receive all the information I need? Thanks

yourslab, 2021-05-31:
I believe there's a limit to how much you can print in a single line. That is defined here:
https://github.com/aggarw13/amazon-freertos/blob/de73cfc57c5f77e73ed813213f11d32c117c56b5/libraries/logging/iot_logging.c#L406-L409

To get around this, you could try to increase that limit. Another option is to add multiple `\r\n` delimiters to the string so that it doesn't get cut off.

Johnnyjax, 2021-05-31:
Thanks for the reply @yourslab. I tried increasing the limit you pointed out but it didn't work and the only the same amount of characters gets printed out as before. Is there another variable I need to adjust?

Johnnyjax, 2021-05-31:
Hello @yourslab, I decided to use the ESP_LOGI function instead and that worked. Thanks

closed 2021-05-31 by Johnnyjax