aws/amazon-freertos #3181: [General] Now that FreeRTOS 202104 is released. How to upgrade amazon-freertos to 202104 ?

teuteuguy, 2021-05-30:
Hi, 

FreeRTOS released 202104 and its not clear at all how to setup a new project for ESP32 using the tutorial provided by AWS: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project

How do you setup a new project and also upgrade to the latest libraries for coreMQTT and OTA ?

pvyawaha, 2021-06-01:
Hello,

Are you looking for integration with coreMQTT or coreMQTT Agent based OTA updates?

pvyawaha, 2021-06-14:
We don't have a specific date but we are actively working on the integration and testing.

LienNguyen2910, 2021-07-15:
teuteuguy, please check out amazon-freertos 202107.00 with pre-integrated projects for ESP32 platforms using coreMQTT, coreMQTT-Agent, OTA libraries, the same as in FreeRTOS 202104.00.

pvyawaha, 2021-08-05:
Hello ,

Please refer latest release and demos  for integration - https://github.com/aws/amazon-freertos/releases/tag/202107.00 

Closing this ticket now, please reopen for same topic or create a new one for other topics.

closed 2021-08-05 by pvyawaha