aws/amazon-freertos #3182: [General] Consuming freertos is my esp-idf project

Devyami-Apcon, 2021-06-01:
**Briefly summarize the issue being raised**
Is it possible to consume freertos v202007(esp-idfv3.3) in my esp-idf project which i will compile using esp-idf v4.0.2? 

**Describe the desired outcome**
Cutting it short, is it fundamentally possible to run both esp-idf and amazon freertos in the same executable ? 
That might eventually mean running 2 freertos on my esp32?

yourslab, 2021-06-03:
Trouble is that esp-idf v4 is using a revision of the FreeRTOS kernel that is different from the FreeRTOS kernel used in amazon-freertos. I've tried this myself before, and you will get build errors because the same function cannot be defined twice, especially when the same function has two different definitions. I understand that there are advantages to using esp-idf v4 such as easy integration with ESP-CAM and other features. However, the reason why the kernel is out of sync is because amazon-freertos tries to keep in sync with the latest version of the FreeRTOS kernel while esp-idf essentially has its own version that interfaces better with ESP boards.

Seems like the best way around this would be to pick and choose libraries from amazon-freertos and import them as esp-idf components. That way, you get the best of both worlds.

Devyami-Apcon, 2021-06-05:
@yourslab thank you for the clarification, cleared the doubt. 

But eventually what you're saying is using aws iot sdk with esp-idf. 

Which I feel needs a lot of porting knowledge. I posted an issue on espressif page too, they too are trying to figure out how exactly to port the latest aws sdk to their environment. 

Can you share any document or link on how can I port Amazon libraries as esp-idf component? 

Thanks

yourslab, 2021-06-05:
Yes, you can start by reading this guide: https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-guides/build-system.html

yourslab, 2021-06-07:
Hi @Devyami-Apcon. Actually, one of our partners from Espressif has successfully consumed FreeRTOS as an esp-idf component. I think this is what you need: https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2

yourslab, 2021-06-10:
@Devyami-Apcon You may also use this if you are only using coreMQTT: https://github.com/espressif/esp-freertos-coremqtt

Do you have any other questions you'd like to ask?

Devyami-Apcon, 2021-06-10:
> Hi @Devyami-Apcon. Actually, one of our partners from Espressif has successfully consumed FreeRTOS as an esp-idf component. I think this is what you need: https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2

@yourslab 
Yes I am using this for refrence. 

Though I had one doubt in this. How can I use multiple src executables in this? Also how to add the entire components folder directly like in esp-idf. 

Here they are using get_filename_component() function of cmake. Why do we need to use it? I have to build each component individual from the root cmakelist file. Unnecessary redundancies. Is there a workaround? 

Can not we directly use add_library and such intrinsic functions ?

Devyami-Apcon, 2021-06-10:
> @Devyami-Apcon You may also use this if you are only using coreMQTT: https://github.com/espressif/esp-freertos-coremqtt
> 
> 
> 
> Do you have any other questions you'd like to ask?

No I am looking for ota as well. 
I guess they are working on integrating the whole aws sdk as a component.

nova2882, 2021-06-22:
Hi @yourslab in a broader scheme are you saying that we should backport libraries to IDF instead of extending the current amazon-freertos.

yourslab, 2021-06-22:
It depends on what your goals are. Currently, amazon-freertos does not have multicore support for Espressif boards and if you would like to use separate IDF components, then it is easier to import those in by backporting the amazon-freertos libraries as an IDF component.

yanjos-dev, 2021-06-30:
I'm closing this issue due to inactivity. Please reopen this issue if you have any further questions.

closed 2021-06-30 by yanjos-dev