aws/amazon-freertos #3186: Compiling ESP32 project with FreeRTOS+TCP fails

htibosch, 2021-06-02:
Hello @shubhamkulkarni97 @tgsong @muneebahmed10 and other Espressif experts:

I downloaded the latest release of AFR, and I am trying to get it running with [FreeRTOS+TCP](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP).

The biggest problem is that make also includes lwIP files, and thus I get many conflicts like `wlanif_input()` being declared twice.
Also the esp version of `NetworkInterface.c` gets an error at `#include "esp_wifi_internal.h"`.

**System information**
- Hardware board: esp32 WROVER
- IDE used: Linux command line
- Operating System: Ubuntu
- Code version: git version 2.25.1
- Project/Demo: The standard ESP32 demo project

**Screenshots or console output**
Please compile the project with the command described here below and see what error will pop up.

**Steps to reproduce bug**
```
. vendors/espressif/esp-idf/export.sh
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -DAFR_ESP_FREERTOS_TCP=1 -S . -B build -GNinja
cd build
ninja
```
I started solving the issues, but I am giving up, hoping that it is easier for you to do.

Thanks, Hein

htibosch, 2021-06-02:
Aniruddha wrote me about PR #3115, which introduced `AFR_FREERTOS_TCP` instead of `AFR_ESP_FREERTOS_TCP`
I tried building the project with both macros, but without success.

AniruddhaKanhere, 2021-06-02:
@shubhamkulkarni97, I am fixing all these issues that @htibosch has pointed out. But, somehow I cannot remove all the lwIP dependencies. I found out that `___idf_log` has a dependency on lwIP. I am still trying to figure out where are these dependencies defined and how can I fix them. Can you help?

AniruddhaKanhere, 2021-06-03:
@shubhamkulkarni97, one instance seems to be coming from [here](https://github.com/espressif/esp-afr-sdk/blob/d58109d325586e86635d5647f71274e62e9fdf4e/tools/cmake/build.cmake#L157). I am still tracking down other instances where lwip is required.

AniruddhaKanhere, 2022-01-04:
Hello @htibosch,

Most ESP32 users get lwIP from Espressif-IDF. So, after a discussion, we decided to simplify the CMake to only use lwIP. I'll raise a PR shortly to remove instances of FreeRTOS+TCP from the CMake files.

I am closing this issue now. Feel free to open another one or re-open this one if you have anymore questions.

Thanks.

closed 2022-01-04 by AniruddhaKanhere