aws/amazon-freertos #3191: when the Internet is not availabl, lwip socket does not close

ohsungec, 2021-06-03:
Connected to wifi ap. However, the following phenomenon occurs when the Internet is not available.
1. MQTT PINGRESP Timeout occurs.
2. MQTT Reference count does not become 0.
3. The log below is not output.
(MQTT connection 0x3f81f610) Connection will be destroyed now.
3. Since the iwip socket is not closed, sockets_allocated does not add 1, so memory cannot be released.
====================================================================
1321 106450 [_createTaskPool] [ERROR][MQTT][1064500] (MQTT connection 0x3f828230) Failed to receive PINGRESP within 10000 ms.
1322 106450 [_createTaskPool] [INFO ][MQTT][1064500] ++_IotMqtt_CloseNetworkConnection
1323 106450 [_createTaskPool] [INFO ][MQTT][1064500] _IotMqtt_CloseNetworkConnection pNetworkInterface->close
1324 106521 [_createTaskPool] [INFO ][MQTT][1065210] (MQTT connection 0x3f828230) Network connection closed.
1325 106521 [_createTaskPool] [INFO ][MQTT][1065210] _IotMqtt_CloseNetworkConnection disconnectCallback.function
1326 106521 [_createTaskPool] [INFO ][HDC][1065210] Shadow Network Disconnect Callback !!
1327 106521 [_createTaskPool] [ERROR][HDC][1065210] Mqtt disconnected due to Keep-alive response not received.
1328 106521 [_createTaskPool] [INFO ][HDC][1065210] hdc_reset_message queue_count[32] xQueueReset
1329 106521 [_createTaskPool] [INFO ][HDC][1065210] MqttGetOperationMode[1]
1330 106521 [_createTaskPool] [INFO ][HDC][1065210] ++MqttShadowDeinit(0x3f828230)
1331 106521 [_createTaskPool] [INFO ][HDC][1065210] ++MqttOutSubscribeUnsubscribeTopics
1332 106522 [_createTaskPool] [INFO ][HDC][1065220] MqttOutReqTopics(47)[hdci/device/hdc_light_01/a1810510521cb68c62/out]
1333 106522 [_createTaskPool] [WARN ][MQTT][1065220] (MQTT connection 0x3f828230) Attempt to use closed connection.
1334 106522 [_createTaskPool] [ERROR][MQTT][1065220] (MQTT connection 0x3f828230) New operation record cannot be created for a closed connection
1335 106522 [_createTaskPool] [ERROR][HDC][1065220] MqttOutSubscribeUnsubscribeTopics out Subscription has error : Error Code = [1]
1336 106522 [_createTaskPool] [INFO ][HDC][1065220] --MqttOutSubscribeUnsubscribeTopics[10]
1337 106522 [_createTaskPool] [ERROR][HDC][1065220] Unsubscribe has error : Error Code = [1]
1338 106522 [_createTaskPool] [INFO ][HDC][1065220] ++MqttDisconnect2(0x3f828230)
1339 106522 [_createTaskPool] [INFO ][HDC][1065220] IotMqtt_Disconnect(0x3f828230)
1340 106522 [_createTaskPool] [INFO ][MQTT][1065220] ++IotMqtt_Disconnect
1341 106522 [_createTaskPool] [INFO ][MQTT][1065220] (MQTT connection 0x3f828230) Disconnecting connection.
1342 106522 [_createTaskPool] [INFO ][MQTT][1065220] ++IotMqtt_Disconnect 4 status[1]
1343 106522 [_createTaskPool] [INFO ][MQTT][1065220] ++_IotMqtt_CloseNetworkConnection
1344 106522 [_createTaskPool] [INFO ][MQTT][1065220] _IotMqtt_CloseNetworkConnection pNetworkInterface->close
1345 106522 [_createTaskPool] [WARN ][NET][1065220] Failed to close connection.
[00000010] DRAM(I[68516]+E[3947432])[4077404] IRAM[61456] free[3992288] FreeHeapSize [4077088k]
1346 106522 [_createTaskPool] [INFO ][MQTT][1065220] (MQTT connection 0x3f828230) Network connection closed.
1347 106522 [_createTaskPool] [INFO ][MQTT][1065220] _IotMqtt_CloseNetworkConnection disconnectCallback.function
1348 106522 [_createTaskPool] [INFO ][HDC][1065220] Shadow Network Disconnect Callback !!
1349 106523 [_createTaskPool] [ERROR][HDC][1065230] Mqtt disconnected due to invoking diconnect function.
1350 106523 [_createTaskPool] [INFO ][MQTT][1065230] --_IotMqtt_CloseNetworkConnection
1351 106523 [_createTaskPool] [WARN ][MQTT][1065230] (MQTT connection 0x3f828230) Reference count changed from 3 to 2.
1352 106523 [_createTaskPool] [INFO ][MQTT][1065230] --IotMqtt_Disconnect
1353 106523 [_createTaskPool] [INFO ][HDC][1065230] --MqttDisconnect2(0x0)->running time(10 msec)
1354 106523 [_createTaskPool] [WARN ][HDC][1065230] >>>>>[MQTT_SHADOW_DEINITED]
1355 106523 [_createTaskPool] [INFO ][HDC][1065230] --MqttShadowDeinit(0x0)->running time(20 msec)
1356 106523 [_createTaskPool] [INFO ][HDC][1065230] hdc_send_message(34, 0, 0x00000000)
1357 106523 [_createTaskPool] [INFO ][MQTT][1065230] --_IotMqtt_CloseNetworkConnection

yourslab, 2021-06-03:
Hi @ohsungec, can you please confirm the version of amazon-freertos that you are using? This bug may only appear in certain versions. Also, I highly recommend that you have your application use coreMQTT instead if possible.

ohsungec, 2021-06-03:
I used commit id #176636b9 in master branch
[image]

yourslab, 2021-06-03:
Okay, thanks. We will investigate further and get back to you as soon as we are able to solve the problem.

ohsungec, 2021-06-07:
Please share your progress.

yourslab, 2021-06-07:
We are still trying to find the best fix for the problem. Another thing to note is that this library will soon be removed from this repository. It is recommended to switch to coreMQTT.

muneebahmed10, 2021-07-21:
Hi @ohsungec,

This looks like it could be a similar issue as #2356. Either way, this deprecated version of the MQTT library was removed from the `main` branch of this repository, so there isn't anything to fix any more. I would recommend migrating to the latest release 202107.00, which uses the [coreMQTT](https://github.com/FreeRTOS/coreMQTT) and [coreMQTT-Agent](https://github.com/FreeRTOS/coreMQTT-Agent) libraries. We realize this may be disappointing if you have already been using the deprecated libraries, but you can find various examples to help getting started with the coreMQTT-Agent library [here](https://github.com/aws/amazon-freertos/tree/main/demos/coreMQTT_Agent) and [here](https://github.com/FreeRTOS/coreMQTT-Agent-Demos).

Please open a new issue if you have any further questions.

closed 2021-07-21 by muneebahmed10