aws/amazon-freertos #3194: how to update shadow on AWS more than 1 times.

SGOSAI602, 2021-06-04:
**Describe the bug**
execute device shadow for aws example only 2 time shadow update on cloud , so how we can number of time or continuous.
we change in below file but its same only two time update shadow.
[https://github.com/aws/amazon-freertos/blob/main/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L40](url)
`#define AWS_IOT_DEMO_SHADOW_UPDATE_COUNT        ( 30 )   /* Number of updates to publish. */` 
**System information**
- Hardware board: espressif esp32-WROOM-32D
- IDE used: esp-idf
- Operating System: Linux 
- Project/Demo: device_shadow_for_aws
- 
**Code to reproduce bug**
`Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory /home/shital/AWS/amazon-freertos
Executing "/home/shital/.espressif/python_env/idf4.2_py3.8_env/bin/python /home/shital/AWS/amazon-freertos/vendors/espressif/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- /home/shital/AWS/amazon-freertos/builde/aws_demos -m '/home/shital/.espressif/python_env/idf4.2_py3.8_env/bin/python' './vendors/espressif/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0' '-B' 'builde'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7584
ho 0 tail 12 room 4
load:0x40078000,len:13724
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2.1-21-gd58109d32 2nd stage bootloader
I (30) boot: compile time 02:14:28
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (86) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (93) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (101) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (108) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (116) boot: End of partition table
I (120) boot: No factory image, trying OTA 0
I (125) boot_comm: chip revision: 1, min. application chip revision: 0
I (132) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2e218 (188952) map
I (213) esp_image: segment 1: paddr=0x0004e240 vaddr=0x3ffbdb60 size=0x01dd8 (  7640) load
I (217) esp_image: segment 2: paddr=0x00050020 vaddr=0x400d0020 size=0xc7140 (815424) map
0x400d0020: _stext at ??:?

I (530) esp_image: segment 3: paddr=0x00117168 vaddr=0x3ffbf938 size=0x01cc4 (  7364) load
I (534) esp_image: segment 4: paddr=0x00118e34 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/shital/AWS/amazon-freertos/builde/../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1851

I (538) esp_image: segment 5: paddr=0x00119240 vaddr=0x40080404 size=0x1e28c (123532) load
I (616) boot: Loaded app from partition at offset 0x20000
I (652) boot: Set actual ota_seq=1 in otadata[0]
I (652) boot: Disabling RNG early entropy source...
I (652) cpu_start: Pro cpu up.
I (655) cpu_start: Application information:
I (660) cpu_start: Project name:     amazon-freertos
I (666) cpu_start: App version:      1
I (670) cpu_start: Compile time:     May 28 2021 02:14:24
[... 388 lines cut ...]
97 2442 [iot_thread] [INFO ][DEMO][24420] memory_metrics::demo_task_stack::before::bytes::5848

98 2442 [iot_thread] [INFO ][DEMO][24420] memory_metrics::demo_task_stack::after::bytes::2888

I (26174) wifi:state: run -> init (0)
I (26174) wifi:pm stop, total sleep time: 13661172 us / 21052245 us

I (26174) wifi:new:<4,0>, old:<4,0>, ap:<255,255>, sta:<4,0>, prof:1
I (26184) WIFI: WIFI_EVENT_STA_DISCONNECTED: 63
99 2543 [iot_thread] [INFO ][DEMO][25430] Demo completed successfully.

100 2544 [sys_evt] [INFO ][DEMO][25440] Disconnected from WiFi access point, reason code: 0.

E (26194) wifi_init: Wi-Fi not stop
I (26204) WIFI: WIFI_EVENT_STA_STOP
I (26204) wifi:flush txq
I (26204) wifi:stop sw txq
I (26214) wifi:lmac stop hw txq
I (26214) wifi:Deinit lldesc rx mblock:10
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
101 2549 [iot_thread] [INFO ][INIT][25490] SDK cleanup done.

102 2549 [iot_thread] [INFO ][DEMO][25490] -------DEMO FINISHED-------
`

yourslab, 2021-06-04:
Hi @SGOSAI602. I know it's confusing, but `AWS_IOT_DEMO_SHADOW_UPDATE_COUNT` does not affect the demo you are running (It only affects older demos that are using). If you'd like to make N shadow updates, it would just be a matter of making N MQTT publishes to the `/update` topic.

I'd also like to share the Shadow demo that utilizes the MQTT Agent, which would be the ideal way to make Shadow updates if you are also expecting to use other MQTT-dependent libraries using the same connection:
https://github.com/FreeRTOS/coreMQTT-Agent-Demos/blob/main/source/demo-tasks/shadow_device_task.c

SGOSAI602, 2021-06-04:
Hi yourslab, 
 Yes we are  confusing and not understand what u said.
we are using this example demo so , In which file and where to change so.
update shadow N times.
[https://github.com/aws/amazon-freertos/blob/30e14afb403d78732388461f14f1d4195e72e4ad/demos/device_shadow_for_aws/shadow_demo_main.c](url)

yourslab, 2021-06-04:
It's in this function call: https://github.com/aws/amazon-freertos/blob/30e14afb403d78732388461f14f1d4195e72e4ad/demos/device_shadow_for_aws/shadow_demo_main.c#L942-L946

The function `PublishToTopic` is defined here: https://github.com/aws/amazon-freertos/blob/30e14afb403d78732388461f14f1d4195e72e4ad/demos/common/mqtt_demo_helpers/mqtt_demo_helpers.c#L1036

yourslab, 2021-06-07:
Do you have any follow-up questions? @SGOSAI602

SGOSAI602, 2021-06-07:
Hi yourslab,

Thanks for your valuable reply.
I tried but I not exactly understand.
I saw PublishToTopic function but I did not understand when and what I change.

yourslab, 2021-06-07:
If you'd like to make a Shadow update N times, you can make a loop that calls that function I mentioned N times. You can change the contents of `pcUpdateDocument` along with `SHADOW_DESIRED_JSON_LENGTH` depending on what are the contents of the Shadow update.

yourslab, 2021-06-09:
Closing this issue because the original question has already been answered. If you still face more issues, feel free to open another.

closed 2021-06-09 by yourslab