aws/amazon-freertos #3196: Compiler not found, you can specify search path with "AFR_TOOLCHAIN_PATH".

adrian-zulnedi, 2021-06-05:
Hello, I have something problem which use Cmake for build FreeRTOS. I don't know how to fix. somebody can please help me? I try install cmake but doesn`t work in terminal.

```
macbookAir:FreeRTOS macbookair$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build
CMake Error at tools/cmake/toolchains/find_compiler.cmake:23 (message):
  Compiler not found, you can specify search path with "AFR_TOOLCHAIN_PATH".
Call Stack (most recent call first):
  tools/cmake/toolchains/xtensa-esp32.cmake:6 (afr_find_compiler)
  /opt/local/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:123 (include)
  CMakeLists.txt:8 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

adrian-zulnedi, 2021-06-05:
I don't know how to handle this problem, because I followed the getting started instruction exactly step-by-step.

yourslab, 2021-06-05:
I don't think you have `make` installed. It looks like you are using Mac, so try running `brew install make` (assuming you have `brew` installed. Also, it's important that you update your `PATH` as stated in the [Standard Setup of Toolchain for Mac OS](https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/macos-setup.html): `export PATH=$HOME/esp/xtensa-esp32-elf/bin:$PATH`

adrian-zulnedi, 2021-06-06:
thanks for your reply I just try your answer but it still doesn`t work.  And `brew install make@ is already installed. so I install again. 

[image]
[image]

adrian-zulnedi, 2021-06-06:
@yourslab 
thanks for you reply I just resolved it. But another have problem when I was install again 
`cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ../build`

```
MacBook-Air:macbookair$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ../build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/adrian_plugo/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Found Git: /usr/bin/git (found version "2.30.1 (Apple Git-130)") 
-- Checking Python dependencies...
The following Python requirements are not satisfied:
pyserial>=3.0
pyparsing>=2.0.3,<2.4.0
Please refer to the Get Started section of the ESP-IDF Programming Guide for setting up the required packages.
Alternatively, you can run "/System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -m pip install --user -r /Users/macbookair/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/requirements.txt" for resolving the issue.
CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:40 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:39 (include)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/macbookair/FreeRTOS-OTA/build/CMakeFiles/CMakeOutput.log".
See also "/Users/macbookair/FreeRTOS-OTA/build/CMakeFiles/CMakeError.log".
MacBook-Air:macbookair$ 

```

yourslab, 2021-06-06:
Run the command that it tells you to run in the error message

adrian-zulnedi, 2021-06-06:
@yourslab sorry I forget that one to tell you , I try too that. but It doesn`t work.

```

MacBook-Air:FreeRTOS macbook$ /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -m pip install --user -r /Users/macbookair/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/requirements.txt
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 111, in _get_module_details
    __import__(mod_name)  # Do not catch exceptions initializing package
  File "/Library/Python/2.7/site-packages/pip-21.1.2-py2.7.egg/pip/__init__.py", line 1, in <module>
    from typing import List, Optional
ImportError: No module named typing

```

adrian-zulnedi, 2021-06-06:
I didn`t know what I must install python module.

yourslab, 2021-06-06:
MacBook-Air:FreeRTOS macbook$ /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -m pip install --user typing

When it says what is missing, simply install it through pip..

yourslab, 2021-06-06:
Since the module requires typing as a dependency, change your version to Python 3 instead of using the default Python 2.7 installed in macOS. I don't think this can work with Python 2.7.

adrian-zulnedi, 2021-06-06:
@yourslab  thanks a lot for your reply. but I try to remove the version 2.7 it cann`t chaning how can I change to new 3.9?
I check in my finder but doesn't have python frame work verison3.X . when I was check python --verison is 3.9.5 but not have file 3.9.

[image]

adrian-zulnedi, 2021-06-06:
I try to remove 2.7 like `sudo rm -rf /System/Library/Frameworks/Python.framework//2.7` is work on terminal but When I was check in gui It still have it. so I try to reset my computer but which not remove the file.

adrian-zulnedi, 2021-06-06:
Did you have something good article to fix the version? or can you teach me how can I using to Python3?

yourslab, 2021-06-06:
https://docs.python-guide.org/starting/install3/osx/

adrian-zulnedi, 2021-06-06:
@yourslab thank you link but I just have try that one. so I try onther one like this . but I can`nt run cmake building again .

[image]

adrian-zulnedi, 2021-06-06:
this is which I change path to 3.9
`/opt/homebrew/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python -m pip install --user -r /Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/requirements.txt`

yourslab, 2021-06-06:
Yes, you installed it correctly but the build script is still using Python 2.7. You need to make sure that the directory containing Python 3.9 is the first directory that appears in your PATH. Also, check to see if you can call `python3` from the terminal. You may refer to this issue here for more details: https://github.com/espressif/esp-idf/issues/6532

adrian-zulnedi, 2021-06-06:
@yourslab thank you very much . I just resolved the version . It have nothing to do with python --verison . I have another case input that data with alias = idf_get. So Installed idf_get then remove that data which directory named build. But still have error again. Did you have something idea for fix this bug?

[image]

adrian-zulnedi, 2021-06-06:
@yourslab Based on the error in the above picture I have foud same errror but doesn't have answer in this URL.
https://www.esp32.com/viewtopic.php?t=19999

adrian-zulnedi, 2021-06-06:
Contents of Error 

```

WARNING: IDF_PATH environment variable is not cleared. 
    If CMake is generating an error, consider clearing the IDF_PATH environment 
    variable, and generating a clean build. This message can be ignored if 
    CMake was successful.
CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:24 (include):
  include could not find load file:

    components
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:39 (include)
  CMakeLists.txt:70 (include)

-- Checking Python dependencies...
Python requirements from /Users/adrian_plugo/esp/esp-idf/requirements.txt are satisfied.
CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:44 (idf_set_target):
  Unknown CMake command "idf_set_target".
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:39 (include)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/adrian_plugo/FreeRTOS-OTA/build/CMakeFiles/CMakeOutput.log".

```

adrian-zulnedi, 2021-06-06:
CmakeOutput.log

[image]

adrian-zulnedi, 2021-06-07:
I check my bash_profile for IDF_PATH when I was erase that IDF_PATH and save with `source ~/.bash_profile` and I runnning again I have another error again. I don`t have Idea for fix this bug.

[image]

adrian-zulnedi, 2021-06-09:
@yourslab Hi , sorry I take renew again the screen shourt. and I just check my IDF_PATH with printenv but still have it.

[image]

adrian-zulnedi, 2021-06-09:
Then I  test with another opertying system (Linux Ubuntu 20.04) but I still same like which I use Macbook M1.

yourslab, 2021-06-09:
@adrian-zulnedi Can you show you print your IDF_PATH? `echo $IDF_PATH`

yourslab, 2021-06-09:
Also, try running `unset IDF_PATH` then try to run `cmake ...` again

adrian-zulnedi, 2021-06-09:
@yourslab thank you so much I can build that cmake. Thanks to you, it really helped me.
Cloud you teach me which you found that article? about `unset IDF_PAT`

```
adriannoMacBook-Air:FreeRTOS adrian_plugo$ echo $IDF_PATH
/Users/adrian_plugo/esp/esp-idf
```

[image]

yourslab, 2021-06-09:
It said consider clearing IDF_PATH so it was just a hunch

adrian-zulnedi, 2021-06-09:
@yourslab thank you very much your are my hero!

yourslab, 2021-06-09:
No problem. I will close this ticket now but please feel free to open another issue if you face more problems in the future. Thank you.

closed 2021-06-09 by yourslab

abhibhatia98, 2021-06-17:
Hi @adrian-zulnedi  & @yourslab 

I am facing the same issue in  windows, please see below screenshot :

[image]

Python version is 3.9.5

Screenshot of environment variables are : 

[image]

Please check if you can help, I will be thankful.

Thanks

adrian-zulnedi, 2021-06-17:
@abhibhatia98 Hi, maybe have error like you when I was starting FreeRTOS. I hope you find it useful.

Did you make ESP-IDF tools? if doesn't have it you can make that in this link

https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/index.html#get-started-get-esp-idf-cmake

```
Step 2.Get ESP-IDF
Windows Command Prompt

mkdir %userprofile%\esp
cd %userprofile%\esp
git clone -b v3.3 --recursive https://github.com/espressif/esp-idf.git

```

if correct to Download you can go to that directory then try `./install.sh` if done try `. ./export.sh` then go back to your FreeRTOS file then try again cmake builder.

you can see my Ubuntu terminal. second picture have error but I think when you make it you not have something error like this . so don`t worry about this error.

[image]
[image]

if  you get this warning message.

```

WARNING: IDF_PATH environment variable is not cleared. 
    If CMake is generating an error, consider clearing the IDF_PATH environment 
    variable, and generating a clean build. This message can be ignored if 
    CMake was successful.
CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:24 (include):
  include could not find load file:

    components
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:39 (include)
  CMakeLists.txt:70 (include)

-- Checking Python dependencies...
Python requirements from /Users/adrian_plugo/esp/esp-idf/requirements.txt are satisfied.
CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:44 (idf_set_target):
  Unknown CMake command "idf_set_target".
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:39 (include)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/adrian_plugo/FreeRTOS-OTA/build/CMakeFiles/CMakeOutput.log".

```

try `unset IDF_PATH` then cmake again.

[image]

[image]

adrian-zulnedi, 2021-06-17:
@abhibhatia98 if just only error in xtensa-esp32-elf you must reinstall again in your local directory,

```
Toolchain Setup
Download the precompiled Windows toolchain:

https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip

Unzip the zip file to C:\Program Files (or some other location). The zip file contains a single directory xtensa-esp32-elf.

Next, the bin subdirectory of this directory must be added to your Path. For example, the directory to add may be C:\Program Files\xtensa-esp32-elf\bin.

```

https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/windows-setup-scratch.html

adrian-zulnedi, 2021-06-17:
@abhibhatia98  If it will help please let me know if there is any progress. Thank you.

abhibhatia98, 2021-06-17:
Hi @adrian-zulnedi, 

Firstly very grateful for your response. 

In this when I followed the second step, `git clone -b v3.3 --recursive https://github.com/espressif/esp-idf.git` it downloaded that repository which looks like this 

[image]

In this there is no install scripts, here I am stuck now ?

Screenshot  of git clone 

[image]

do you have any idea ?

Thanks 
Abhishek

adrian-zulnedi, 2021-06-17:
@abhibhatia98  maybe you can install like this video.Then If can be start you can try cmake build on your FreeRTOS folder.

https://www.youtube.com/watch?v=K58Y2dFsQ2A

```
ATTENTION: (ESPRESSIF OFFICIAL SITE DOCUMENT)

Now you can run get_idf to set up or refresh the esp-idf environment in any terminal session.

Technically, you can add export.sh to your shell’s profile directly; however, it is not recommended. Doing so activates IDF virtual environment in every terminal session (including those where IDF is not needed), defeating the purpose of the virtual environment and likely affecting other software.
````

abhibhatia98, 2021-06-17:
Hi,
Ok so I have setup the esp idf this way also , actually this is esp idf 4+ version , and the above your comments are related for idf3.3 version.(I have 2devices and have tried different version on both)

So as showing in this video I have done this way nd also yeat hello world example too that works fine.

After that I have downloaded the freeRTOS project from aws console and goes to that directory and run vendors\espressif\esp-idf\export.bat command as mention in aws page and that gives me error.

adrian-zulnedi, 2021-06-17:
@abhibhatia98 Cloud you show me that error details?

abhibhatia98, 2021-06-17:
Yiu can find that here https://github.com/aws/amazon-freertos/issues/3235

For the time being I can try on the same environment as you.I have idea that you ar working on ubuntu, could you please share the eap idf version and from where you have downloaded the freeRTOS project like from aws console or from github ?
I will so all same setup on ec2 instance. ans try that.

Thanks

adrian-zulnedi, 2021-06-17:
@abhibhatia98  

#STEP1 my esp-idf version.

[image]

# STEP2 I Download that xtensa-esp32-elf file on this page. I using Linux 64bit.

```

Toolchain Setup
ESP32 toolchain for Linux is available for download from Espressif website:

for 64-bit Linux:

https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

mkdir -p ~/esp
cd ~/esp
tar -xzf ~/Downloads/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

```

#STEP3 ABOUT FreeRTOS.

I using on aws FreeRTOS service. you can check that FreeRTOS guide

https://docs.aws.amazon.com/freertos/latest/userguide/freertos-getting-started.html

```

@abhibhatia98 I'm not work at freertos and beginners of freertos. So, actually I can't tell detail to you.  sorry ...
But I guess about when I was using opereting system in Macbook (intel and M1) have same error like your Windows10.
So when I was change in Linux Ubuntu 20.04 is cleary work (actually 80%. little bit some have error but is not difficult.)

adrian-zulnedi, 2021-06-17:
about little bit error you can try setup again. reinstall .

abhibhatia98, 2021-06-18:
Hi @adrian-zulnedi , well I have still issue in windows build, but for the time being I have setup the tool in ec2 instance in ubuntu environment, and that works well. Now I can build and compile them. Now I have binaries on ec2 instance, and I have downloaded that build folder to my local.

Could you please provide the screenshot of running flash command ?

Thanks 
Abhishek

abhibhatia98, 2021-06-21:
Hi @adrian-zulnedi, 

I have one more doubt, that after `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B your-build-directory `  command you ran `make all -j4 `  ??

Actually the first command works well, but when I ran second command it gives me error, when I downloaded the project from freeRTOS console.

Thanks