aws/amazon-freertos #3205: [General] <Failed to resolve component 'freertos' for ESP32>

sherindeena, 2021-06-09:
When I am trying to build the example I am getting this error for ESP32:
Failed to resolve component 'freertos'

D:\amazon-freertos\vendors\espressif\esp-idf\examples\wifi\getting_started\station>idf.py -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 build
Executing action: all (aliases: build)
Running cmake in directory d:\amazon-freertos\vendors\espressif\esp-idf\examples\wifi\getting_started\station\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DCCACHE_ENABLE=1 d:\amazon-freertos\vendors\espressif\esp-idf\examples\wifi\getting_started\station"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.31.1.windows.1")
-- Component directory D:/amazon-freertos/vendors/espressif/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/amazon-freertos/vendors/espressif/esp-idf/components/unity does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/amazon-freertos/vendors/espressif/esp-idf/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/amazon-freertos/vendors/espressif/esp-idf/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: D:/amazon-freertos/vendors/espressif/esp-idf/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/amazon-freertos/vendors/espressif/esp-idf/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: D:/amazon-freertos/vendors/espressif/esp-idf/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at D:/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'freertos'.
Call Stack (most recent call first):
  D:/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:212 (__build_resolve_and_add_req)
  D:/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:213 (__build_expand_requirements)
  D:/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:213 (__build_expand_requirements)
  D:/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  D:/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "D:/amazon-freertos/vendors/espressif/esp-idf/examples/wifi/getting_started/station/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

Can you please help me with this?

Johnnyjax, 2021-06-09:
Hello, I think you have to run the build command in the amazon-freertos source directory instead of D:\amazon-freertos\vendors\espressif\esp-idf\examples\wifi\getting_started\station. cd to D:\amazon-freertos and run the command there instead

mingyue86010, 2021-06-14:
Hi @sherindeena, I just check back to see if the above answer resolve your problem? Thx~

sherindeena, 2021-06-17:
@Johnnyjax thank you so much. it solves my issue.

closed 2021-06-17 by sherindeena