aws/amazon-freertos #3209: error: expected initializer before '__result_use_check' ,'__alloc_align' when I was use `make all -j4`

adrian-zulnedi, 2021-06-09:
Hi, I have something problem again with cmake builder. That Cmake builder can run but when I was run `make all -j4` have an error. I just do it step by step for preparing firmware (esp32) with AWS FreeRTOS. Some body can please help me?

@yourslab  Thank you during this period. I would be very happy if you could teach me again.

```
FreeRTOS_OTA/FreeRTOS/build$ make all -j4

In file included from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/cstdlib:75,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ext/string_conversions.h:41,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:6400,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/string:52,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/stdexcept:39,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/array:39,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/tuple:39,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/unique_ptr.h:37,
                 from /Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/memory:80,
                 from /Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/components/nvs_flash/src/nvs.hpp:18,
                 from /Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/components/nvs_flash/src/nvs_api.cpp:14:
/Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/Users/adrian_plugo/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)

make: *** [all] Error 2
```

adrian-zulnedi, 2021-06-09:
I just try remove that [build] on FreeRTOS and another case I make  [build] in FreeRTOS_OTA  folder.

Then I starting to use 
`cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ../build`

And `make all -j4` again but still have same error.

adrian-zulnedi, 2021-06-09:
Addition Information Error

Start with clear I have error package like this.(This picture sentence is Japanese. translate in english is mean can`nt open that package name)

Download xtensa-esp32-elf which if I run have error like this two picture.
https://docs.espressif.com/projects/esp-idf/en/v4.2.1/esp32/api-guides/tools/idf-tools.html

[image]
[image]

this one ↓ I download extensa file another one.
https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/linux-setup.html

[image]

yourslab, 2021-06-09:
The error is occurring because you are using the wrong compiler. Make sure you are using `xtensa-esp32-elf-gcc` because other compilers won't be able to build.

To fix this, I think you are using toolchain for Linux but you should be using MacOS toolchain instead: https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/macos-setup.html

adrian-zulnedi, 2021-06-09:
@yourslab thank you for reply my question again. I try that link when first I remove the xtensa-esp32-elf in my folder.
But If I was call get_idf I have error in `xtensa-esp32-elf-gcc` folder. 

[image]

yourslab, 2021-06-09:
Looks like a permission issue with your Mac. You need to update your Preferences so that it can be opened.

adrian-zulnedi, 2021-06-10:
@yourslab I just trying Preferences for permission data `extensa-esp32-elf` I think is work it but, Something like error or have a few warning.　Did you have idea for fix it? I just trying a few time for initiallize my data name `FreeRTOS_OTA` named build and FreeRTOS/build. I remove that data and I renew download `extensa-esp32-elf` on terminal again.

[image]

Download omitted

[image]

error or warning contents

```

Traceback (most recent call last):
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 2622, in parseImpl
    if (instring[loc] == self.firstMatchChar and
IndexError: string index out of range
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/tools/ldgen/generation.py", line 619, in add_sections_info
    results = parser.parseString(first_line)
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1814, in parseString
    raise exc
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1804, in parseString
    loc, tokens = self._parse( instring, 0 )
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 3705, in parseImpl
    loc, resultlist = self.exprs[0]._parse( instring, loc, doActions, callPreParse=False )
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1548, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 4067, in parseImpl
    return self.expr._parse( instring, loc, doActions, callPreParse=False )
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 1550, in _parseNoCache
    raise ParseException( instring, len(instring), self.errmsg, self )
pyparsing.ParseException: Expected "In archive" (at char 1), (line:2, col:1)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/tools/ldgen/ldgen.py", line 113, in <module>
    main()
  File "/Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/tools/ldgen/ldgen.py", line 88, in main
    sections_infos.add_sections_info(sections_info_file_obj)
  File "/Users/adrian_plugo/FreeRTOS-OTA/FreeRTOS/vendors/espressif/esp-idf/tools/ldgen/generation.py", line 621, in add_sections_info
    raise ParseException("File " + sections_info_file.name + " is not a valid sections info file. " + p.message)
  File "/Users/adrian_plugo/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/pyparsing.py", line 272, in __getattr__
    raise AttributeError(aname)
AttributeError: message
make[2]: *** [esp-idf/esp32/esp32.project.ld] Error 1
make[1]: *** [esp-idf/esp32/CMakeFiles/ldgen_esp32.project.ld_script.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Linking C static library libbootloader_support.a
warning: /Library/Developer/CommandLineTools/usr/bin/ranlib: archive library: libafr_3rdparty_mbedtls.a the table of contents is empty (no object file members in the library define global symbols)
[ 91%] Built target idf_component_bootloader_support
[ 79%] Built target afr_3rdparty_mbedtls
Scanning dependencies of target idf_component_bootloader_support_sections_info
[ 92%] Generating idf_component_bootloader_support.sections_info
[ 92%] Built target idf_component_bootloader_support_sections_info
Scanning dependencies of target ldgen_section_infos
[ 92%] Built target ldgen_section_infos
Scanning dependencies of target ldgen
[ 92%] Built target ldgen
Scanning dependencies of target bootloader
[ 94%] Building C object CMakeFiles/bootloader.dir/dummy_main_src.c.obj
[ 96%] Linking C executable bootloader.elf
[ 98%] Built target bootloader
Scanning dependencies of target app
[100%] Generating bootloader.bin
esptool.py v2.8
[100%] Built target app
[ 79%] No install step for 'bootloader'
[ 79%] Completed 'bootloader'
[ 79%] Built target bootloader
make: *** [all] Error 2

```

yourslab, 2021-06-10:
Hi @adrian-zulnedi. I'm not entirely sure as to why you are facing this problem with the esp-idf. Have you tried downgrading Python to 3.7? Also, try to delete `build` directory then re-run again.

adrian-zulnedi, 2021-06-10:
@yourslab hi , I try removed inside data that build directory on FreeRTOS_OTA then I re-run again . but It doesn`t work 100%.
I still Returned to the first stage problem again. when I put `make all -ja` on terminal that over 16% have error again.

adrian-zulnedi, 2021-06-10:
this is version of extensa

```

adriannoMacBook-Air:build adrian_plugo$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

adrian-zulnedi, 2021-06-10:
I guess which one must download file for Standard Setup of Toolchain for Mac OS? 
I forgeted which can be 100% that I put on terminal `make all -j4`
when you give me this link  it stoped beetwen 16%~20% .
https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/macos-setup.html

yourslab, 2021-06-10:
That is the right link. Note that you should be installing and using idf provided in `FreeRTOS/vendors/espressif/esp-idf`.

adrian-zulnedi, 2021-06-10:
Hi, @yourslab  I try another OS (Linux Ubuntu) is work it but still have again like yesterday.
I take a lot of screen shourt , sorry if hard to see this article.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

adrian-zulnedi, 2021-06-10:
Documents

https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/linux-setup.html

```

for 64-bit Linux:

https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

```

```
mkdir -p ~/esp
cd ~/esp
tar -xzf ~/Downloads/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
```

yourslab, 2021-06-11:
Actually, on Ubuntu, it built correctly but has a few warnings. You should be able to flash already.

adrian-zulnedi-plugo, 2021-06-11:
@yourslab thank you I will check again but I have question what is mean about flash? can you show me that command line?

adrian-zulnedi, 2021-06-12:
@yourslab thank you that right I just try that flash so it maybe can be . but I didn't know that serial port for esp32 in ubuntu when I use in MacBook I installed that serialport UART driver but if I wan`t to use on linux how can I change that serial port?

adrian-zulnedi, 2021-06-12:
Hi @yourslab I think this RTOS is work it.(But something have an error on my program so maybe later I check again.) 
I am apreciate with your Gentleman correspondence. Next time my project is to renew that device is mean Over The Air (OTA).
I really luck to gladd @yourslab . Thank you very much!

[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2021-06-12 by yourslab